hoglet67/RGBtoHDMI #1: Idea: use EDID for automatic adjustment to monitor geometry

BigEd, 2018-01-25:
If it can be done in bare metal, perhaps read the monitor's capabilities by EDID and adjust automatically?

In linux, you'd use `tvservice -d edid.dat ; edidparser edid.dat`

Perhaps TAG_GET_EDID_BLOCK in rpi-mailbox-interface.h is a clue.

The hope being that this removes the need to adjust the config file for different monitors.

hoglet67, 2018-01-25:
Yes, that would be a great feature.

What I'm not sure about is how easy it is to change the actual HDMI screen mode that's being used programatically. Currently this is set in config.txt, e.g.:
```
# 1920x1080p @ 50Hz
hdmi_group=1
hdmi_mode=31
overscan_top=-48
overscan_bottom=-48
overscan_left=240
overscan_right=240
framebuffer_aspect=-1
```
Dave

BigEd, 2018-01-25:
So do we need a baremetal equivalent to `tvservice --explicit=<string>`?
Ref:
https://raspberrypi.stackexchange.com/a/40681

hoglet67, 2018-01-25:
This looks relevant:
https://github.com/raspberrypi/firmware/issues/637

BigEd, 2018-01-25:
Good find!