hoglet67/RGBtoHDMI #38: V3: Drop HAS_MULTICORE from code base

hoglet67, 2019-03-09:
dmb@quadhog:~/atom/RGBtoHDMI/src/scripts$ grep HAS_MULTICORE ../*.[chS]
../armc-start.S:#ifdef HAS_MULTICORE
../armc-start.S:#ifdef HAS_MULTICORE
../armc-start.S:#ifdef HAS_MULTICORE
../armc-start.S:#ifdef HAS_MULTICORE
../arm-exception.c:#ifdef HAS_MULTICORE
../capture_line_atom_4bpp.S:#ifndef HAS_MULTICORE
../capture_line_atom_8bpp.S:#ifndef HAS_MULTICORE
../capture_line_mode7_4bpp.S:#ifndef HAS_MULTICORE
../defs.h:#define HAS_MULTICORE
../rgb_to_hdmi.c:#ifdef HAS_MULTICORE
../rgb_to_hdmi.c:#ifdef HAS_MULTICORE

IanSB, 2019-04-11:
Have you actually tried the current code on a Pi 3 or 3+ recently since all the optimisation changes were made?

hoglet67, 2019-04-11:
No I haven't, have you?

hoglet67, 2019-04-11:
BTW, I really have little interest in supporting more than one hardware platform, unless there's a really good reason.

IanSB, 2019-04-11:
I was just curious if the arm optimisations made any difference. Also I suppose that one day the core in the Pi 3 might be used on a next generation Pi zero.
I just tried on a Pi 3+ but it crashed: (I had to update bootcode.bin etc to the latest versions to get it to boot at all)

INFO: ***********************RESET***********************
INFO: RGB to HDMI booted
INFO: CPLD  Design: Normal
INFO: CPLD Version: 6.4
EMMC: BCM2708 controller power-cycled
EMMC: controller did not reset properly
WARN: Failed to initialize file system
EMMC: BCM2708 controller power-cycled
EMMC: controller did not reset properly
EMMC: BCM2708 controller power-cycled
EMMC: controller did not reset properly
WARN: Failed to initialize file system
INFO: Loading file /config.txt
EMMC: BCM2708 controller power-cycled
EMMC: controller did not reset properly
WARN: Failed to open /config.txt (result = 13)
EMMC: BCM2708 controller power-cycled
EMMC: controller did not reset properly
WARN: Failed to initialize file system
EMMC: BCM2708 controller power-cycled
EMMC: controller did not reset properly
EMMC: BCM2708 controller power-cycled
EMMC: controller did not reset properly
WARN: Failed to open profile /Profiles/Default.txt (result = 13)
main running on core 0
starting core 1
starting core 2
starting core 3
INFO: Analyze Csync: polarity changed to inverted
INFO: Detected polarity state at startup = +H-V (Separate H & V CPLD)
INFO: -----------------------LOOP------------------------
INFO: Analyze Csync: polarity unchanged (inverted)
INFO: Detected polarity state = +H-V (Separate H & V CPLD)
INFO:         clkinfo.clock = 16000000 Hz
INFO:      clkinfo.line_len = 1024
INFO:     clkinfo.clock_ppm = 5000 ppm
Undefined Instruction at 01F038DC
Registers:
...

hoglet67, 2019-04-11:
Seems the EMMC library (in sdcard.c) contains a hard-coded peripheral address:
```
#define EMMC_BASE    0x20300000UL
```
We should include ../rpi-base.h and change it to:
```
#define EMMC_BASE    (PERIPHERAL_BASE  +  0x300000UL)
```

hoglet67, 2019-04-11:
Regarding the illegal instruction, you can produce a full object dump using
```
arm-none-eabi-objdump -D rgb-to-hdmi > file.txt
```
It may be different in your kernel, but here's the offending region:
```
01f03898 <waitlo72>:
 1f03898:	ee198f1d 	mrc	15, 0, r8, cr9, cr13, {0}
 1f0389c:	e058800e 	subs	r8, r8, lr
 1f038a0:	42688000 	rsbmi	r8, r8, #0
 1f038a4:	e3580b4a 	cmp	r8, #75776	; 0x12800
 1f038a8:	e5948000 	ldr	r8, [r4]
 1f038ac:	c2288502 	eorgt	r8, r8, #8388608	; 0x800000
 1f038b0:	e3180502 	tst	r8, #8388608	; 0x800000
 1f038b4:	1afffff7 	bne	1f03898 <waitlo72>
 1f038b8:	ee198f1d 	mrc	15, 0, r8, cr9, cr13, {0}
 1f038bc:	e058800e 	subs	r8, r8, lr
 1f038c0:	42688000 	rsbmi	r8, r8, #0
 1f038c4:	e3580b4a 	cmp	r8, #75776	; 0x12800
 1f038c8:	e5948000 	ldr	r8, [r4]
 1f038cc:	c2288502 	eorgt	r8, r8, #8388608	; 0x800000
 1f038d0:	e3180502 	tst	r8, #8388608	; 0x800000
 1f038d4:	1affffef 	bne	1f03898 <waitlo72>
 1f038d8:	e1510000 	cmp	r1, r0
 1f038dc:	0e1f7f3c 	mrceq	15, 0, r7, cr15, cr12, {1}   <<<<<<<<<<
 1f038e0:	e2511001 	subs	r1, r1, #1
 1f038e4:	0e1f6f3c 	mrceq	15, 0, r6, cr15, cr12, {1}
 1f038e8:	1affffd8 	bne	1f03850 <measure_n_loop>
 1f038ec:	e0560007 	subs	r0, r6, r7
 1f038f0:	42600000 	rsbmi	r0, r0, #0
 1f038f4:	e8bd9ff0 	pop	{r4, r5, r6, r7, r8, r9, sl, fp, ip, pc}
```
Which is:
```
measure_n_loop:
        WAIT_FOR_CSYNC_1
        WAIT_FOR_CSYNC_0
        cmp    r1, r0
        mrceq  p15, 0, r7, c15, c12, 1
        subs   r1, r1, #1
        mrceq  p15, 0, r6, c15, c12, 1
        bne    measure_n_loop
```
I think on the Pi 3 the cycle counter is in a different Co Pro register:
```
   mrceq p15,0,r7,c9,c13,0
```
So a few thinks to fix.

But it would be interesting to try this again, I agree.

IanSB, 2019-04-12:
I made the above change to sdcard.c and changed measure_n_loop to:
````
measure_n_loop:
        WAIT_FOR_CSYNC_1
        WAIT_FOR_CSYNC_0
        cmp    r1, r0
#if defined(RPI2) || defined(RPI3)
        mrceq  p15, 0, r7, c9, c13, 0
#else
        mrceq  p15, 0, r7, c15, c12, 1 
#endif        
        subs   r1, r1, #1
#if defined(RPI2) || defined(RPI3)
        mrceq  p15, 0, r6, c9, c13, 0
#else
        mrceq  p15, 0, r6, c15, c12, 1 
#endif        
        bne    measure_n_loop
````
and that fixed the reset error and crash and the code gets a lot further but it hangs at setting up the frame buffer (the screen stays on the four colour squares):
````
...
INFO: Actual frame time = 16704540 ns (non-interlaced), line time = 45765 ns
INFO: Window: H = 63549 to 64124, V = 16586289 to 16736364, S = +H+V
INFO: Looking for autoswitch match = 45765, 365, 1
INFO: Autoswitch match: EGA (3) = 45558, 45969, 365, 1
INFO: Setting subprofile to 3
INFO: Analyze Csync: polarity unchanged (inverted)
INFO: Detected polarity state = +H-V (Separate H & V CPLD)
INFO:         clkinfo.clock = 16257000 Hz
INFO:      clkinfo.line_len = 744
INFO:     clkinfo.clock_ppm = 4500 ppm
INFO:     Nominal 100 lines = 4576400 ns
INFO:      Actual 100 lines = 4576570 ns
INFO:           Clock error = 37 PPM
INFO:  Error adjusted clock = 16256396 Hz
INFO:         GPCLK Divisor = 12
INFO: Target core frequency = 390153504 Hz
INFO:       Final core freq = 400000000 Hz
INFO:    CPLD sampling freq = 97538376 Hz
INFO:       Lines per frame = 365, (365.001)
INFO: Actual frame time = 16704495 ns (non-interlaced), line time = 45765 ns
INFO: Window: H = 45558 to 45969, V = 16628670 to 16778685, S = +H-V
INFO: Initialised Framebuffer: 800x400
````
From the above it looks like it's setting up the profile correctly but the core freq is not being set as well as the hangup.

hoglet67, 2019-04-12:
I think the core frequency not being set relates to using the latest firmware. The same happens on the Pi Zero. I filed a bug on the Pi Firmware issue tracker, but it seems to have been ignored:
https://github.com/raspberrypi/firmware/issues/1022
I really would like top get to the bottom of this issue. It's one of the reasons that I wanted to try using PLLA as the clock source. See #9.

Regarding the black screen. I suspect the ARM code has got an incorrect address for the framebuffer address. I did have issues with this for a while. There are a couple of versions of that code, controlled in defs.h by USE_PROPERTY_INTERFACE_FOR_FB. I think on the Pi 3 it was only working if called before the caches were enabled.

I do have an old Pi 3, but it's now partly broken. It boots Raspbian fine, but gets rather warm, and won't boot either PiTubeDirect or RGBtoHDMI. I think it's now got some damaged GPIOs that have gone short circuit.

IanSB, 2019-04-12:
It looks like the Pi3B+ requires the buggy firmware just to boot so it will never work on that model until the issue is fixed. I reverted to a Pi3 and eventually got it to boot by uncommenting USE_PROPERTY_INTERFACE_FOR_FB and making a debug build (i.e. bash ./configure_rpi3.sh -DDEBUG=1).
For some reason the non debug build doesn't work, it boots and the system is running as the serial log responds to keypresses but there is no HDMI output at all (the monitor goes to sleep so not just a black screen)
Testing:
It's still glitchy with double height enabled and also when mode 7 deinterlacing is enabled.
Interestingly there is less glitching with the more complex capture loops. e.g. in mode 0 the default loops are glitchy but the inband loops less so.
One strange thing I noticed is that mode 7 bob deinterlace drops multiple characters at one position:
[image]
Similarly advanced deinterlace starts messing up at the same point:
[image]
This is with deinterlace disabled:
[image]
I recall Dominic mentioning that cache misses can take many cycles which might explain the gap in the bob deinterlace image above so perhaps the issue is that the screen memory is cached by default on the Pi3 but not on the Pi zero.

hoglet67, 2019-04-12:
In my experience, although the memory bandwidth on the Pi 3 is higher, so is the latency when there is a cache miss, which we are more sensitive to.

BTW, I posted to photos of the Atom Video to HDMI adaptor PCB on Stardot:
https://stardot.org.uk/forums/viewtopic.php?f=44&t=16907

It's pretty much finished now.

IanSB, 2019-04-12:
> In my experience, although the memory bandwidth on the Pi 3 is higher, so is the latency when there is a cache miss, which we are more sensitive to.

I just did some memory reading tests (read same sequential 4K of memory 1024 times) and the results were as follows:
````
For the Pi zero:
INFO: Time to read main memory   =   3164982
INFO: Time to read screen memory = 103189128
For the Pi 3:
INFO: Time to read main memory   =   2113826
INFO: Time to read screen memory = 120997838
````
So the screen ram is uncached but the question is where is that huge stall coming from in the bob code as it is missing several psync edges. There can't be any cache misses in the code part so maybe it's running out of write buffers but if that was the case there would be a noticeable difference between 4bpp and 8bpp but it doesn't look like there is any difference. Perhaps there is variable read/write latency even for uncached memory.

> BTW, I posted to photos of the Atom Video to HDMI adaptor PCB on Stardot:

Looks good.

hoglet67, 2019-04-12:
Yeh, I agree. If you look at the second image, it looks like the shifts are approx 8 pixels, which is two psync edges. Given this is Mode 7, that's a stall of 600-700ns (or cycles).

One technique we used to profile critical code in PiTubeDirect was to set/clear a spare GPIO pin around a critical section of code, and then look at it on a scope. Not sure whether that would be useful here. But it might allow us to see whether the stalling is the memory read, the memory write, the GPIO read, or all of the above.

I'm not really convinced this is worth pursuing though.

Dave

IanSB, 2019-04-12:
> I'm not really convinced this is worth pursuing though.

Yes, I'm not going to look at this any further at the moment as the Pi zero already works well enough and is much cheaper. The Pi 3 is usable on the beeb/electron and for CGA/EGA PC timings as long as you disable double height frame buffers (which disables scanlines) and switch off deinterlacing for mode 7.

I think we should push for a fix on the clock change or come up with a direct register workaround as there may come a time when a later revision of the Pi zero BCM2835 requires newer firmware.

hoglet67, 2019-04-26:
Closing as wontfix, as it's possible in the future we'll try again to get this working on the Pi 3.

closed 2019-04-26 by hoglet67