hoglet67/RGBtoHDMI #43: Screen capture extremely slow or not working

IanSB, 2019-03-11:
It takes about 40 seconds to write a capture to a sandisk card and it seems to hang up completely when writing to kingston or verbatim cards (waited a few mins).

hoglet67, 2019-03-11:
I'm in the middle of debugging this.

There seems to be two issues:

1. I'm seeing occasional SD command errors, which are resolved if I slow the SD Card clock down. It's currently fixed at 25MHz.

2. After each multi-block write command (of 8 512b sectors) there is a full re-initialization of the EMMC hardware and SD Card. This I think is taking ~100-200ms and is what's clobbering the write performance.

I need to test if both of these error also happen on Dom's Pi1MHzBus port of BeebSCSI.

The files are also currently huge (~1MB) as there is no compression. Switching to LodePNG will fix this I hope, but might mask the SD card write issue, which is more concerning.

Dave

hoglet67, 2019-03-11:
Above post edited.

For reference, some links related to bugs in John Cronin's SD Card Library:
https://www.raspberrypi.org/forums/viewtopic.php?f=72&t=94133

hoglet67, 2019-03-11:
Ian,

I've added a work around to the SDCard driver that really helps with the performance.

As far as I can tell Multi Block Writes (CMD25) were seriously broken, because the card ended up in the wrong state. It least for the cards that I have, this workaround really helps.

I'm going to have a play LodePNG library now....

Dave

hoglet67, 2019-03-11:
Ian,

I've now switched over to the LodePNG library (both are still available in the code).

I've also slowed the SD Card clock down from 25MHz to 20MHz. This seems to have helped a bit, but I'm still seeing occasional errors like:
```
EMMC: BCM2708 controller power-cycled
INFO: Screen capture starting, file = capture34.png
INFO: Screen capture PNG length = 10396, writing data...
SD: error sending CMD24, error = 01048576.  Retrying...
SD: error sending CMD24, error = 00065536.  Retrying...
```
Those error codes are 2^20 and 2^16 which correspond to:
#define CMD_TIMEOUT(a)      (FAIL(a) && (a->last_error & (1 << 16)))
#define DATA_TIMEOUT(a)     (FAIL(a) && (a->last_error & (1 << 20)))

See if this works any better for you.

I'll keep digging into these errors. I did read there were some bugs in the SD card driver code to do with setting the clock speed.

Dave

hoglet67, 2019-03-11:
Last word on this (I hope)...

I've just done some better fixes for the multi block write and timeout issues.

I've not seen any problems since, and I've tried a couple of different cards.

Closing for now.

closed 2019-03-11 by hoglet67