hoglet67/RGBtoHDMI #49: V3: Add back HDMI Calibration (long press of cal button?)

hoglet67, 2019-03-11:
Tried this a couple of days ago, and it nearly worked, but some of the OSD messages were lost.

IanSB, 2019-03-12:
Perhaps restore both buttons as they were and press both calibration buttons simultaneously to do a screen grab (this would also work in a menu). You would have to change the action on those two buttons to happen on release rather than press to get that working reliably.

IanSB, 2019-03-24:
Another idea:
Use short press for HDMI calibration and long press for screen capture. If you press the menu button before the long press has timed out you, the menu is put on screen before the capture is made thus addressing https://github.com/hoglet67/RGBtoHDMI/issues/48 as well

hoglet67, 2019-04-26:
I've added:
- short cal button press <0.5 s = HDMI Calibation
- long cal button press >= 0.5s = Auto Calibration

Doing this exposed a nasty issue with palette mailbox messages getting lost. That's fixed in a seperate commit: 873166aa

closed 2019-04-26 by hoglet67