hoglet67/RGBtoHDMI #50: V3: Design new configuration file format

hoglet67, 2019-03-11:
My current thought is to stop using cmdline.txt, and replace it with:
- a base config file for common settings
- one or more machine (or system) specific config files

Or maybe three levels:
- a base
- a machine type specific (e.g. Beeb)
- a machine instance specific (e.g. Dave's Issue 3 Beeb)

The same multi-line syntax the config.txt would be used

A particular setting could be set at any level, and the mode specific specific file would override the base file. I'm thinking these would just be read in order.

The system would search the SD card for these files, and populate a menu that would allow allow the user to easily select a new configuration.

Initially these files would be managed manually.

Later, any updated settings could optionally be written back to the currently selected configuration. (This needs careful thinking through. How to track what has changed, and is different to the value in the base config).

Sounds simple, but I bet it's not!

One wrinkle is how to deal with machines that have different modes (like the Beeb with Mode 0..7 and Mode 7). Are these done with separate files? Or with sections within a file (like config.txt allows conditional filters):
https://www.raspberrypi.org/documentation/configuration/config-txt/conditional.md

IanSB, 2019-03-11:
> Later, any updated settings could optionally be written back to the currently selected configuration. (This needs careful thinking through. How to track what has changed, and is different to the value in the base config)

Perhaps we need to manage the settings files in folders, a default folder with original files that can only be edited manually and a current folder where any changed configs would be saved with the same name which would override the default unless a restore default setting option was selected in which case the current file is deleted and the default takes over.
e.g. the default folder contains:
BBC.txt
Electron.txt
ZX81.txt
UK101.txt
PC CGA.txt
PC EGA.txt
Custom1.txt
Custom2.txt
Custom3.txt
etc.
The current folder woud be empty and the menu would show the filenames as a list of options to select
if you edit one and save settings it would create the same named file in the current folder which would then override the default until deleted by a restore default option

IanSB, 2019-03-11:
Also should be able to select the screen resolution, refresh rate and scaling kernel.
We can't change these directly in the hardware but could save an updated config file and prompt for a reboot.

IanSB, 2019-03-12:
> One wrinkle is how to deal with machines that have different modes (like the Beeb with Mode 0..7 and Mode 7). Are these done with separate files? Or with sections within a file (like config.txt allows conditional filters):

I think in the same file similar to the command line with its 06 & 7 settings

IanSB, 2019-04-07:
I think this can be closed now as outstanding issues are covered by 
https://github.com/hoglet67/RGBtoHDMI/issues/73
https://github.com/hoglet67/RGBtoHDMI/issues/74
https://github.com/hoglet67/RGBtoHDMI/issues/75

closed 2019-04-07 by hoglet67