hoglet67/RGBtoHDMI #54: V3 Investigate possibility of adding half odd/even to CPLD

IanSB, 2019-03-12:
I've just been testing mode7 at 192 MHz clock rate with a new mode 7 half odd/even sample loop and it seems to be completely stable. Although this is well out of spec it is potentially useful for those beebs with a very asymmetric 12 Mhz clock as it doubles the resolution and it removed the last twinking pixels on one of my systems. Unfortunately with double the psync rate there is no time to read memory and so deinterlacing doesn't work.
Could this be implemented in the CPLD with a switch that enabled half odd/even mode? (i.e. put every other pixel in the shift reg and halve psync)
This would work with the existing capture loop without any changes.

To get most benefit you would probably need to increase the offsets to 4 bits to make full use of the double resolution and there's not enough registers to do that but maybe worth some experimentation.
The odd/even bit would be acting as the global MSB of all offsets (or maybe acting as the global LSB would be better?)

hoglet67, 2019-03-12:
I've had a go at this: you can find it in the cpld_v4 branch.
https://github.com/hoglet67/RGBtoHDMI/tree/cpld_v4

I had a lot of trouble getting it to fit, and in the end had to trade extra macro cells for product term savings (by pipelining a couple of signals). This means that currently all macro cells are used. I could do with one more, to give psync an extra cycle of skew (i.e. 2 rather than 1, to be safe).

There's definitely not enough registers to increase offset to 4 bits. But you do have the full resolution, because of the choice between half-odd and half-even. Auto calibration would need to search both, and it's possible there are edge cases if the ideal sampling points straddle the boundary.

I thought about reducing delay, but doesn't that need to be 4-bits for the worst case alignment in Mode 7? On my beeb delay is currently 11. BTW, there was additional work in the CPLD so that delay works units of whole pixels in he half sampling mode. This cost an extra counter bit.

According to the timing analysis, it's guaranteed to work at ~90MHz (which is the same as before).

I'm far from sure we should go with this... it's consumed pretty much all the remaining resources in the CPLD. Hence it's currently in a separate branch. I want to see if it gives better calibration results for me as well, on Beebs that are artificially clobbered (i.e. change the capacitor back again).

By the way, did you do the capacitor mod to your problematic beeb? It would be good to see what's actually wrong with the 6MHz clock.

Dave

hoglet67, 2019-03-12:
Done a couple more changes, to allow for arbitrary skew on PSync, and save a register at the same time! Amazing what's possible with a bit of head scratching. I'm now warming to this version....

IanSB, 2019-03-13:
This is looking very promising.
In my case, when even is selected I get very poor calibration with every position having some amount of noise. When odd is selected I get excellent calibration with a wide band of noise free settings.
If we go with this, if one of the odd / even options is already selected in mode 7 then the calibration should try both and switch to the one with the best result.
I've also been trying this version with some TTL VGA signals and managed to get 640x480@60Hz locked cleanly at 6 bits per sample. I have a card with both analog VGA and TTL EGA type connectors but it only seems to output VGA timings on the EGA connector but after some further code optimisation I managed to get it to lock cleanly with no noise. This is with a pixel clock of 25.175Mhz so real EGA should work no problem.
[image]
settings were:
sampling=0,3,3,3,3,3,3,0,6,4,1 geometry=10,25,81,490,648,490,0,8,151050000,4800,5000,0
One issue is that we definitely need a sync inversion bit. I've added one as a test and it fitted OK but the other issue is that to make inverted sync detection easier I think that VSYNC should be separate from HSYNC (on the CSYNC line) so we need to remove the xor of the two and feed the VSYNC directly to the PI for software control. 
I think this could be achieved by connecting the link through to the spare pin on the CPLD an then onto the GPIO 22 (making GPIO 22 an input instead of an output and making spare an output) or alternatively cut the track to GPIO 22 and feed VSYNC straight in.

hoglet67, 2019-03-13:
That's great news about TTL VGA, and rather amazing. That's a PSYNC clock speed of 6.29375MHz compared to the normal sampling Beeb rate of 2MHz (If I have my sums right). So more than 3x faster. That's pretty amazing. You should post that captured screen shot in thread!

On the half-rate sampling, something is now quite adding up for me here....

All that's happening in this mode with a double speed clock is that the sample positions are closer together. If you are seeing a wide band of noise free settings (i.e. say 3 or more) then in normal mode at least one position should also be noise free. If not, then something unexpected is happening.

I need to do some more testing with my Beeb with a socketed capacitor.

Dave

IanSB, 2019-03-13:
> All that's happening in this mode with a double speed clock is that the sample positions are closer together. If you are seeing a wide band of noise free settings (i.e. say 3 or more) then in normal [...]

I think it's working more or less as expected.
With 96Mhz my best alignment was at around 5 +-1 but it wouldn't always give perfect results and there was still some twinkling.
With 192 Mhz most of the even alignment was poor which corresponds to the lower half of the 92 Mhz alignment (0-3) the odd alignment was good which corresponds to the upper half of the 92 Mhz alignment (4-7) since 5 is in the middle of that range it's not surprising that there is now a relatively good band around the centre of the offset range of 0-7 in 192 Mhz mode. Effectively I'm getting the equivalent of 5 +- half a pixel at 92 Mhz and that half seems to pick the best result.

hoglet67, 2019-03-14:
I think this can now be closed.

closed 2019-03-14 by hoglet67