hoglet67/RGBtoHDMI #66: Improvement: Configurable button actions

hoglet67, 2019-04-04:
Add support for detecting long and short button presses:
- A short button press action could happen on release
- A long button press action could happen on the "long press detection" timeout

Make the binding of the various actions to buttons configurable.

Actions would be:
- menu launch
- hdmi calibration (i.e. genlock)
- auto calibration (i.e. sampling clock calibration)
- take screen shot

Any others?

This would fix #24

IanSB, 2019-04-05:
scanlines on / off
interpolation on / off
(or maybe a combination of the two on a single button)
NTSC artifacting on / off

hoglet67, 2019-04-26:
This is now implemented.

Feel free to add additional actions - there are three palace holders (A5_SPARE, A6_SPARE, A7_SPARE)

closed 2019-04-26 by hoglet67