hoglet67/RGBtoHDMI #67: Improvement: Consider whether the pixel sampling geometry option is obsolete

hoglet67, 2019-04-04:
At the very least, it's confusing that we have very similar options in both the geometry and sampling menu.

The one in the sampling menu ("Sample mode") specific to the "Normal CPLD" and controls the hardware sampling rate / size:
   "Normal (3bpp)",
   "Double (6bpp)",
   "Half-Odd (3bpp)",
   "Half-Even (3bpp)"

The one in the geometry menu ("Pixel sampling")  controls how the software pixel copy loops handle each pixel:
   PS_NORMAL,    // Each sampled pixel is mapped to one pixel in the frame buffer
   PS_NORMAL_O,  // Odd pixels are replicated, even pixels are ignored
   PS_NORMAL_E,  // Even pixels are replicated, odd pixels are ignored
   PS_HALF_O,    // Odd pixels are used, even pixels are ignored
   PS_HALF_E,    // Even pixels are used, odd pixels are ignored
   PS_DOUBLE,    // pixels are doubled

Some of the pixel sampling options were useful for the Atom (using the Beeb adapter) with older CPLDs.

A second consideration is it allows a 2x width and 2x height frame buffer to be used, to get the better looking OSD font.

The benefit of rationalising this would be to reduce the size of  capture line tables, meaning there is more change they could all be fully implemented. 

In the capture line table, we have the following tables (15 in total):
- default (4bpp and 8bpp) x (normal vs normal_odd vs normal_even vs half_odd vs half_even vs double)
- mode 7 (4bpp)
- atom (4bpp and 8bpp)

Each table has 8 entries:
- 3bpp vs 6bpp (pixel samples)
- Off vs Fast vs Inband vs NTSC artifacting (palette control)

IanSB, 2019-04-14:
I've been thinking about this and have come up with a use case for something along the above lines but implemented in a different way.

Currently PS_NORMAL_O, PS_NORMAL_E & PS_DOUBLE achieve the same thing: writing two pixels for each sample read. The normal versions run at double clock rate so the resolution of the sample points is higher but that only matters with an asymmetric clock and I don't know of anything other than BBC teletext that has that.  I don't think PS_DOUBLE actually works properly at the moment in integer / non integer scaling as the frame buffer size is derived from the capture area (should work in FB scaling modes)
The above were mainly implemented to workaround problems using low pixel clock frequencies which have now been fixed.

Similarly PS_HALF_O and PS_HALF_E write one pixel for every 2 sample clocks, again increasing the resoloution of the sample position but we have equivalent functionality in hardware now for the only known use case (at least for 3 bit mode).

There is a use case for writing two pixels for each sample and that is to help with interpolation:
A very small frame buffer ends up looking extremely soft when scaled to full screen whereas a pixel doubled buffer (in both directions) looks much better.

Pixel doubling on it's own does have its problems as it reduces the number of integer scaling options (i.e. no odd multiples of the undoubled size) but I have found a workaround for that on the vertical scaling:

Currently you can double vertically by switching on FB double height and I found you can integer scale a double height buffer by 2.5x instead of a single height buffer by 5x and it looks the same because although the odd and even pixels are integer scaled by different amounts (2x and 3x) they are both the same value so it ends up looking like a single pixel scaled by 5x.

So I'm proposing replacing the "Pixel Sampling" option with a "FB double width" option which should have the same effect as PS_NORMAL_O, PS_NORMAL_E, PS_DOUBLE but allow double width buffers with odd multiple scaling and essentially replicate the vertical functionality. I think we can also eliminate the PS_HALF_O and PS_HALF_E  due to their partial replication in hardware.

hoglet67, 2019-04-15:
> So I'm proposing replacing the "Pixel Sampling" option with a "FB double width" option which should have the same effect as PS_NORMAL_O, PS_NORMAL_E, PS_DOUBLE but allow double width buffers with half multiple scaling and essentially replicate the vertical functionality. I think we can also eliminate the PS_HALF_O and PS_HALF_E due to their partial replication in hardware.

This does make sense, but...

The main actual active user of PS_HALF_O/PS_HALF_E is Trixster, for his Atom. The unit he's using with the Atom contains CPLD v2 (plus some extra line termination resistors). I know he doesn't have a CPLD programmer to do the update, so I'm reluctant to make changes that will mean he can't use the latest software.

I know it's more work for us, but could we either:
1. Just remove Geometry/Pixel Sampling if the CPLD version is v6 or later, or
2.. Park working on this item until the next release (say another 3 months) and I'll try to work out a plan to get him upgraded in the mean time

IanSB, 2019-04-15:
Well I could implement the doubling option side of things and just remove PS_NORMAL_O, PS_NORMAL_E, PS_DOUBLE for now, and remove the remaining PS_HALF_O and PS_HALF_E and the Pixel Sampling line itself at a later date.
In order to avoid adding another line to the menu (and increase redraw time) the existing FB double height option could be changed to FB Double Size with options Off / Height / Width / Height & Width

hoglet67, 2019-04-15:
That seems a good plan. I will make contact with Andy and chat to him about the best way of getting him upgraded. What's your feeling about the the current CPLD version (6.4 I think) is now stable? Is there anything else you have in mind that might need to be changed?

IanSB, 2019-04-15:
I haven't thought of anything else for the CPLD yet (!), although I have just sent 3 boards with programmed CPLDs to PC testers so it might be worth waiting for some early feedback from them in case of any significant show-stopping problems. Unfortunately they aren't in the UK so there will be some delay due to international postage.

Looking at the Atom capture loops, the pixels seem to be presented as two adjacent 4 bit pixels with the top 4 bits unused and then the resulting pixels are doubled up. Can that be changed to look like the 6 bit/pixel modes as we could then use the standard capture loops with this new double width option. That would potentially allow things like in band control and NTSC artifacting to work on the Atom in future. (The 6847 was used in the tandy color computer which made use of NTSC artifacting)

hoglet67, 2019-04-15:
> Looking at the Atom capture loops, the pixels seem to be presented as two adjacent 4 bit pixels with the top 4 bits unused and then the resulting pixels are doubled up. Can that be changed to look [...]

Yes, that could easily be done. The only reason it's like this is because it predates the 6-bit capture mode, and I was wondering whether it would make sense to send 3x 4-bit pixels. But were way pase needing to do that now. I'll change this tomorrow.

IanSB, 2019-04-16:
> Yes, that could easily be done.  I'll change this tomorrow.

With the latest commits I just made, double width buffers work so you should be able to create a new Atom profile that works with the standard capture loops (see comments in the pull request about the workaround).

IanSB, 2020-05-06:
This has been addressed by limiting the option to legacy mode only

closed 2020-05-06 by IanSB