hoglet67/RGBtoHDMI #69: Feature: Save changes back to the SD Card

hoglet67, 2019-04-04:
Changes are possible under 4 menus:
- processing
- settings
- geometry
- sampling

The challenge is:
- tracking what has changes
- deciding which level of profile a particular change belongs in

By level of profile, we mean either:
- /Default.txt
- /SomeProfile.txt
or
- /Default.txt
- /SomeProfile/Default.txt
- /SomeProfile/SomeSubProfile.txt

A possible starting point is that all changes should be saved to the most specific level of profile that currently exists. i.e. the default files are manually maintained.

Can we think of any exceptions where this is clearly wrong?

IanSB, 2019-04-05:
I've been thinking about and there are issues with all the different approaches.
One of the primary reasons for the default folder is to make maintenance of the default profile set easier as in many cases only the default will need to be updated and I agree that /Default.txt should be manually maintained with any changes to a standard profile saved to /SomeProfile.txt

The problem comes with the switching sets: 
In all cases, geometry and sampling changes should be saved to /SomeProfile/SomeSubProfile.txt
The problem happens with where to save other settings. In order to allow temporary changes to another setting like scanlines,  /Default.txt and then /SomeProfile/Default.txt are only read once when the switching profile is loaded and as each  /SomeProfile/SomeSubProfile.txt is auto selected (which normally have just geometry and sampling) it allows the other settings to persist across the auto-select.

If we save those other settings to /SomeProfile/SomeSubProfile.txt then each time a sub-profile with those settings is auto-selected you will lose the persistence of the temporary change and in the case of the auto-select on/off setting that would make that option virtually unusable.

To work around this, the other (non-geometry/sampling) options should be saved to /SomeProfile/Default.txt so they are common to all sub-profiles.

There are some circumstances where it would be preferable for one sub-profile to have a different setting to the other sub-profile and I think for the moment that will have to be managed manually especially because if you make a custom change to one sub-profile you have to put the opposite setting in all the other sub-profiles to make sure the setting is undone when they are auto selected.
(This already is being used to switch genlock off for the 70Hz "CGA/EGA on VGA"  modes) 

Are you going to look at the saving code or do you want me to do it?

hoglet67, 2019-04-05:
> Are you going to look at the saving code or do you want me to do it?

It's about third on my list things to work on at the moment, so if you want to have a go then feel free.

IanSB, 2019-04-07:
I think this can be closed now as outstanding issues are covered by 
https://github.com/hoglet67/RGBtoHDMI/issues/74
https://github.com/hoglet67/RGBtoHDMI/issues/75

closed 2019-04-07 by hoglet67