hoglet67/RGBtoHDMI #72: Bug: System crash when using 1280x720@50 resolution/BBC Micro profile

hoglet67, 2019-04-07:


IanSB, 2019-04-07:
Yes a lot of the lower resolutions crash

hoglet67, 2019-04-07:
Any theories as to why?

IanSB, 2019-04-07:
Nothing definite, but it seems to be related to the setup and clearing of the frame buffer because that ends up with vertical line garbage but when it gets as far as capturing a frame the capture area looks the right size and clean. Maybe related to https://github.com/hoglet67/RGBtoHDMI/issues/60

hoglet67, 2019-04-07:
The bug is down to the Pi Firmware limiting the max (virtual) framebuffer height to 2560.

I added some debugging:
```
INFO: Initialised Framebuffer
INFO: Size: 1112x720 
INFO: Virtual Size: 1112x2560 
INFO: Pitch: 560 bytes
INFO: Framebuffer address: DF89D000
```
The virtual height should be 720x4 = 2880.

If I then add the max_framebuffer_height property to 8192 then that allows larger values:
```
INFO: Initialised Framebuffer
INFO: Size: 1112x720 
INFO: Virtual Size: 1112x2880 
INFO: Pitch: 560 bytes
INFO: Framebuffer address: DF872000
```
And the crash doesn't happen.

I'll push this fix now.

commit 333ad78a (2019-04-07): Increased Pi Config max_framebuffer_size; fixes #72 and maybe also #60

Change-Id: I8df9438d21406f7af2af692b51b0f6ebfa1917e8
closed 2019-04-07 by hoglet67