hoglet67/RGBtoHDMI #73: Improvement: Use sub-profile switching sets for Beeb default/mode7 modes

hoglet67, 2019-04-07:
This allows us to get rid of the sampling7 and geometry7 params, which were always a bit of a bodge.

IanSB, 2019-04-07:
Mode switching shouldn't be any slower as all sub-profiles are pre-loaded (as raw txt) rather than read from the sd card during a mode switch. There is a tiny delay while the txt is parsed but that is insignificant.
The way I see this working would be as follows:
The settings would have a BBC Mode 7 true/false switch which just informs the system that the current profile is a mode 7 profile.
Auto switching would then have something like the following options:
Off / Mode7 Timing / Standard Timing
So when set to  'Mode 7 Timing' it will select a profile with BBC Mode 7 set to True when mode 7 timing detected and any other profile when mode7 not detected
When set to 'Standard Timing' the sub-profiles work as now.

IanSB, 2019-04-07:
Should the BBC Mode 7 true/false actually be an option in the sync setting in geometry as its sync related.
i.e. SYNC =+H+V/ -H+V/ +H-V/ -H-V/ Composite/ BBC Mode7

hoglet67, 2019-04-07:
> Should the BBC Mode 7 true/false actually be an option in the sync setting in geometry as its sync related.

Not sure, it's both are legitimate composite sync signals, so it does feel like we would be slightly overloading this setting.

But if it works out easier to do this way, then I don't really have an objection.

IanSB, 2019-04-08:
This will require more than one "mode0-6" profile because the timing for TV0,0 and TV0,1 is going to be different (the existing interlace on/off detection code can be removed).