hoglet67/RGBtoHDMI #76: Improvement: Make return and keymap properties configurable through UI

hoglet67, 2019-04-07:


IanSB, 2019-04-07:
Probably best if a reboot is required for the changes to take effect so you don't end up in an inoperable state part way through changing things. Also maybe the two button reset should restore a default state.
Could be part of the same menu that changes the config for https://github.com/hoglet67/RGBtoHDMI/issues/66

hoglet67, 2019-04-27:
I've just added the "Return Position" parameter to the OSD.

I don't think adding the keymap and actionmap parameters is worth the effort. They are multi-character strings, so would need to be broken out into 11 separate properties. They are also too many corner cases to allow live edit (as you noted Ian).

I'm going to close this issue as done for now.

closed 2019-04-27 by hoglet67