hoglet67/RGBtoHDMI #77: Palettes

IanSB, 2019-04-07:
Should palettes be driven by txt files like modes and profiles or at least have the option for additional custom palettes in this way?

hoglet67, 2019-04-07:
Yes, I think that now we can read files that makes sense.

Do you think we should allow profile specific palettes, or just have a single global folder?

IanSB, 2019-04-07:
I think initially a single folder as many profiles will be sharing the same palette.
e.g with the PC profiles I already have 3 different switching sets that all have different CGA / EGA etc timings but all share the same palettes. The BBC ones share with the Electron & Oric and maybe others.

hoglet67, 2019-04-24:
I'm also wondering if there is a case for at least some of the palettes to be provided by the CPLD driver.

(I would have said profile, but as we are now pro grammatically generating some of the Atom palette that becomes harder). 

We have a set of Atom palettes which only work correctly with the Atom CPLD based interface.

If using the Normal CPLD based interface with an Atom, then
- in 3bpp sampling, orange is mapped to yellow, and the default RGB palette is probably best
- in 6bpp sampling, orange can be see, but only with the ATOM_EXPERIMENTAL palette which is no longer present (I think I deleted this in error a few days ago)

IanSB, 2020-05-06:
This has been added in the latest release
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200504_d19df1c

closed 2020-05-06 by IanSB