hoglet67/RGBtoHDMI #78: Investigate high HDMI clock rates

IanSB, 2019-04-07:
According to info online the HDMI clock can run mach faster than 165Mhz to allow for higher resolutions. I tried using that to generate 70Hz 1080p to a HDMI 2.0 TV but it didn't work so maybe other registers need to be changed or different divisors used.
Some support for using >165Mhz has been added but is currently disabled
see here:
https://www.raspberrypi.org/forums/viewtopic.php?t=172057
which mentions hdmi_pixel_freq_limit

hoglet67, 2019-04-07:
I think the 165MHz comes from HDMI 1.0 (and DVI). The next version, HDMI 1.3, allows for up to 340MHz pixel clocks.

Do you have any displays that officially support HDMI 1.3. It would need to be WQXGA (2560×1600) to required that sort of resolution.

The highest DVI pixel clock supported by my largest display (a HP LP2465) is 162MHz, so I wouldn't be able to test this.

IanSB, 2019-04-07:
Yes I have a WQXGA (2560×1600) display and a 4K Tv (tried on 4K tv) which has HDMI 2.0 I think.

IanSB, 2020-09-08:
Addressed in:

https://github.com/hoglet67/RGBtoHDMI/pull/156/commits/04b0397c8f495e57b514f08e4761ad7536a3eb56

closed 2020-09-08 by IanSB