hoglet67/RGBtoHDMI #81: Feature: Auto delay adjust for 1280x1024 monitors

IanSB, 2019-04-07:
You can still buy 1280x1024 monitors (unlike 1600x1200) and they seem to be quite common. They are closer to 4:3 than current widescreen monitors and should be ideal for the beeb as integer scaling from 640x256 is straightforward. However each mode on the BBC has a different horizontal position so you either have to capture more than 640 pixels which eliminates simple integer scaling or manually adjust the delay after each mode change. Maybe some sort of automatic adjustment could be implemented. This issue also affects PCs where the text and graphics versions of the same resolution have different offsets due to the character ROM delay. I have seen this sort of auto adjustment on LCD monitors following a mode change when the image moves left and right by a few pixels but the main issue with the BBC is detecting the mode change as there isn't always going to be a sync disturbance. It might have to be an auto adjust triggered by one of the configurable buttons.

hoglet67, 2019-04-07:
Wouldn't this require some sort of a calibration screen that ensured the first column of pixels we being used?

On a modern PC it's less of an issue, because the background is rarely black.

IanSB, 2019-04-07:
> Wouldn't this require some sort of a calibration screen 

Potentially but most non-text output on the beeb is games and quite a lot have non-black backgrounds. Worst case it's a completely manual adjustment using the configurable buttons.

IanSB, 2019-04-07:
Maybe continuously capture 8 or 16 pixels on each side of the active 640 pixels and as soon as any non-zero pixel is detected in those areas, adjust the delay so that those pixels are in the visible area. Might be visually annoying when it shifts but it shouldn't happen that often.