hoglet67/RGBtoHDMI #82: Improvement: Add protection to sync and rgb inputs

IanSB, 2019-04-07:
I think I just damaged a CPLD by plugging the 'hat' PCB back into an unpowered Pi while being connected to the computer RGB & sync source. (This was on a PC so Pi powered via 5v supply)

I have done this multiple times but after doing it this time, the Vertical sync was very jumpy, mostly when pressing the buttons but occasionally when it was just sitting there.
The source had +ve Hsync (mostly low) and -ve Vsync which would be mostly high so maybe allowed enough current to flow to cause issues.

I tried reprogramming the CPLD but that didn't help so I replaced it and so far that appears to have fixed it so we definitely need some protection on the inputs. Any thoughts on the best approach for this? Diodes or something else?

hoglet67, 2019-04-08:
In theory, it's possible to damage a XC9500XL CPLD input by feeding a 5V signal into it when the device is unpowered. There's a bit written about this on page 12 of the datasheet. So this might be unrelated to the act of plugging this back into the Pi.  With inputs > 3.5V you do need to be careful about power supply sequencing.

The only occasion I've damaged a RGBtoHDMI setup was a different case. The hat and the Pi were initially separate. The hat was connected to a switched Beeb. The Pi was connected to an LCD monitor that was in stand-by. All I did was plug the two parts together. Both were unpowered at the time. This ended up blowing the a GPIO input on the Pi, likely the pin that first made contact.
Because my monitor is un-earthed and the Beeb is earthed, it's possible for the monitor to float at quite a high voltage wrt ground. This voltage, I think, ended up being discharged via  the Pi's GPIO protection diode, which ended up causing damage.

I'm now very careful to disconnect the HDMI cable before plugging a Pi and a Hat together.

Your case may well be CPLD damage, but did you check by trying another Pi?

hoglet67, 2019-04-08:
I had a couple more thoughts about this.

**1. Driving an unpowered CPLD**

An unpowered XC9500XL series part should be able to handle up to 4V on it's inputs. Here's the reference:
https://www.xilinx.com/support/documentation/user_guides/ug445.pdf#page=17

It would be interesting to measure your VSYNC signal to see if it actually gets to much more than 4V. It really depends on the driver technology, but a 74LS TTL driver cannot drive much above this anyway, as there is a diode in the output path.

You say the issue seemed intermittent. In my experience damaged I/Os are usually permanently stuck, so I wonder if there is another cause...

**2. Intermittent connection between the Pi and the Hat**

It's interesting that you suggested the problem was intermittent was associated with pressing the buttons. I've had a problem exactly like this, and it turned out to be a poor/intermittent connection between the Hat and the Pi. In my case, the dodgy connection was on one or two of the pixel bus signals, and this would manifest as vertical lines on the screen that came and went as you flexed the boards.

Do you use spacer between the hat and the Pi?

Without spacers, I've found that over time (months) the slight flexing of the connector when you press the buttons causes damage to the socket on the hat. And after a while, you start to get bad connections.

The 0.1" female headers are not really designed for repeated insertions, especially cheap, unbranded parts. The harwin ones from RS are only rates for about 50 cycles. I'm sure I've exceeded this by now on my first board (which is now slightly dodgy) if not screwed together.

What make of headers did you end up using?

Bottom line, I actually suspect that you didn't have a problem with the CPLD after all. If it's the connector that is dodgy, then unplugging an re-plugging may help for a while, but eventually the problem will recur, especially if movement is possible in the connector.

IanSB, 2019-04-08:
> Your case may well be CPLD damage, but did you check by trying another Pi?

Yes I tried another Pi - no difference.
> What make of headers did you end up using?

The Harwin ones in your BOM. No spacers but I have some now. 
I reseated the hat several times but it made no difference. I don't think I've cycled the connector 50 times yet.

The hat was connected to a powered PC and plugged into an unpowered Pi but that was connected to a monitor so it was earthed and the PC was earthed as well. 

I tried moving / reseating the board again with the new CPLD but it remained completely stable.
The 'bad' CPLD came off physically undamaged so maybe I'll fit it to another board sometime to see if it still has the problem.

IanSB, 2020-05-06:
This has been addressed by the addition of the Extender board layouts.

closed 2020-05-06 by IanSB