hoglet67/RGBtoHDMI #84: Bug: Saved Profiles use DOS line endings

hoglet67, 2019-04-10:
It seems that Saved Profiles use DOS line endings, and our original profiles use windows line endings.

```
dmb@quadhog:/media/dmb/67FC-977D/Saved_Profiles$ find . -name '*.txt' | xargs file
./BBC_Micro.txt: ASCII text, with CRLF line terminators
```

Just in case is causes obscure bugs in the future, we should pick one and stick to it

```
dmb@quadhog:/media/dmb/67FC-977D/Profiles$ find . -name '*.txt' | xargs file
./Atom.txt:                         ASCII text
./Atom_CPLD.txt:                    ASCII text
./BBC_Micro.txt:                    ASCII text
./BBC_Micro_24MHz.txt:              ASCII text
./Custom_1.txt:                     ASCII text
./Custom_2.txt:                     ASCII text
./Default.txt:                      ASCII text
./Electron.txt:                     ASCII text
./Oric.txt:                         ASCII text
./UK_101.txt:                       ASCII text
./ZX80-81.txt:                      ASCII text
./OAK_OTIVGA/CGA_Graphics.txt:      ASCII text
./OAK_OTIVGA/CGA_Text.txt:          ASCII text
./OAK_OTIVGA/Default.txt:           ASCII text
./OAK_OTIVGA/EGA.txt:               ASCII text
./OAK_OTIVGA/MDA.txt:               ASCII text
./PC/CGA.txt:                       ASCII text
./PC/Default.txt:                   ASCII text
./PC/EGA.txt:                       ASCII text
./PC/Hercules.txt:                  ASCII text
./PC/MDA.txt:                       ASCII text
./TSENG_ET3000AX/CGA.txt:           ASCII text
./TSENG_ET3000AX/CGA_Keen.txt:      ASCII text
./TSENG_ET3000AX/CGA_on_VGA.txt:    ASCII text
./TSENG_ET3000AX/Default.txt:       ASCII text
./TSENG_ET3000AX/EGA.txt:           ASCII text
./TSENG_ET3000AX/EGA_on_VGA_25.txt: ASCII text
./TSENG_ET3000AX/EGA_on_VGA_28.txt: ASCII text
./TSENG_ET3000AX/VGA_Graphics.txt:  ASCII text
./TSENG_ET3000AX/VGA_Text.txt:      ASCII text
```

IanSB, 2019-04-10:
That's a feature :)
I chose DOS endings because the ones most likely to be creating, viewing and messing with profiles are likely to be the PC users.

hoglet67, 2019-04-10:
I see from the comment in the pull request that you intent to re-fresh the released profiles anyway.