hoglet67/RGBtoHDMI #87: Review contents of processing and settings menus

IanSB, 2019-04-13:
I think we need to determine what each of these two menus is for and review what items belong in each one.
I think the processing menu should contain options that affect the current profile and that the user might want to change to customise their preferences and the settings menu should contain elements that are generally only changed when switching profiles or rarely changed for testing etc.
e.g should palette move to settings because it only changes for specific profiles while palette control remains in processing?
Similarly should scaling move to processing as it is a user preference
Should the menus be renamed?

hoglet67, 2019-04-13:
The split I had originally in mind was was:
- Processing - configuration related to additional processing of the actual video content in some way
- Settings - configuration related to how the system operates generally

So with this definition, scaling probably should go into the processing menu.

By all means lets talk about changing this though.

But it has to make sense as far as the user is concerned. From your description above I'm struggling with that aspect of it. We should be making this sort of change to improve the user experience, making it easier for them to find a feature they are looking for,

Why not sketch out what would they would be renamed to, and what would be in each?

IanSB, 2019-04-13:
I was thinking along the lines of:
Preferences: Things a user would want to adjust when within a profile)
Settings: Things that have to be set to make the profile work and if you change them they will mess up the profile:
So:
Preferences:
Palette control: (enabling/disabling in band control and NTSC artifacting)
Scanlines:
Scanline Level:
Scaling:
Mode 7 deinterlace

Settings:
Palette: If you change this you get wrong colours
Border Colour: (new feature I just added) primarily for zx80 white background
Vsync Type: changing gets jumpy display (electron)
Vsync indicator: only used for debugging really
Vlock mode: should always be locked really (if in & out freqs don't match it is inhibited anyway)
Vlock line: should be left at 10, adjusting only makes lag worse
Vlock Adjust: Changing makes wildly out of spect timings
Num Buffers: really should be left at single now that OSD and vlock issues sorted.
Debug: again only needed for testing

You might want to change the palette to get a monochrome, green or amber screen effect but that would require different palette entries for each of RGB, RGBI, RrGgBb, Atom etc so I think a better option there would be a new colour option in the preferences menu to allow the current palette to be converted to monochrome / green etc programmatically and that would also work with user palette files.

That leaves Input Mux which really feels like it should be in the Sampling menu as it is CPLD specific and associated with the 3/6 bit switch. It certainly would be in that menu if it was controlled by an internal bit rather than a GPIO.

There are also Output resolution and Interpolation which would also belong in the preferences menu if / when they are adjustable on a per profile basis.

hoglet67, 2019-04-27:
I would suggest putting Palette and Border Colour in Preferences (together with Output Colour and Output Invert). It makes more sense, I think, to have all the colour related things together. As it currently is, I continually find myself looking in the wrong place for Palette.

I also agree about moving Input Mux to the CPLD Menu. I've just done that (164b005). I'll do the other menu tweaks now.

hoglet67, 2019-04-27:
I've updated the Preferences  and Settings menus (e817994).

Do say if you really don't like the palette being under preferences, but I do think this is ultimately a user preference (e.g. the choice between several valid palettes for the Atom).

closed 2019-04-27 by hoglet67