hoglet67/RGBtoHDMI #90: Update cpld_atom to use the standard capture line code

hoglet67, 2019-04-16:
1. Update Atom CPLD to pack into 6-bit wide pixels

Done but not pushed yet.

2. Update cpld_atom  to use:
- capture_line_normal_4bpp_table
- capture_line_normal_8bpp_table

Done but not pushed yet.

3. Resolve palette issue

The colour map used by the Atom CPLD is:
0x00 - Black
0x01 - Red
0x02 - Green
0x03 - Yellow
0x04 - Blue
0x05 - Magenta
0x06 - Cyan
0x07 - White
0x08 - Orange
0x09 - Bright Orange
0x0A - Dark Green (Text Background)
0x0B - Dark Orange (Text Background)

In 8-bpp mode, the palette is configured as per the above

In 4-bpp mode, capture_line_atom_4bpp did the following transformation on the fly:
- 0x08,0x09 => 0x03 (yellow)
- 0x0A,0x0B => 0x00 (black)

This was so only colours 0-7 were actually used, leaving bit-3 in the frame buffer free for the menus.

We need to think of a way to achieve the same net effect with the standard code.

Options I can think of:

1. In the Atom CPLD, change the color encoding to use bits 4 and 5, i.e.:
- 0x13 = Orange
- 0x23 = Bright Orange
- 0x10 = Dark Green
- 0x20 = Dark Orange

Now, simply ignoring the higher bits, the 4bpp capture code will get the right result.

Problem is, I don't think this will fit in the CPLD.

2. In the Atom CPLD, have a pin (or scan bit) to select between 3 and 6 bit capture.

Might fit, not tried.

3. Keep the custom capture code for 4bpp mode (yuk!)

4. Drop 4bpp capture mode for the Atom CPLD

We would need to add a way for the CPLD driver to indicate to the higher levels that only 8 BPP frame buffers are compatible.

Ian, any thoughts on this?

IanSB, 2019-04-16:
If you can get option 1 working then by all means do it but otherwise I think option 4 is sensible. There isn't any real need to operate in 4bpp mode as it requires 8bpp for correct operation. Same as all the PC stuff really, if I switch to 4bpp mode in 6 bit capture I get all wrong colours. They might be slightly less wrong than the Atom but wrong none the less. Setting the bit depth correctly is just part of the configuration, just like selecting the palette in the first place.

There isn't much left that uses 4bpp mode anyway, just mode 7 (which won't work in 8bpp mode at all so is similar to the Atom but the other way around). Even modes0-6 uses 8bpp by default now to support the 127 colour in band mode.

I wonder if I could use 8bpp for mode 7 and use the extra bits as the comparison buffer (their state has no effect on the ouput as the palette is programmed repeating the same 8 colours).

hoglet67, 2019-04-16:
I've managed to do option (1)

1a3b861b and ef8f4a75

It works very well in both 4BPP and 4BPP modes.

I'll push the updated profile tomorrow.

IanSB, 2019-04-21:
You might want to switch on double height and width for the Atom profiles.
Also are the Atom palettes technically correct?
The 6847 only outputs 4 luminance levels but the standard saturated 8 colour RGB part of the Atom palette converts to 8 separate luminance levels.

hoglet67, 2019-04-21:
Yes, I'll switch on double height/width.

The original Acorn Atom Colour Card output TTL-level RGB, which is why I'm using 100% R/G/B. None of them directly applied a analog YUV->analog RGB transformation.

There is a separate Atom Monochrome palette, which accurately reflects what you see if you just hook up the Y output to a monochrome monitor. i.e. just 4 levels.

It might be possible to calculate a colour palette such that when fed through the monochrome palette filter, results in the right luminescence levels. I have no idea what that would do to the colour levels thought.

IanSB, 2019-04-21:
> The original Acorn Atom Colour Card output TTL-level RGB

The 50Hz Mk II colour card certainly did a conversion to TTL RGB but the original colour card was just a PAL encoder that produced 60Hz PAL and worked with very few colour TVs.
See: https://stardot.org.uk/forums/viewtopic.php?t=11321
I don't know if that did a similar conversion.
You might find a 6847 palette in a tandy color computer emulator or maybe a dragon 32 emulator.

hoglet67, 2019-04-21:
Interesting that you found the "Acorn PAL Encoder" design - I had completely forgotten about it.

I've actually never seen one, so I don't really know what the colours look like in practice.

For reference, the schematic is here. 
http://stardot.org.uk/mirrors/www.bbcdocs.com/filebase/hardware/circuit/atom_colour_encoder_cct.png

It uses 2x LM1496 balanced modulators. The Dragon 32 uses the LM1889.

I've just added two more Atom Palettes:
- One taken directly from XRoar
- The other made up by myself, such that the Y value is correct, but still using saturated colours.

Here's the XRoar one:
[image]

IanSB, 2019-04-23:
I had a look at this myself and I don't think you can ever get a perfect palette that both looks good and is a true representation of the intended colours at the same time.
You either end up with something that looks good but isn't reversible back to YUV due to clipping (like the Xroar one) or is reversible (to 4 level mono) but has poor contrast or poor saturation (like the Acorn PAL one).

The problem is that the maximum Y level output of the 6847 cannot be treated as 100% because you only get 100% Y with zero U and V and it uses 100% both for white and some bright colours which is why it's referred to as "Buff" and not white.
I created a YUV to RGB algorithm that converts the actual voltage levels from the 6847 YUV outputs as described in the datasheet to 0-255 RGB and that produces exactly the XROAR values when luma and chroma are set to 100% but the values have to be clipped as they go out of range which means they can't be reversed back to YUV.

However I may have come up with a compromise which gives reasonable brightness and saturation while still giving a good representation of the actual 6847 colours and it's reversible back to 4 levels of Y.

I've attached an experimental version of OSD.C & OSD.H which uses the above algorithm and allows adjustment of the luminance percentage by using the genlock line value (temporarily renamed as luminance%) while automatically desaturating each colour individually until the RGB values are reversible back to YUV and thus 4 levels.

[osd.zip](https://github.com/hoglet67/RGBtoHDMI/files/3105390/osd.zip)

Build using these files, call up the settings menu, change the palette to "Atom desaturated" and then adjust the LUMINANCE% value until you get a good looking result (around 80-85% seems a good compromise). Too high and colours like yellow get desaturated to white and too low and the image is too dim.

The calculated values are output in the serial log so once you select a suitable luminance value the output RGB values can be used in an updated OSD.C
With a luminance value of around 80-85 I think this looks better than Atom Pal and maybe Xroar and it's reversible to monochrome. It does tend to push the green and cyan close together but that might be improved by further tweaking of the saturation of those two colours. 

Note due to rounding errors the Y value can be off by 1 on some conversions so the final RGB values may need some manual tweaking to get consistent Y results as even an off by 1 shows up when you switch the output colour to mono/green etc.

hoglet67, 2019-04-23:
That's a nice piece of work there Ian. We should try to keep this code in place, but with a hard coded luminance value of 80%.

One thing I would change though. I think the dark green/dark orange text background are way too bright. Looking at the code, you have them at the same luminance value as red/blue (i.e. 650mV). In practice, I they are actually black (720mV).

If you hook up just the composite video output of the Atom (pin 9 of PL4) to a suitable TV, you can verify there is no difference between the text background and true black (e.g. the border). I think the data sheet also confirms this. I'm pretty sure they only become visible because they have different UV values.

Out of interest, I've just dug out my old MK3.5 Atom Colour Card from Phill, schematic attached.
[AtomColour.pdf](https://github.com/hoglet67/RGBtoHDMI/files/3107192/AtomColour.pdf)
[LMH1251.pdf](https://github.com/hoglet67/RGBtoHDMI/files/3107193/LMH1251.pdf)

This looks a bit like your Atom desaturated palette, but not really the same. I think this is because the LMH1251 is expecting Rec. 601 U/V signals, which the Atom PA and PB are not. 

I think the answer, ultimately, is that there is no right answer. i.e. it's all quite subjective, and which you prefer is probably influenced by what you are used to.

Dave

hoglet67, 2019-04-23:
If you set the Y value of dark green / dark orange to 720mV (same as block), but then change yuv2rgb
```
        double y = (double) (luma_scale * 255 * (770 - y1_millivolts) / (770 - 420))/100;
```
then it looks quite good (completely subjective!)

The reason I picked 770 is that's what the 6847 outputs during the back porch, which I think is then used by most systems as the black level reference.

You do have to reduce the brightness on the monitor though to compensate, so black is black, not grey.

But then you see the text background as a very dark green/orange. But if you then turn the color down to 0 in the TV settings, they become indistinguishable from black.

IanSB, 2019-04-23:
> I think the answer, ultimately, is that there is no right answer. i.e. it's all quite subjective, and which you prefer is probably influenced by what you are used to.

Yes, but at least this version produces correct monochrome levels which must mean it's somewhat close.
Perhaps we can get Fwapp from the stardot post linked above to post some screenshots of his pal encoder output (he's building one of these converters for a ZX81 so might be interested in helping)

Here's another test version with some more tweaks:
1. Changes suggested above
2. Converted to integer arithmetic to avoid some annoying rounding errors
3. restrict rgb range to 2-253 (also helps with rounding errors)
4. Add separate U and V scaling using the vsync indicator to switch it on and off:
Valid YUV signals have different max ranges for U and V (+-0.436 for U and +-0.615 for V)
The old code scaled both the same but you can now switch on the above using the vsync indicator to get unequal U & V scaling. This means that some colours can have a higher saturation without going out of range and in this case switching it on boosts magenta, cyan & green.

Have a try and see which one you prefer.
Note you selected 80 as the optimim value but that produces "off by 1" errors for the luminance.
With Unequal U & V off, 81 produces matching luminance values
With Unequal U & V on, 79 produces matching luminance values

Can you try both and let me know which one you prefer.
Also please post some screen grabs from the atom with those palettes as I've been testing with a BBC so only looking at the first 8 colours.
[osd-test-v3.zip](https://github.com/hoglet67/RGBtoHDMI/files/3108572/osd-test-v3.zip)

hoglet67, 2019-04-23:
Here's some snaps from osd-test-v3.zip:

With Unequal U & V off, 81:
[image]

With Unequal U & V on, 79:
[image]

Now, after quite a bit more fiddling, I think it's probably a mistake to have set the black reference level to 770. Main reason being black is very much grey, and I have to drop the brightness on my TV/Monitor from 50% to 20% to get it looking "right" again. Doing this only for the Atom is going to be annoying.

Instead, if we want to see a slightly difference colour for the text background, then I think I prefer upping the Y level from 720 to 710 or 700, for just the two text background colours. But this is technically incorrect, so probably we should include both options (like I did with normal and extended).

Here's two more snaps, with black reference back at 720, and 710 for the two background colours,

With Unequal U & V off, 81:
[image]

With Unequal U & V on, 79:
[image]

Here's two more snaps, with black reference back at 720, and 700 for the two background colours,

With Unequal U & V off, 81:
[image]

With Unequal U & V on, 79:
[image]

Also, I'm starting to wonder if we are doing something wrong with gamma. There's quite a difference between what you see on the screen (with a normal brightness level of 50%) and what you see in the PNG on another monitor. The biggest difference is perceived black level in the PNG is much lower. But then again, the black level on my TV changes massively with vertical viewing angle anyway.

Hope this helps, but I suspect it won't!

hoglet67, 2019-04-23:
When it comes down to it, I don't think I would ever choose to use a palette where the text background colour was anything other than black (same as the border).

I think this is just because I'm not used to seeing a difference between the border and the text background:
[image]
It's actually much more obvious than this when viewed on my TV, hence the earlier comment about gamma.

But I do very much like the rest of the colours, including buff rather than white, which is how it's described in the data sheet.

So, for me the best is probably a black reference level of 720, Y levels of 720 for all three of the "black" colours, a luminance of around 80%, and probably Unequal U & V off.

IanSB, 2019-04-23:
> probably Unequal U & V off.

Yes it looks like it's pushing the orange towards pink but it was worth a try (That option was trying to completely fill the YUV colourspace without going overrange)

> Also, I'm starting to wonder if we are doing something wrong with gamma.

Gamma is really under the control of the display device so we shouldn't be changing it.

>  I think it's probably a mistake to have set the black reference level to 770

Technically it's the right thing to do. NTSC has different levels for blanking and black so US TVs were adjusted to compensate for the offset. PAL has the same level for blanking and black so any NTSC source will always have a slightly grey background when displayed on PAL devices.

Sounds like we need maybe four Atom palettes:
Atom Mk1 (with black backgrounds on text)
Atom Mk1 full (with pedestal and dark green/orange backgrounds on text)
Atom Mk2 (with TTL RGB + orange)
Atom Mk2 full (with TTL RGB + orange + dark green/orange backgrounds on text)

There isn't any need for an Atom Mono as that is now covered by selecting mono on the Mk1 colour card settings. I don't think there is a need for an 8 colour Mk2 colour card palette as that can be created either by using the standard RGB palette which repeats every 8 colours or switch to 4 bpp mode with one of the Mk2 colour card settings.

I'll merge the yuv2rgb code back into the original osd.c

hoglet67, 2019-04-23:
I'd like to keep two additional ones:
- Atom Acorn - this is pure TTL RGB, with orange mapped to read
- Atom XROAR - because this is a useful data point for what the Dragon people think the 6847 looks like

IanSB, 2019-04-23:
> Atom Acorn - this is pure TTL RGB, with orange mapped to red. This is what the Acorn 202,006 card looks like.

I think you get that if you select the standard RGB palette because of the way you mapped the colours on the six bit capture side (the RGB entries repeat every 8 colours)

hoglet67, 2019-04-23:
Yes, I think you are right,

IanSB, 2019-04-23:
Actually just checked and it could be done but the two oranges are currently mapped to 0x0B and 0x13 which both map to colour 3 which is yellow. This also breaks the 4 bpp mode as well I think.
To map to red they should map to 0x09 and 0x11.

The dark green / orange map to 0x08 and 0x10 which will correctly map to black with the RGB palette and also in 4bpp mode.
Can this be fixed in the CPLD?

hoglet67, 2019-04-23:
Ah yes, that was a deliberate choice that I would rather not change.

Mapping of orange->yellow is (subjectively of course) much better (because it's a much brighter colour, so the text is clearer). I could happily use that mode for long periods, which is not the case when mapping orange->red.

Both "Atom Acorn" and "Atom XROAR" are mainly for reference use. I would not expect anyone to choose to use them. So I guess both of these need to stay.

Dave

hoglet67, 2019-04-24:
I think this issue is now all done.

closed 2019-04-24 by hoglet67