hoglet67/RGBtoHDMI #91: Improvement: Allow CPLD driver to provide a default profile

hoglet67, 2019-04-17:
I regularly switch one system between a Beeb and an Atom.

When switching from Beeb to Atom, it's actually quite hard to change the profile from BBC_Micro ro Atom_CPLD, because the menu is very flickery, and barely visible.

I think this is just because the Atom is running at 60Hz, and the Beeb profile is 50Hz.

Anyway, it would be nice if the CPLD driver could provide a default profile, that would be used as a fallback if profile.txt was missing.

I quickly tried this, and it doesn't work quite as expected, because file_read_profile() is called in a way that causes profile.txt to get written (even if there is no change).

(I think as a general principle we should try to minimise unnecessary writes to the SD card, as you are more likely to end up with corruption if the use powers off the system at the wrong time).

Anyway, it would be good to only write profile.txt when the user actually changes the profile.

IanSB, 2019-04-24:
> When switching from Beeb to Atom, it's actually quite hard to change the profile from BBC_Micro to Atom_CPLD, because the menu is very flickery and barely visible.

I think this is because of the mode 7 auto switching code which works even when the menu is on screen. 
For the PC style mode changes, I disabled auto switching when the menu is on screen to avoid loss of menu but it isn't disabled for BBC auto switching so when you feed Atom signals into a BBC profile it gets confused and starts switching between mode 7 and modes 0-6.
I think that the code that handles switching needs to be improved so it doesn't lose the menu on a false trigger. This is probably the last preparatory change required before implementing full BBC profiles.

IanSB, 2019-04-24:
Thinking about it, what's needed for the BBC side is to improve the robustness of the mode7 / mode 06 detection when fed non-bbc signals or if that's not possible, reducing the number of times it exits rgbtofb with a detection (e.g. delaying mode 7 detection for a few seconds after entering rgbtofb would at least reduce the period of menu flashing to once every few seconds.

hoglet67, 2019-04-25:
I've investigated this issue some more.

What's interesting is that it was specific to the Atom CPLD, and didn't occur when a Atom was connected via the Beeb interface (using the Normal CPLD).

What's happening is:
- Beeb profile accidentally selected, with autoswitch=mode7
- Atom timing looks like mode 7, so mode7 = 1
- If mode=7, the OSD is written just once (i.e. the mode 7 line capture code preserves the OSD bit)
- The Atom CPLD uses 6bpp sampling, but we don't have mode7 6bpp line capture code
- So instead, the default line capture code is used
- Which then clobbers the OSD

I've just committed a quick work around which forces autoswitch to off if the Atom CPLD is detected.

I was wondering if the CPLD should advertise whether autoswitch should be supported (e.g. by renaming cpld->old_firmware() to cpld->get_capabilities(), and defining some capabilities like
- SUPPORTS_CONTINUOUS_PSYNC
- SUPPORTS_PSYNC_DELAY
- SUPPORTS_AUTOSWITCH
I held off, because getting the right set of capabilities is important, otherwise we'll only make the problem worse.

IanSB, 2019-04-25:
Yes, I think cpld->get_capabilities() would be useful for supporting multiple CPLD variations.

IanSB, 2019-04-26:
Rather than disable autoswitching with the Atom CPLD, perhaps disabling mode 7 switching would be preferable as there is a possible use case for normal auto switching with the Dragon 32. If you connect the Atom CPLD board to a Dragon 6847 it will work at 50Hz so both 50 and 60 hz profiles could be put into an autoswitching set.

hoglet67, 2019-04-27:
That's a sensible suggestion, which I've just implemented (e5e5771)

hoglet67, 2019-05-06:
Done...

closed 2019-05-06 by hoglet67