hoglet67/RGBtoHDMI #96: Profiles for old CPLD

IanSB, 2019-04-21:
At the moment, the profiles don't work properly on the old CPLD due to the change of sync edge which means the horizontal capture offset is incorrect. We could either use separate profile sets for the old CPLD or auto adjust the value from the standard profile setting if an old CPLD is detected but that will only work for BBC modes as the adjustment will be different depending on the pixel clock speed.

hoglet67, 2019-04-25:
It's not just the profiles that are incorrect.

I've just tested the Atom with Normal CPLD v1.0 and it seems there is insufficient range in H Offset. Even set to 0, sampling starts too late:
[image]
I don't understand why this would be the case.

As a sanity check, I reverted back to the last released version of Master, and the Atom with Normal CPLD v1.0 works fine.

I don't get the same issue with CPLD v1.0 on the Beeb. i.e. if you set H_OFFSET to 0, then things are pretty much centered.

hoglet67, 2019-04-25:
I figured out what was going on here.

For CPLD v1 to work with the Atom, you need a profile that runs at 2x the pixel clock, and then use Odd/Even sampling to discard the extra samples.

Here's the working profile
```
sampling=0,0,0,0,0,0,0,0,8
geometry=16,11,304,240,304,240,3,8,14318181,912,0,262,4,1
input_mux=1
palette=8
```
Critical params are the clock of 14.318181MHz and the line length of 912 clock cycles.

I've split the Atom profiles into:
- Atom_V1
- Atom_V2
- Atom_V6
to reflect the CPLD versions.

See 72631aa3

This is mostly for our sanity during testing!

I'm not sure what we should release here.

IanSB, 2020-05-06:
This has been addressed in the latest release with legacy mode:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200504_d19df1c

closed 2020-05-06 by IanSB