hoglet67/RGBtoHDMI #99: PSYNC issues with CPLD 1.x and CPLD 2.x

hoglet67, 2019-04-24:
Ian,

I'm been doing some regression testing with the older CPLD and found an issue.

With the latest build dev software, I'm seeing PSYNC sampling issues when using CPLDs earlier than version 3.0.

If I regress to commit 4f0c567 then things are stable again.

It looks like currently the default and default_double capture loops are incorrectly using the WAIT_FOR_PSYNC_FAST macro.
```
capture_line_default_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST              // expects GPLEV0 in r4, result in r8
capture_line_default_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST              // expects GPLEV0 in r4, result in r8
capture_line_default_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST              // expects GPLEV0 in r4, result in r8
capture_line_default_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST              // expects GPLEV0 in r4, result in r8
capture_line_default_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST         // expects GPLEV0 in r4, result in r8
capture_line_default_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST         // expects GPLEV0 in r4, result in r8
capture_line_default_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST         // expects GPLEV0 in r4, result in r8
capture_line_default_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST         // expects GPLEV0 in r4, result in r8
capture_line_default_double_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST              // expects GPLEV0 in r4, result in r8
capture_line_default_double_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST              // expects GPLEV0 in r4, result in r8
capture_line_default_double_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST              // expects GPLEV0 in r4, result in r8
capture_line_default_double_4bpp_8bpp.S:        WAIT_FOR_PSYNC_EDGE_FAST              // expects GPLEV0 in r4, result in r8
```
I think he bug was introduced in 76f56927 just over a week ago.

I could easily change this back, but though it best to first check with you that this really is just an oversight.

Dave

IanSB, 2019-04-24:
> though it best to first check with you that this really is just an oversight

Yes it was just an oversight, I think I was copying blocks of code around when I reworked the capture loops and forgot to change the PSYNC macro on the two 3 bit defaults.
I've just sent a pull request with the fix

hoglet67, 2019-04-24:
Merged, thanks.

closed 2019-04-24 by hoglet67