hoglet67/RGBtoHDMI #104: Use the camera interface for 24 BPP capture

IanSB, 2019-04-30:
The camera interface is high bandwidth so could be used to acquire 24bpp video for use with later systems like the Archimedes, Amiga etc.

hoglet67, 2019-04-30:
Have you come across any hardware register documentation (or source code) describing how it works?

I looked a couple of years ago, and didn't find much.

IanSB, 2019-04-30:
Some info here
https://www.raspberrypi.org/forums/viewtopic.php?t=148753

HDMI to PI camera interface
https://www.raspberrypi.org/forums/viewtopic.php?t=212783
https://www.raspberrypi.org/forums/viewtopic.php?f=38&t=120702

PAL or NTSC to PI camera:
https://www.raspberrypi.org/forums/viewtopic.php?t=195139
("Adv7282 converts any ntsc or pal source to csi")

Also Ultibo (bare metal OS) seems to support camera capture
https://forum.lazarus.freepascal.org/index.php?topic=38325.0

emesto, 2022-01-01:
This is an interesting idea. but it would require some sort of converter. 

there is this chip from Texas instruments that can convert 24 bit parallel data to serial data that would most likely work on the CSI port.
it is called [SN65LVDS311](https://www.ti.com/lit/ds/symlink/sn65lvds311.pdf?ts=1640986589240&ref_url=https%253A%252F%252Fwww.ti.com%252Fproduct%252FSN65LVDS311). 
this chip however needs 1.8volt to operate.  therefore you would need a level translator to interface with the 5volt levels on the amiga. (first bump in the road I would say)

This is the closest way to use the CSI port I think, but I doubt it would be easy.
I am not sure it would manage to sync to the scanline/frame and/or any of the system clocks for LISA chip either. 

another way to use the CSI port would be to find a proper trippel input video ADC for the CSI port.
Or even a custom FPGA chip.

however you would probably have to make a custom program to read from the CSI port either way, since this project is so minimalistic as it is. 
It would have been interesting too try it tough. 

more information on the CSI interface can be found [Here](https://www.mipi.org/specifications/csi-2)

-EM-

abrugsch, 2022-01-31:
@captain-amygdala has been working on an FPGA board to go between denise and the Pi CSI port for PiStorm (seen here: https://twitter.com/Claude1079/status/1454107559778996226 ) 
with a PoC using a cypress fx2lp board: https://twitter.com/Claude1079/status/1365377191819812868

and the PiStorm32 for A1200 is getting a similar treatment. might be worth harassing him to get the denise intersposer board finished ;)

captain-amygdala, 2022-01-31:
Yes, I use a FPGA (Lattice Crosslink with hard macro MIPI D-PHY). An other, and probably much simpler, method would be sampling the 24bit 28MHz pixels with the Raspberry SMI unit. The SMI is only 16bit wide. But when using a external latches for the upper bits the SMI should be perfectly capable of sampling 24bit 28MHz. Synchronisation to the pixelclock can be achieved trough the latches and using the SMI external DMA pacing pins.