hoglet67/RGBtoHDMI #107: Spare PCBs on eBay

haptic, 2019-06-06:
Hi @hoglet67 

I have been unable to register with the stardot forum and I cannot find contact details for you anywhere so apologies for raising an issue here as the only way of getting in touch.

I put up some spare PCBs on ebay without thinking it might be an issue. I see from the forum (which I can read but to which I cannot post) that you are not happy about this so I have taken the listing down. 

I am happy to send you the remaining PCBs to do what you wish with them - or I can just to dispose of them.

My sincere apologies for not contacting your first. Please let me know what I can do to make amends.

Kind regards

Tim

hoglet67, 2019-06-06:
Hi Tim,

Thank you for contacting me. By all means continue to sell the boards you have on ebay.

I was just slightly taken aback at seeing these appear on ebay without any prior warning.

I'll post a response on startdot saying we had this conversation.

all the best

Dave

haptic, 2019-06-06:
Thank you Dave. I completely understand: I should have tried harder to find a way of contacting you after I failed to register at stardot and I apologise again. I think this is an awesome project and I genuinely just wanted to help promote it. I'm very happy you have been so gracious about it.

Cheers

Tim

PS I'll put the listing back up now.

hoglet67, 2019-06-06:
If you are still interested in registering with Stardot I'm happy to contact one of the admins. Just let me know the username and email address you used.

closed 2019-06-06 by hoglet67

haptic, 2019-06-06:
Thanks Dave. That's very kind. Actually its a browser issue blocking me - I get a message telling me I cannot register because I have not accepted the cookie policy but with no option available to accept it. I will try again later with a different computer. If I get further and then get stuck waiting for an admin to confirm then I'll take you up on your kind offer. Cheers

hoglet67, 2019-06-06:
Do you have a ad blocked in the browser? Try disabling that temporarily:
https://stardot.org.uk/forums/viewtopic.php?p=203977#p203977

hoglet67, 2019-06-06:
Out of interest, who did you use to make the PCBs?

haptic, 2019-06-06:
>> Do you have a ad blocked in the browser? Try disabling that temporarily:

That worked! Thank you.

>> Out of interest, who did you use to make the PCBs?

JLCPCB - I've just posted to your thread on stardot to answer the question that I think prompted your question here!