hoglet67/RGBtoHDMI #118: Seven bit mode

IanSB, 2019-12-21:
It should be possible to tweak this project:
https://github.com/c0pperdragon/A-VideoBoard
to output four or seven bit TTL for C64 and Atari respectively (C64 has 16 and Atari has 128 colour possible combinations).
This should be fine for the C64 but currently we are 1 bit short for Atari machines.

We should be able to get a 7 bit mode running without any hardware changes by using the VSYNC input as the seventh bit. (The 7 bits will fit into the 8 bit frame buffer without any problems as we only use 1 bit for the menu).

To do this without hardware changes would mean that only 1 seven bit pixel could be acquired per psync cycle as there are only 12 GPIO pins to transfer data which would halve the max pixel clock rate but as it currently maxes out at 28Mhz in six bit mode that would mean 14 Mhz in seven bit mode and the Atari has a ~7.16Mhz dot clock so there should be plenty of headroom.

There are a couple of registers left in the RGB CPLD so enough to add 1 bit to the capture mode setting but I don't know if the remaining logic could be squeezed into it.
In any case if it proves impossible, a new 7 bit CPLD could be designed.

hoglet67, 2019-12-23:
>  BTW do you think it would be possible to get a 7 bit mode implemented in the RGB cpld?
( see #118 )

I tried a simple experiment - just extend the scan chain by 1 bit (i.e. 1 extra register), and this failed to route (even with this exhaustive fit mode enabled).

The only way I could get it to route was ti unconstrain the pinout (by removing the .ucf file)

So I think it's unlikely this will ever fit unless the existing logic can be somehow simplified.

One thing I meant to ask you - one of the recent changes (swap_bits) has made the logic for the G functions a bit more complex. Can you explain what this is actually doing?

If you want to pursue this, I would suggest starting a new CPLD design that strips out the multiple sample offsets. That frees up a big chunk of space. As far as I'm aware, only Beeb Mode 7 has such an asymmetric pixel clock.

For future reference, here's the diff of what I tried:
```
diff --git a/vhdl/RGBtoHDMI.vhdl b/vhdl/RGBtoHDMI.vhdl
index 4c24da5..b45c3a6 100644
--- a/vhdl/RGBtoHDMI.vhdl
+++ b/vhdl/RGBtoHDMI.vhdl
@@ -68,7 +68,7 @@ architecture Behavorial of RGBtoHDMI is
     constant VERSION_NUM_RGB_ANALOG : std_logic_vector(11 downto 0) := x"C71";
 
     -- Sampling points
-    constant INIT_SAMPLING_POINTS : std_logic_vector(23 downto 0) := "000000011011011011011011";
+    constant INIT_SAMPLING_POINTS : std_logic_vector(24 downto 0) := "0000000011011011011011011";
 
     signal shift_R  : std_logic_vector(3 downto 0);
     signal shift_G  : std_logic_vector(3 downto 0);
@@ -105,7 +105,7 @@ architecture Behavorial of RGBtoHDMI is
     -- pixel clock is a clean 16Mhz clock, so only one sample point is needed.
     -- To achieve this, all six values are set to be the same. This minimises
     -- the logic in the CPLD.
-    signal sp_reg   : std_logic_vector(23 downto 0) := INIT_SAMPLING_POINTS;
+    signal sp_reg   : std_logic_vector(24 downto 0) := INIT_SAMPLING_POINTS;
 
     -- Break out of sp_reg
     signal invert   : std_logic;
@@ -166,7 +166,7 @@ begin
     half     <= sp_reg(18);
     delay    <= unsigned(sp_reg(20 downto 19));
     rate     <= sp_reg(22 downto 21);
-    invert   <= sp_reg(23);
+    invert   <= sp_reg(24);
 
     -- Shift the bits in LSB first
     process(sp_clk, SW1)

```

IanSB, 2019-12-23:
>  Can you explain what this is actually doing?

It is a roundabout way of making a 3 input to 2 bit encoder.
There are 3 comparators on the green input. 
In one mode I use this as 2 video comparators (sensing 3 levels of video) and a sync slicer.
In the other mode I use this as 3 video comparators (sensing 4 levels of video)
The two options above are switched by the mux input

The lo comparator feeds the lo green bit and the hi comparator feeds the hi green bit with the third comparator feeding the vsync input. 
When mux is low, the possible combinations on the green bits fed to the pi are:
00 when both comparators low
01 when the lo comparator is active
11 when both the lo and hi comparators are active.
Any sync detected by the G/Y sync comparator is available on the Vsync input

When mux is high, the possible combinations on the green bits fed to the pi are:
00 when all 3 comparators low
01 when the  lo comparator is active
10 when both the lo and Vsync comparators are active (Vsync going active caused 01 to become 10)
11 when all three comparators are active (Vsync being active causes 11 to swap but that is no change)

hoglet67, 2019-12-23:
Thanks, that makes sense, in a roundabout sort of way. ;-)

I built up the V2 analog boath this afternoon, and it's working nicely now with my Atom.

I'll create a new issue to record my experiences, so they don't get lost here.

IanSB, 2020-01-01:
If the proposed 7 bit mode used a new cpld design it could be increased to 8 bits per pixel by using the P19 pin as well (currently used for clamp and analog board detection).

Also it should be possible to capture three 8 bit samples in two psync edges, splitting the middle sample into the unused 4 bits of each psync edge.

That would imply a max clock rate of three quarters of the current maximum which would be around 21 Mhz. (3/4 * 28Mhz). I think that would be enough to capture the elan / enterprise which has 8 bit TTL RGB available internally. (3 bits G & R + 2 bits B).
see: http://enterprise.iko.hu/schematics/EP64-3~0.jpg

IanSB, 2020-09-08:
I think this has been covered by multiple commits for >6 bit support together with C64 and Atari support:

https://github.com/hoglet67/RGBtoHDMI/pull/154/commits/91564b9ed91b63f258ab2fcea3559bdb8b0df5b8
https://github.com/hoglet67/RGBtoHDMI/pull/156/commits/8ee368fdf21279c255e5a00607fa21b7032405bf

closed 2020-09-08 by IanSB