hoglet67/RGBtoHDMI #120: Support for v1 and v2 cplds

IanSB, 2019-12-22:
Now we have separate folders for the different cplds, if you want to keep support for v1 and v2 versions in the field you could extend the bbc driver to detect those earlier versions and change the driver name to bbcv1 bbcv2 etc so they would then have their own profiles folders. (Those profiles differ from later versions)

hoglet67, 2019-12-22:
Yes, I was also thinking we should do this.

For the pre-6.x CPLD I was thinking we would only include an Atom and a Beeb profile, to so it's not a pain to maintain.

One reason I checked in every BBC CPLD version was to confirm the version numbers where h_offset needs to change by a large amount.  I have this written down somewhere now.

I need to do the same with the Atom.

I'm actually quite keen to do this, as people are very reluctant to upgrade.

Dave

IanSB, 2019-12-22:
I've just changed the code to hide the geometry menu's deprecated pixel sampling option in rgb/analog/yuv modes. It's still available in bbc mode but that could be modified further so it's only visible in bbcv1 and bbcv2 modes if you want to make the above changes.

IanSB, 2020-05-06:
This has been addressed in the latest release 
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200504_d19df1c

closed 2020-05-06 by IanSB