hoglet67/RGBtoHDMI #123: Update on the new V2 Analog Frontend

hoglet67, 2019-12-23:
I thought I would create a issue to record my experiences with the V2 Analog Frontend, so I can find it again in the future.

I built up the V2 Analog Frontend board this afternoon and it's working nicely now with my Atom.

Firstly, here's your notes from the Stardot PM:

> One small mod is required:
> Cut the track from the clamping capacitor CL1 going to pin 12 of U1 (AGREEN) and connect pin 12 to the other side of CL1 (CGREEN). Originally I had hoped to get composite sync from the 6847 Y signal by trying to extract it from the DC side before the clamp (it couldn't be on the other side as the sync signal is used to derive the clamping signal). That option wasn't reliable at all so I put that signal on the other side of the clamp with the two other comparators. I then fed the 6847 TTL HSYNC into the sync input which provided the clamp and then extracted the composite sync from the Y input that way but only used it to determine the VSync as the TTL HSYNC was already available.
>
> I found the Atom would only work with back porch clamping on the YUV cpld but the dragon would work with either back porch clamping on the YUV cpld or sync tip clamping on the RGB cpld.
>
> One other change I intend to do for the next issue is to add a schottky diode in the +5v supply from the picoblade connector. That way a standard beeb 6 way DIN can be on the other end of the picoblade connector and it could still be plugged into an Amstrad without causing damage as that has mono video on the +5v pin.
>
> Using the 6 pin DIN as the standard might be the way to go as all other computers could then have adapter leads made up from that, i.e. 6 pin DIN cable socket to Atom/Dragon 10 way IDC, 6 pin DIN cable socket to spectrum, 6 pin DIN cable socket to oric etc (I used a 10 way cable mounted IDC plug wired up in the same way as an Atom as the connctor hanging out the back of the dragon)

Here's what I'm seeing with the Atom:

With the RGB CPLD there is always a bit of occasional noise, no matter how I tweak the thresholds. I think you reported the same above. With the YUV CPLD that noise goes away, and everything is perfect.

Note, this currently running with Cx1 - Cx6 (DAC decoupling caps) not fitted.

In the YUV CPLD there are three features that might be helping, that you loose with the RGB CPLD:

1. the L and C noise filters are clearly making a difference. If you disable them the zero-error calibration window narrows dramatically (from 4 positions wide to 1 position wide).

2. the back porch clamp will do a better job of setting a refrence black level than a sync tip clamp, because the black porch level is very close to reference black.

3. the PR/PB signals are sub-sampled, and that sample is taken in the middle of the two pixels. On my original Atom experiments, sub-sampling in the middle made a significant difference. The reason being that on the Atom much of the noise on the 6847 video outputs is caused by the 6847 address bus outputs switching, and I think this happens to shift the sampling point away from this.

A couple of further comments.

Point (3) above (chroma sub-sampling) is only possible with the 6847 because the colour will never change more frequently than ever two pixel clocks (i.e. the effective chroma resolution is half the luma resolution). This is true across all screen modes. It does feel feel specific to the 6847. I'd be surprised if it were true on other systems.

Do you have any other systems that produce YUV (component) video?

One other feature of the original Atom CPLD were some tricks to avoid ever having to distinguish between the 6847's Black and WL levels. (The 6847 puts out 4 different Y levels, in the datasheet referred to as Black, WL, WM and WH). It turns out it's possible to determine each of the 6847's discrete colours with just two Y thresholds: one set between WL/WM and the other set between WM/WH. In the original design, this "logic" was directly implemented in the CPLD. Am I right in thinking it's now replicated in the way the palette is setup?

Anyway, I'm really liking this board. I'm probably going to strip down my V1 board and build up a second V2 board. I'll maybe also post a bit on Stardot, but that might have to wait until after Christmas.

Dave

IanSB, 2019-12-23:
> the PR/PB signals are sub-sampled, and that sample is taken in the middle of the two pixels

It would be useful if you could tweak the YUV cpld to make that switchable with a control bit on the register chain.

> Am I right in thinking it's now replicated in the way the palette is setup?

Yes, the palette replicates the VHDL logic.

> Do you have any other systems that produce YUV (component) video?

Yes, TI-994/a and 48K spectrum. I have had some interesting results with them so far but neither is really usable yet. 
Both of these have many different luma and U/V levels which makes it difficult to discriminate between colours but I used similar tricks to your Atom luma threshold above to infer colours using incomplete data. 
The TI video chip has published info for the YUV levels and I used that to create the following table:  (The first value is the luma voltage and the numbers in brackets are the U/V values that occur at each voltage)
0    (0,0) 
0.40 (+1,-.132)
0.47 (-.321,+679) (-.453,-.642)
0.53 (-.509,-.755) (+.868,-.075) (-.377,+.868) (+.377,+.491)

0.67 (-.377,-.566) (-.377,+.868)
0.73 (+.434,-.887) (-.755,+.189)
0.80 (-.566,+.189) (0,0)

1.00 (0,0)

The blank lines show where luma discrimination is mandatory because the luma values either side of the blank line share the same U/V values. So in the above there must be at least 3 levels of luma discrimination. (These are at relatively wide voltage gaps)
The current interface only discriminates + , 0 and - for the U and V values so I set the thresholds to +.225 and -.225 such that anything between those values is 0, anything above is + and anything below is -. This results in the following:
0    (0,0) 
0.40 (+,0)
0.47 (-,+) (-,-)

0.53 (-,-) (+,0) (-,+) (+,+)

0.67 (-,-) (-,+)
0.73 (+,-) (-,0)

0.80 (-,0) (0,0)

1.00 (0,0)

Due to U/V clashes in the above, actually 5 luminance levels would be needed to fully discriminate so I had to include the 1.00 (0,0) (white) in with the 0.80 which also has (0,0) (grey) as I only had 4 levels.

The main issue is that the thresholds between the luma levels in this configuration are very narrow and susceptible to noise. Using the original 3 levels above with more bits for U/V discrimination would probably be much better.

The spectrum 48K is similar, it has about 15 different luminance levels and it would be impossible to reliably discriminate those but using just 3 luminance levels and the +/0/- U and V levels it should be possible to reliably extract the 8 basic colours but not distinguish between their normal and bright variants. (Which would be the equivalent of connecting a Spectrum 128 or +2 to a beeb style TTL RGB monitor with no intensity input)
There is one remaining problem with the Spectrum: The R-Y (V) output inverts every other line presumably to make it easier to PAL code the signal in the colour encoder chip without any external PAL switch circuitry.
That inversion would have to be undone in the YUV cpld so another bit to enable the feature and some way of synchronising the PAL switch after field sync.

Here are the spectrum YUV signals from colour bars using normal and bright versions of each colour:
[image]
Yellow is Y (inverted) , Cyan is B-Y, Purple is R-Y

> In the mean time, if there is anything you would like me to try

Perhaps investigate if the Rx decoupling caps really make any significant difference. The other thing will likely be the termination mod as mentioned in https://github.com/hoglet67/RGBtoHDMI/issues/119 once I've tried it myself.

hoglet67, 2019-12-24:
Thanks for that info Ian, very interesting...

I've just been reading my copy of The Spectrum ULA by Chris Smith (do you have this?) and it's got some good diagrams on the YUV analog output. Detecting the base 8 colours should be very doable, with just three levels of UV, and probably just two levels of Y. But I think you are right, detecting the bright highlight will require far more Y resolution than we have. And looking at your diagram, if the 0 level of B-Y and R-Y are significantly different, then you are stuffed.

The YUV CPLD changes should be straight forward (assuming there is space):
- make the chrominance sub-sampling controllable
- handle the R-Y inversion on alternative lines

I'm not allowed back in the Lab until the weekend now, so I won't be doing much until then.

I had one other thought I had last night. For a future design, to increase the resolution without needing more DACs, you could use a chain of 1% resistors between pairs of DAC outputs to generate N regularly spaced thresholds. You still need multiple comparators, but this might be more flexible. It would allow R-Y and B-Y to have different offsets. It's an alternative to clamping I guess.

Dave

IanSB, 2019-12-24:
> The Spectrum ULA by Chris Smith (do you have this?) 

No I don't but I have heard of it.

>  looking at your diagram, if the 0 level of B-Y and R-Y are significantly different, then you are stuffed

I did manage to get some reasonable results by specifying a wide band for the "0" level that encompassed the 0 level of both B-Y and R-Y and some of the lower levels of colour values but I was tripped up by the half chroma sampling.

BTW one other thing to look at is the effect treating G differently (by clamping it) has on the RGB noise performance.
It seems to be OK under normal circumstances but if you push the DAC thresholds to the edge of the noise band you get colour fringing indicating that the G is slightly phase shifted. I'm not sure if this is something to be concerned about.

hoglet67, 2019-12-28:
Hi Ian,

Hope you had a good Christmas.

I've done some work just now on the YUV CPLD in a new branch called spectrum:
https://github.com/hoglet67/RGBtoHDMI/commits/spectrum

There are three changes to the YUV CPLD VHDL:
- v4.0 adds a register to the scan chain that enables/disables chroma sub-sampling
- v4.1 reduces the macrocell usage (there is some risk this might have messed things up)
- v5.0 adds a register to the scan chain that enables/disables inversion of the R-Y channel on alternate lines

I've done some testing of this on my Spectrum 48K Issue 4B , and it's not yet working reliably.

The problem I think is that the inversion of the R-Y channel really needs to happen only on the even numbered lines within the 312 line field. At the moment, the CPLD does not see any kind of vertical sync pulse (at least, not the way I have it wired up), so it's not possible to synchronise this to the even lines.

I'm wonder whether we could feed a signal from the Pi into the CPLD (possibly overloading the mux signal) that could be used for frame synchronization. Alternatively, the I wonder if the CPLD could detect the field sync pulse itself somehow? But that would need a longer counter than we currently have space for. Did you have any thoughts on this?

Anyway, I thought I would push this up to github so you can take a look and have a play.

One thing to be wary of is that when the OSD is displayed I think the display is updated every two fields, and a consequence of this is that sometimes this interacts wierdly with the R-Y inversion, especially if this is out of sync. 

Dave

IanSB, 2019-12-28:
> The problem I think is that the inversion of the R-Y channel really needs to happen only on the even numbered lines within the 312 line field

Yes, the PAL switch needs to be synchronised for things to work as mentioned in my first post above.

> possibly overloading the mux signal

Yes I think that's probably the way to go as the MUX in YUV mode currently controls the 3/4 level switch but blipping it after field sync shouldn't cause any problems as there is no video capture during that time.
I think the way to do it would be to expose a new "synchronise" method in the cpld driver(s) and call that method after field sync detection has finished. In other CPLDs that should do nothing but in the YUV cpld it should briefly invert the MUX line if config->alt_r is enabled and that should force the PAL switch (inv_R) to 0 or 1 (not sure which way round the logic would be)

> One thing to be wary of is that when the OSD is displayed I think the display is updated every two fields

With the longer menus, even the 8x8 font started dropping fields so I gave up on trying to keep 50Hz capture during menu display and defaulted to the mode 7 font even in 8 bit mode.
Making that work nicely with the PAL switch might get messy.

Perhaps we should look at getting some of the capture interrupt driven.

BTW do you see the same DC offset in R-Y and B-Y ?

hoglet67, 2019-12-28:
I've found a fairly efficient way to detect the FS pulse in the CPLD. It re-uses the counter to measure the width of the HS pulse. If the HS pulse is longer that 511 clock cycles (9us in the case of the Spectrum) then this clears the R-Y inversion signal.

This is pushed as CPLD v5.1.

It may not be the best approach (using mux might be more flexible), but it lets us see if there are father problems, which I think there are.

Here's what I'm seeing now:
[image]

The top 7 lines have black text, the bottom 7 have white text.

I think possibly the errors might be caused by the delay being introduced by clamping Y.

Dave

IanSB, 2019-12-28:
It could be tilt introduced by clamping being messed up during the field sync pulses.

hoglet67, 2019-12-28:
No, it's actually a bug I just introduced in trying to save some macrocells.

So I reverted that bit, and removed temporarily removed the filtering, and this looks better:

[image]

hoglet67, 2019-12-28:
I think I've found a way to fit everything in.

v5.2 now pushed.

This does need to be re-checked on the Atom, to make sure the filters are still working properly. I haven't done that yet.

IanSB, 2019-12-28:
I just issued a pull request for a few minor changes I made recently including a fix for the very small screencaps on the Spectrum (also affected other computers). You may want to merge this at the same time as merging the new branch back in to consolidate everything.

hoglet67, 2019-12-28:
I've merged everything back to dev now.

I'll do a bit more testing on the Atom tomorrow to make sure nothing has regressed.

I'm still not able to get a perfect picture on the Spectrum.

For example, with the following test program:
[image]

I can't get a perfectly stable picture:
[image]

Here are my current sampling settings:
[image]

I'll see if adding a decoupling capacitor to the DAC helps.

Dave

IanSB, 2019-12-29:
I've pushed an updated spectrum palette and profile directly to your spectrum branch
The palette now uses only 2 levels of luma discriminiation with some overlap which gives +-10 levels of noise free threshold. The U/V lo/hi adjustments seem to have +-5 levels of noise free threshold.
Only the G/Y lo adjustment is used now, the hi is set to 255 to disable it.

The main problem left is that there seems to be some vertical tilt which is affecting the sync separation. As I adjust the sync threshold I can get clean video at the top, middle or bottom but not reliably all at the same time. Not sure what's causing the tilt, it may be that the clamp circuit is loading the Y signal and affecting it's level or there may be an issue with the Spectrum itself.

IanSB, 2019-12-29:
Just had a thought:

There is a second sync separator on the clamped side of the Y signal feeding the Vsync input.
If the normal sync separation was used just to generate the clamp, the secondary sync input which would be clamped could drive the timing chain and that would hopefully eliminate any sync separation tilt.

However I don't think that would fit in as it would require 2 sets of timing counters, one for the clamp and one for the psync. This could be tested with sync tip clamping (i.e. drive the clamp directly from the sync input and drive the timing chain from the vsync input) as that would eliminate the need for the second counter.

hoglet67, 2019-12-29:
Hi Ian,

I've updated to the latest code and profile you pushed to the spectrum branch.

I'm not sure I'm seeing quite the same issue that you are seeing with vertical tilt.

Could you post a capture of the sort of artifact you are seeing (ideally with my small random test program)?

Do you have any scope pictures that show the vertical tilt?

To get a stable image, I needed to adjust three settings:
- Offset from 4 to 1 (this had a big effect)
- DAC-F (Sync) from 222 to 255 (this had a big effect)
- DAC-D (RB/UV Lo) from 125 to 138 (this had a minor effect on the last few twittering pixels)

[image]

I think the sync change is needed I think because there is a glitch on ULA Y output in the back porch region.

With these settings, the main part of the image is perfect.
[image]

The only artefacts are where the border changes to blanking, and this transition probably shouldn't be visible anyway (i.e. reduce the Max H Width Geometry to 344).

I can think of a few possibilities:
- there are differences between our Spectrums (what issue is yours?)
- there are differences between our cabling (I'm picking the signals directly off the ULA pins 15-17)
- there are differences in our analog adapter boards (did you definitely fit a 10uF cap as C3)

Dave

IanSB, 2019-12-29:
My Spectrum is an issue 3
I'm also picking up directly from the ULA
There is definitelty a 10uF capacitor fitted.

However I can see AC loading effects even when the spectrum is disconnected from the RGB/YUV board. e.g. if I change the border from 7 to 0 the video will move up/down on the scope and doing this when connected to the RGB/YUV board results in a clean image going noisy.

This is the signal I get on the Y output:
[image]
The large pulse is the field sync and the block of lines following it are a black border with line sync pulses followed by another block of lines with white video background (inverted video).
The ULA outputs are driving the internal circuitry and that might be affecting things.
Also my spectrum has been modded for video out on the RF modulator's phono connector and that might be affecting things although it's not connected to a video out at the moment.

hoglet67, 2019-12-29:
Here's a couple of similar scope shots....

Border 0

[image]

Border 7

[image]

The Y cursors are showing:
- Y1 = black = 2.422V
- Y2 = white = 1.469V

I don't see any perceivable shift in the black level when changing the border.

My FS pulse also appears much squarer than yours,

Can you try to quantify the black level shift you are seeing?

I'm wondering why our clamping circuit is not compensating for this shift.

Dave

hoglet67, 2019-12-29:
OK, I've tried another test, using my 6-line test program above.

With this, if I change the BORDER from 7 to 0 then I do see some disturbance, even though there is no perceivable shift in the waveform. What's happening in my case the R-Y inversion is occasionally taking the wrong phase with BORDER 0.

If I reduce the DAC-F sync threshold from 255 to 250, it goes away.

A couple more tests:
- with a white border, DAC-F needs to be 243-255
- with a black border, DAC-F needs to be 241-252

(otherwise the Y-R inversion occasionally glitches)

Can I just confirm that I have things connected in the same way that you do...

Specifically, I have the Y output of the ULA (pin 17) feeding both the G and Sync inputs of the Analog board.

(So clearly our clamping circuit doesn't come in to play regarding the Sync)

hoglet67, 2019-12-29:
Ian,

> However I don't think that would fit in as it would require 2 sets of timing counters, one for the clamp and one for the psync. This could be tested with sync tip clamping (i.e. drive the clamp [...]

Try this version of the CPLD and see if it helps:
[YUV_CPLD_vv53.zip](https://github.com/hoglet67/RGBtoHDMI/files/4008440/YUV_CPLD_vv53.zip)

I've remove the L and C filters, and added a second counter.

The HS input (DAC-F = Sync) is now used only to generate the clamp output.

The VS input (DAC-E  = G/Y Sync) is used for everything else, including the CSync output.

(I have both set to 215)

If this helps, I'll push it to a branch while we consider what next.

Dave

hoglet67, 2019-12-29:
Sorry, there was an error in v53, here's v54 that fixes that.

[YUV_CPLD_v54.zip](https://github.com/hoglet67/RGBtoHDMI/files/4008500/YUV_CPLD_v54.zip)

This doesn't seem to work particularly well for me (v52 and v52 seem better)

I'm rather confused what's going on here. I think I need to hook up some wires to the clamp signal and the clamped Y. But that will have to wait until tomorrow.

BTW, v53 and v54 are on a spectrum2 branch if you want to examine them.

Dave

IanSB, 2019-12-29:
> Specifically, I have the Y output of the ULA (pin 17) feeding both the G and Sync inputs of the Analog board.

Yes, that's right. Same here.

Your symptoms above are effectively a less severe version of what I'm seeing.
The sync threshold has a significant effect on results and adjusting it varies the sampling point by something like +- 1 pixel. As the DC level of the sync is itself varying across the frame I'm seeing the start point of the sampling vary vertically across the frame such that the video is clean at the top but noisy at the bottom or the reverse of that depending on the sync adjustment.

Also I noticed glitches in the sync pulses like we had with my Master 128 so maybe it would benefit from some deglitching on the sync pulse input like the RGB cpld.

[image]

Here is a comparison of border 0 and border 7:

Border 0
[image]

Border 7
[image]

There is a shift of ~150mV in the border area

hoglet67, 2019-12-29:
I've also notices those glitches, so yes, some filtering may help.

The sync pulse is a bit assymetic, specifically:
- the leading (rising) edge is ~50ns (from black to sync)
- the trailing (falling) edge is ~200ns (from sync to black)

So it may also be beneficial to use the leading edge as the reference, rather than the trailing edge.

Dave

IanSB, 2019-12-29:
> Sorry, there was an error in v53, here's v54 that fixes that.

I tried v54 but it didn't seem to help

However I did get the following result while varying the G/Y sync threshold which indicates the glitches are having an effect at some threshold levels:
[image]

Still getting left to right movement in the sample position:
[image]

One other thing I noticed is that the peak sync tip level is >3.3v at some points so maybe the regulator should be increased to 4v to allow full range adjustment (There is a similar issue with the TI99/4a where the chroma is sitting on a pedestal and is >3.3v)

IanSB, 2019-12-30:
I've pushed another updated spectrum palette and profile directly to your spectrum branch.
This won't help with the current problems but it adds detection of 2 levels of white as that can be done using the unused comparator on the luminance. As white is used for the default border and background this might be more useful than any of the other colours.
Note G/Y Hi should be set to the previous G/Y lo setting and the G/Y lo setting should be adjusted for detection of the two white levels: (The noise free adjustment is about +-7)

[image]
To disable it and revert to one level, set G/Y lo to 0

hoglet67, 2019-12-30:
Ian,

I've just pushed YUB CPLD v56 that is based off v52 (single counter) and adds a 2-bit HS counter. This should ignore glitches of 1, 2 or 3 samples (i.e. upto about 50ns).

I would have liked to use a 3-bit counter, but it didn't fit.

(On my Spectrum the glitch in the back porch is 35ns wide at it's base.)

This does significantly improve the range of Sync settings before the display starts breaking up:
- v52 I can use 242-253
- v56 I can use 196-253

Like you, I'm seeing a shift in sampling point when I switch from a black to white border:
- black border: offset 5 is best (not quite perfect though)
- white border: offset 3 is best (close to perfect)

I don't really understand why this shift in sampling point is occurring, because on my system the shape/position of the sync pulse is unchanged. Possibly there is some phase shift occurring across our clamping circuit, and this varies depending on the DC level.

Dave

hoglet67, 2019-12-30:
Ian,

One last exeriment for this morning...

I've just pushed YUV CPLD v57 that uses the leading (rising) edge of HS as the timing reference.

With this, I'm seeing far less of a shift in sampling point between the black/white borders.

I'll pause now, and see how you get on with v56 and v57.

Dave

hoglet67, 2019-12-30:
I think there's one more uncontrolled factor possibly coming into play here - the quality of the power supply. If I power the Spectrum from my bench power supply, there is generally less noise than when using the (non-original) 9V power brick. Just something to bear in mind!

Also, I'm now less worried about the mis-sampling at the edge of the border area. If you look on the scope, there does seem to be significany jitter in this transition. There is also jitter between the two edges of the sync pulse.

Dave
Dave

IanSB, 2019-12-30:
The 57 version is a lot more stable and I can get a good sampling point which works at all vertical positions:
[image]

Also changing the border to 0 doesn't affect it:
[image]

Does this version still work with the Atom?

hoglet67, 2019-12-30:
> Does this version still work with the Atom?

It should do (the Atom's sync pulse width is 4.61us (16.5/3.579545) which is almost identical to the Spectrum's 4.57us (32/7).

But I haven't actually tested it.

I've had an idea for a way to significalty optimize the YUV CPLD, saving 6 (or possibly 8) registers. This would allow the edge to be software selectable, and give us some room for future enhancements.

Dave

IanSB, 2019-12-30:
> I've had an idea for a way to significalty optimize the YUV CPLD, saving 6 (or possibly 8) registers

Sound good, lots of room for more features :)

I just tried your torture test above and that was also noise free:
[image]
[image]

I'm guessing your sampling values will be slightly different to mine but I think it should be possible to automatically optimise them similar to the sampling position as long as the optimisation starts from the assumption that they are approximately correct and only varies them up/down by a small percentage.
[image]

hoglet67, 2019-12-30:
One thing I noticed is that the auto calibration doesn't work.

I think it's because it's operating directly on the frame buffer, before the palette is applied.

IanSB, 2019-12-30:
Just tried the Atom and that worked OK

One problem I noticed with the current and other recent cpld versions is that you can get the system to lock up if you adjust the sync level incorrectly. I think this might be due to the psync pulses disappearing when sync is stuck high (or maybe low). 

e.g. with the spectrum profile if you set the Sync value to 0 you can't make any further adjustments and have to hit reset. This can also have an effect when changing profiles. This doesn't affect the v33 cpld and you can set sync to 0 on that one without hanging the Pi.

I think psync continues on the RGB cpld when sync is high or low so would it be possible to get psync to continue on the YUV cpld. (It's not possible to put a software timeout on psync as that would impact the max sample frequency)

IanSB, 2019-12-30:
> I think it's because it's operating directly on the frame buffer, before the palette is applied.

Yes, there are now several possible values in the frame buffer that produce the same value on screen (this was an attempt to improve noise immunity) so the calibration will have to use a 256 byte lookup table to map equivalent values.

hoglet67, 2019-12-30:
> One problem I noticed with the current and other recent cpld versions is that you can get the system to lock up if you adjust the sync level incorrectly. I think this might be due to the psync pulses disappearing when sync is stuck high (or maybe low).

I've just pushed a fix for that (v5.8)

IanSB, 2019-12-30:
If you manage to save all those registers, one additional thing now required is the two bit delay value as used in the RGB cpld. It was hard coded for the Atom but now needs to be adjustable.

IanSB, 2019-12-31:
Some new changes pushed to the Spectrum branch:
I've refactored the 6847 palette generation and added an equivalence table so that 6847 and Spectrum palettes containing multiple entries for the same colour can be compared correctly by the auto calibration algorithm. This seems to work as without the equivalence table I can't get a zero error point on the Spectrum but with it I get two zero error points. 

The 6847 palette generation was rewritten to make the equivalence generation easier. Before it used the original Atom palette code with a workaround for the RGB/YUV board. Now it's the other way around. 

I have tested this on an Atom with the RGB/YUV board and with the original through hole Atom board and both produce the right palettes with the test program.

I also tweaked the palette generation slightly:
Previously the MK1 Atom palettes (now Dragon/CoCo) were very desaturated to make them reversible to 4 level mono but as that didn't look as saturated as the real Dragon composite output I have increased the saturation to try and match that more closely. If the mono/green/amber modes are selected the saturation is reduced to the previous level so that true 4 level mono/green/amber video is output.

Finally I thought the oranges in the MKII palettes were a bit dull and so I changed them to use the same calculated oranges as the improved dragon/coco palettes - you can change them back if you don't like them!

hoglet67, 2019-12-31:
Ian,

I've just pushed YUV CPLD v60 that adds:
- selectable timing reference edge
- 5-bit offset (which effectively gives the 2-bit of delay)

I've also updated the YUV CPLD Driver to expose these parameters, where I have brought delay out as a seperate 2-bit value. The split between delay and offset is a little complex, because it depends on whether colour sub-sampling is enabled. But we can revise the driver model if needed.

I have not yet updated any of the profiles, so they will default to trailing edge and zero delay.

A did a couple of optimizations that reduced usage from:
- v5.8 (71 macrocells / 217 product terms) 
- v5.9 (63 macrocells / 226 product terms) 
- v5.A (57 macrocells / 267 product terms)
- v6.0 (59 macrocells / 296 product terms)

The key idea behind this rework is running the counter and PSYNC generation in phase with the pixel sampling, allowing the internal 12-bit shift register to be eliminated.  This wasn't possible in the RGB version because we had 6 different offset values. 

The current CPLD usage is:
```
Function    Mcells      FB Inps     Pterms      IO
Block       Used/Tot    Used/Tot    Used/Tot    Used/Tot
FB1          13/18       41/54       88/90       6/ 9
FB2          18/18*      32/54       42/90       8/ 9
FB3          11/18       44/54       83/90       9/ 9*
FB4          17/18       33/54       83/90       6/ 7
             -----       -----       -----      -----
             59/72      150/216     296/360     29/34
```
So whilst there are plenty of free macro cells, product terms are now the limiting factor. There are various ways to trade one for the other, so I'm not too concerned. 

In summary, we can probably squeeze one or two more things in, but it's likely to be increasingly painful.

Do you have a feel for the range of pixel clocks we need to deal with? I see that currently in the profiles we have the TI99/4a @ 5.369MHz up to the Atom at 7.159MHz. There are two reasons I ask:

1. The end of clamp is a fixed number of clock cycles after the reference edge (either 240 or 240+256 depending on which edge). If the pixel clock is very slow, then the clamp might extend beyond the end of the back porch. This may well be an issue on the TI99/4a, and we might need to shorten the clamp period.

2. Some of the CPLD functions are quite complex now, and this may limit the speed of the design. There is some timing overhead to functions with large numbers of product terms. We have some now with 21 product terms, which I think adds a couple of 2ns to the max clock cycle time.

Dave

IanSB, 2019-12-31:
If I set the sync edge to leading I get loss of sync on the spectrum profile with v6.0

hoglet67, 2019-12-31:
> If I set the sync edge to leading I get loss of sync on the spectrum profile with v6.0

That's rather unexpected... I've just checked with my Speccy and it works fine for me.

Does the Delay setting have an effect?

IanSB, 2019-12-31:
I adjusted the sync slice level and got sync back (reduced from 230 to 222) but it's clearly behaving differently to the earlier version, perhaps some interaction with the glitch removal.

hoglet67, 2019-12-31:
Hmmm, I still can't reproduce this.

Can you try CPLD versions 5.8, 5.9 and 5.A and see if any exhibit the behaviour. All these use the leading edge of HSYNC. The 5.9 and 5.A CPLD files are in the vhdl_YUV_6bit directroy.

Can you also confirm exactly what you mean by "loss of sync"? Is this is "no sync detected" error? Or do you mean the system just hangs.

The HSYNC deglitching logic has not changed at all between 5.8 and 6.0.

> Changing offset and delay have no effect

Even when sync is in range?

Can you do a clean rebuild of your kernel? (make -B)

hoglet67, 2019-12-31:
I still can't reproduce quite what you are seeing, but the only change I can see that might cause anything remotely like this is change to the way the clamp signal is generated.

Can you test v6.1 (just pushed) to see if this resolves the issue for you?

IanSB, 2019-12-31:
V5.8, 5.9 and 5.A all work OK

> Can you also confirm exactly what you mean by "loss of sync"?

Trailing edge:
[image]

Leading edge:
[image]

As mentioned if I adjust the sync threshold I do get a lock over a very narrow range
Offset and delay work normally when sync is locked, they don't have any effect on the loss of sync

Tried a clean rebuild with no difference

In this failure mode the clamp is on all the time except for field sync and that is having an effect on the input Y signal on the spectrum side of the clamping capacitor. I think the clamp must be behaving differently than before and this is causing a feeback loop.

IanSB, 2019-12-31:
V6.1 fixed it.
As mentioned above, what was happening was that the clamp was on for a long period and that had an effect on the relatively high impedance spectrum side of the clamping capacitor which moved the Y signal and that resulted in loss of sync detection which left the clamp permanently on.
Ideally, there shouldn't be a condition that leaves the clamp permanently on, even when there is loss of sync.

hoglet67, 2019-12-31:
OK that's great, and does delay work as well?

So did you have a scope connected to clamp when this happened?

How long was clamp being stuck on for?

I still can't see what set of conditions would cause this.  There needs to be a rising edge of hsync (after inversion) to initially set clamp (and reset the counter), then the counter must somehow avoid the clamp_end count value. This could only happen if there was a continuous sequence of hsync glitches close together (i.e. < ~10us appart) that make it through the glitch filter.

I would like to try to understand this case.... (or it will come back and bite is later on)

Dave

hoglet67, 2019-12-31:
I thought of another possibility that might explain this.

The new clamp behaviour in v6.0 was to enable the clamp on the trailing edge of hsync, rather than using a count. I wonder if the following sequence was happening....

1. There is a glitch in the middle of the hsync pulse
2. That glitch is wide enough to get through the hsync filter
3. This causes the clamp to immediately be engaged
4. But we are in the middle of HSYNC, so Y is clamped to the wrong level (much too high)
5. This breaks sync detection

If this glitch happens in every sync pulse, I can see how the condition would persist forever.

You must have a much more signifcant glitch in the middle of hsync than I have.

Which suggests making the HSYNC deglitch counter longer.

Can you try to measure the width of the glitches you are seeing?

IanSB, 2019-12-31:
> So did you have a scope connected to clamp when this happened?
> How long was clamp being stuck on for?

It happens immediately at startup so is difficult to catch but after it gets stuck this is the state of the sync input and the clamp output:

[image]

Cyan is the sync input and these are now field sync pulses only (remember my field sync pulses are slightly higher than line sync -see above grabs) so after the drop in Y video level due to the terminating effects of the clamp, only the field sync is getting through the sync slicer.
This results in the clamp (yellow) being on all the time except during field sync itself.

> does delay work as well?

Yes. With the RGB cpld, the delay is 0-15 by using two bits from the h-offset as well (It has to be this range as mode7 needs at least 0-11 to get alignment). 
Do you want me to change the YUV to work in the same was as all the hooks are already there?

hoglet67, 2019-12-31:
Thanks for the scope shot.

I've just been messing about with v6.0 plus a somewhat crippled hsync filter. I couldn't replicate exactly what's happening on your system, but I certainly noticed at certain sync values their is a feedback loop and the you get continuous oscillations of the clamping output. These are clearly visible on the Spectrum's UHF output. I also replicated this on the v3.3 CPLD and was able to see these as well. So not a new issue, just new to me.

I think you have mentioned this before, but I handn't really understood that it could affect the unclamped Y from which sync is derived, and thus can become self-fulfilling.

Thinking about this, the original clamping circuit does rely on a low(ish) source impedence for Y, so that the largish 0.1uF coupling capacitor can be quickly charged to the right voltage.  And once in the steady state, there won't be much current flowing any way.

> Do you want me to change the YUV to work in the same was as all the hooks are already there?

Yes, please do if you think that would be useful.

> If I change the sync threshold to get just field sync pulses feeding into the v61 cpld I get 1 short clamp pulse and the clamp stays low for the rest of the field.

Does this happen for both settings of edge?

Can you confirm that doesn't happen with (say) v5.8 and v3.3?

(I need to solder a wire onto the clamp signal so I can scope it as well; I'll do that tomorrow)

Dave

IanSB, 2020-01-03:
> Thinking about this, the original clamping circuit does rely on a low(ish) source impedence for Y

I might experiment with different clamping resistor values to determine the highest value that produces good results.

> Does this happen for both settings of edge?

Only on leading edge

> Can you confirm that doesn't happen with (say) v5.8 and v3.3?

It only happens with v6.0 and then only on leading edge.

> leading edge of hsync as the timing reference has introduced various edge cases

One option might be to try the two counters approach again as there are probably enough registers available now although I don't know if the logic will fit.
i.e. one counter for psync which can be leading or trailing edge triggered and another counter for the clamping pulse which is always trailing edge triggered. This counter could be a reduced number of bits as it only needs to count from the trailing edge of sync to the end of the clamp pulse and it could be stopped one cycle after that point and reset on the next trailing edge. Hopefully running the clamp off the trailing edge won't introduce any jitter back into the sampling.

This has the advantage that the clamp will work with any sync pulse length. (e.g. the ZX80/81 has a sync pulse of around 7uS although in practice they would have to work with sync tip clamping due to lack of a back porch).

Actually that would be another useful feature: The ability to switch between back porch and sync tip clamping as I'd like to keep all the mono profiles in the YUV cpld.

BTW I've got user definable palettes working:
On startup the Pi creates a palettes folder and checks for the presence of the default palette set (one file per palette) and if those files are not present, it creates them. Then it enumerates the Palettes folder and presents the filenames found as the palette options. This means you can add extra palettes to the folder and also edit the existing palettes (if they already exist it doesn't overwrite them).

Do you want me to push that to the spectrum branch or wait until you have merged that back into dev?

hoglet67, 2020-01-03:
As v6.1 appears stable, I've just merged the spectrum branch back to dev.

Feel free to commit this back to dev, or just generate a pull request in the normal way.