hoglet67/RGBtoHDMI #126: Vsync detection issues

IanSB, 2020-01-08:
I've looked into the UK101's Vsync issues some more and this is what I've found so far:

The default Vsync detection determines whether the field is odd or even by checking the time between the last rising edge of the larger field sync pulse(s) and the rising edge of the first hsync pulse after field sync. If this is > ~32uS it is an even field, otherwise an odd field.

With 6845 generated sync pulses, the vsync starts either 1/3 or 2/3 of a line after the last hsync pulse and occupies 2 lines. As a result it obscures 2 hsync pulses in both fields. In order to get the video of both fields to line up with interlaced video sources, one extra line after field sync in the odd field has to be skipped before the start of active video. This vsync detection method also works for most non-interlaced video sources because both frames are either < 32uS or > 32uS so either all frames get the skipped line or none of them do.

The Electron causes a problem with this because although it outputs an interlaced video signal, the field sync pulse starts at 0 or 1/2 of a line instead of 1/3 and 2/3. It is still 2 lines long and the vsync that starts at half way still obscures 2 hsync pulses but the vsync that starts at 0 obscures 3 hsync pulses and as a result it doesn't need to skip the extra line in the odd field.
Ironically this option is required for the interlaced Electron output even though not skipping a line is the way to ensure that non-interlaced video is handled correctly.

The UK101's Hsync is derived from counter chain output C6 so the leading edge is derived from a crystal but the length and thus the trailing edge is determined by a monostable triggered by C6. This is also the case for the Superboard but with C7 instead of C6 (due to the approx half crystal speed).

I think this means we would get more reliable calibration by using leading edge hsync.

The UK101's Vsync is derived from counter chain output C15 but in this case it triggers two monostables, one to delay the position of the Vsync pulse and one to determine the length of the Vsync pulse. This means the Vsync can drift both in position and duration with respect to the Hsync and that can confuse the interlace detection code resulting in jumpy video. 

The Superboard's Vsync is also derived from counter chain output C15 but it only triggers one monostable for vsync length so the start of the Vsync pulse will always be phase locked to the start of the hsync pulse and thus it is 50% less likely to experience problems.   

For both the UK101 and the Superboard, if the time between the end of the vsync and the next hsync drifts to around 32uS then vsync will jitter between odd and even field detection with resulting jumpy video and that's why selecting Electron type sync eliminates vsync jitter.

So at a bare minimum, The UK101 and Superboard profiles should have leading edge hsync and Electron type Vsync.

However there still remains an issue when the end of vsync drifts close to the first hsync so there is a very narrow pulse at that time as that will cause jitter as one hsync pulse is randomly merged into the end of the vsync pulse. This would affect both the Superboard and the UK101 and switching to leading edge vsync detection wouldn't help as there would still be the random loss of the first hsync pulse.

The software currently determines the line length so a possible solution would be to measure the average time beween the last valid hsync before entering vsync and the first valid hsync after leaving vsync, dividing by the line length and ensuring the duration is the same for all fields by skipping hsyncs as required

Note: this means we will be using leading edge hsync with other computers/clock speeds, not just the Spectrum. As you are already looking at eliminating the psync delay, make the counter value the same for leading and trailing edges. This will mean a big shift when switching edges but that can be automatically compensated in the software by adjusting h-offset and as the software knows the clock rate it can do it accurately for all clock speeds.

hoglet67, 2020-01-08:
Some randon thoughts as I try to digest this....

> With 6845 generated sync pulses, the vsync starts either 1/3 or 2/3 of a line after the last hsync pulse and occupies 2 lines. As a result it obscures 2 hsync pulses in both fields.

Just one clarification...The Beeb and Master are implement this slightly differently. The Beeb uses a NOR gate, so behaves as you indicate above. The Master uses an XOR gate, so inverted hsync pulses are still visible.

> The UK101's Vsync is derived from counter chain output C15 but in this case it triggers two monostables, one to delay the position of the Vsync pulse and one to determine the length of the Vsync [...]

Looking at the time constants of the two monostables:
- IC28a 74123 uses 27K and 220nF, which gives a time of 0.33 x 0.22 x 27000 = 1,960us = ~30 lines
- IC65a 74123 uses 10K and 100nF, which gives a time of 0.33 x 0.10 x 10000 = 330us = ~5 lines

With 5% tolerance components, the drift in the start of VSYNC will be upto 1.5 lines. So even switching to the leading edge, it's not clear this will work reliably, even this leading edge of VSYNC will move significanltly (i.e. the variation from the first monostable is dominant).

I would be tempted first to try replacing the resistors and capacitors with modern 1% tolerence types and see if this helps (if you haven't already done this).

Stepping back, it feels like we've already invested a lot of time/effort/testing in the current method of vsync detection. If I recall, it was quite painful to get this working with the Beeb, Master and Electron. I'm not against adding a second method, if that proves to be better in some situations. But I don't think we should be replacing what we already have. Especially if this is just to deal with one arguably poorly designed system. 

> Note: this means we will be using leading edge hsync with other computers/clock speeds, not just the Spectrum. As you are already looking at eliminating the psync delay, make the counter value the [...]

I would like to get rid of this delay, but I think there is a possible race condition when using the leading edge....

As far as I remember the software is currently doing:
1 - WAIT_FOR_CSYNC_0
2 - WAIT_FOR_CSYNC_1
3 - skip HOFFSET PSYNC edges
4 - start sampling on each PSYNC edge

The RGB CPLD is inserting a small delay between (2) and (3) where PSYNC is unchanging, but it's still enough that sampling is consistent.

The YUV CPLD currently inserts a much longer delay, so PSYNC is stable for longer after the end of HSYNC, regardless of which edge is used.

Now, if we update the YUV CPLD so that PSYNC starts toggling short time after that start of HSYNC, then it seems we have a race condition. More specifically, the skip PSYNC edge code (3) may now start at the same time as a PSYNC edge. Then we'll see significant sampling jitter.

I will make the CPLD change and try this out.

Dave

I

This makes sense I think.

IanSB, 2020-01-08:
> But I don't think we should be replacing what we already have.

Any new vsync technique would always be a switchable option.

IanSB, 2020-01-09:
I've updated the first post to correct an error in the Electron sync description

IanSB, 2020-10-08:
This was addressed in https://github.com/hoglet67/RGBtoHDMI/pull/154/commits/7f575bf138f9cf73b7a447691d87c0d5912df4a0

closed 2020-10-08 by IanSB