hoglet67/RGBtoHDMI #127: Spectrum Regression

hoglet67, 2020-01-08:
Ian,

I've just setup the Spectrum again, and I'm seeing very poor results, much worse than before.

In particular, the DAC-C (RB/UV Hi) adjustment is now very narrow.

I've testing with CPLD v6.1 so that I can work out when this started. In each case, I'm rebuilding the kernel, deleting the old Profiles and copying new kernel Profiles.

The last commit which works well is: d45d2e5d

I can vary DAC-C from 172 to 186 and the picture stays stable.

Skipping forward two commits to: dc13e0f9 it's much worse.

The good range of DAC-C is now just 185-186.

The work that was done between those commits is some minor Profile tweaks and adding the User Defined Palettes.

So I think this is a Palette Issue.

It is possible the Spectrum 48K 9-Colour Palette got changed somehow in this process? It seems to have lost some of it's noise masking properties.

(I'm at a loss to think of any other explanation.....)

IanSB, 2020-01-09:
I changed the Spectrum palette slightly to increase the overlap which improved noise immunity slightly for me but I guess not for you so I've reverted that change.

hoglet67, 2020-01-09:
Thanks for looking at this - it's fixed the problem.

closed 2020-01-09 by hoglet67