hoglet67/RGBtoHDMI #132: Eliminate fixed delay from YUV CPLD

hoglet67, 2020-01-09:
I've committed some code for testing to a branch called yuv80

This contains a new version of the YUV CPLD that replaces the large fixed delay with a smaller programmable delay.

Would you mind giving this a try on your Spectrum (which has a worse glitching than mine)? It's quite possible there will be problems.

You'll need to:
- rebuild the kernel to get the new cpld_yuv code
- update your spectrum profle (this just changes the value of HOFFSET)
- reprograme the CPLD with version 8.0

The CPLD works a bit differently now:
- the counter has been reduced from 10 bits to 7 bits 
- the Edge setting controls which edge of HSYNC resets the counter
- PSYNC is stopped on the trailing edge of HSYNC
- PSYNC is restarted when the counter expires

To ensure the software correctly synchronises to the first edge of PSYNC, there is still a need for a small delay between the rising edge of HSYNC and the first edge of PSYNC.

If this delay is too short, then you get lines shifted by whole quads. Here's a couple of examples:

[image]

[image]

This can in theory occur when the reference Edge is either Leading or Trailing, but I'm only actually seeing it when edge is leading. 

I've made this delay programmable, by extended the hardware "offset" value from from 5 bits to 7 bits.

Dave

IanSB, 2020-01-09:
I tested this on my spectrum and I'm seeing the following:

No glitching with trailing edge in all possible offset and delay combinations

Glitching the same as above with leading edge when delay = 14 and offset = 0 to 4 also when delay=15 and offset = 7

Does this need to be tested with any other computers?

hoglet67, 2020-01-09:
It might be worth testing your UK101.

As the delay has changed, the current DELAY and HOFFSET values in the profiles will need adjusting. This is probaby deterministic if I think about it a bit. I'll apply myself to this tomorrow.

Dave

hoglet67, 2020-01-10:
I've just pushed a v8.1 that fixed a hang if the sync threshold was out of range (i.e. v8.0 re-introduced the the same bug we had before).

I've also updated the YUV Profiles, increasing HOFFSET by 16 in each case. I've tested the Atom, Spectrum and Superboard Profiles. All seem fine.

Dave

hoglet67, 2020-01-10:
Closing for now, please re-open if I've broken anything....

closed 2020-01-10 by hoglet67