hoglet67/RGBtoHDMI #134: Multiple Pi version support

IanSB, 2020-01-28:
I've been doing some work on improving the support for multiple Pi versions. This was mainly because the zero is often out of stock so it would be useful if it worked with the Pi 1 models A & B, especially as the Pi 1 model A seems to be well stocked by RS & Farnell at a reasonable price:
https://uk.rs-online.com/web/p/processor-microcontroller-development-kits/8332699/

The main issue is that the cpu clock is used for measuring the time on many things which wouldn't work at frequencies other than 1000Mhz (1nS) so I've made changes that auto adjust the measurements depending on the CPU speed. (My Pi1s wouldn't overclock to 1000Mhz)

I also found the PLLs weren't being calculated properly and are actually running at twice the speed we thought originally due to the prediv bit so the PLL code has been rewritten to take that into account. More info in this linux source code here:
https://github.com/torvalds/linux/blob/43570f0383d6d5879ae585e6c3cf027ba321546f/drivers/clk/bcm/clk-bcm2835.c
This also confirms the usage of the PLLs:
PLLA is the auxiliary PLL, used to drive the CCP2 (Compact Camera Port 2) transmitter clock.
PLLB is the CPU clock
PLLC is the core PLL, used to drive the core VPU clock and the UART etc
PLLD is the display PLL, used to drive DSI display panels. (not HDMI)

Along the way I found out why you couldn't get PLLA running on the Pi2 and Pi3 because the CM_PLLA register was defined as an absolute address rather than an offset from PERIPHERAL_BASE so that was only valid for the Pi 0 & 1.

As I had a Pi 4 I also did some work to get that running and it now boots up and displays output but it can't read the SD card or the HDMI registers (I assume the new video core has different register locations) so it can't load profiles or genlock but other than that it works with the default BBC values and doesn't glitch in mode 7 like the Pi2 & 3. I also did some work to improve mode 7 on the Pi2 & 3 and they are much better but still not perfect.

Config.txt now has entries for all Pi versions and you can change CPU/core and sdram speeds.

If you want to try it out, it's currently in the "multipi" branch on my Github:
https://github.com/IanSB/RGBtoHDMI/tree/multipi

hoglet67, 2020-01-28:
Hi Ian,

Some nice detective work there....

I do think we need to think very carefully about whether we want to officially support anything other than the Pi Zero. Multiple Pi Models has made release testing of PiTubeDirect an absolute nightmare.

Even the Pi 1B, with the same BCM2835, adds some complexity.  For example, the GPIO assignment  on the 40 pin connect changed between version 1.0 and version 2.0 of the Pi 1B.  What's the hardware revision number of your Pi 1B? (Reported during startup in debug mode).

Pimorani seems to be better at holding stocks of the Pi Zero 1.3 than anyone else:
https://shop.pimoroni.com/products/raspberry-pi-zero

Also, the Pi Zero WH is not limited to one per order.

Finally, this site is very useful:
http://www.thepilocator.com/

IanSB, 2020-01-29:
> What's the hardware revision number of your Pi 1B? (Reported during startup in debug mode).

BOARD_REVISION : 00000010

>For example, the GPIO assignment on the 40 pin connector changed between version 1.0 and version 2.0 of the Pi 1B.

Do you have a reference for that, I couldn't find any mention on the official page:
https://www.raspberrypi.org/documentation/usage/gpio/
In any event even if we don't officially support other boards, the above changes mean that the pi zero can be overclocked or even underclocked (to reduce temps)

> Pimorani seems to be better at holding stocks of the Pi Zero 1.3 than anyone else:

I noticed that all UK suppliers were out of stock for about a month before Christmas.

>I do think we need to think very carefully about whether we want to officially support anything other than the Pi Zero.

I think the only practical alternative would be the model A or 3A as they have the same width as the Pi zero but twice the depth so it might be possible to make a larger PCB with analog support and proper connectors. However neither of them work properly at the moment. The model A needs a mild overclock for glitch free operation and the 3A will suffer from the same issues with mode 7 as the 3B although I have an idea that might eliminate the mode7 problems with the pi2 & 3. 

[image]

hoglet67, 2020-01-29:
> BOARD_REVISION : 00000010

OK, that's a Pi 1 Model B+

> > For example, the GPIO assignment on the 40 pin connector changed between version 1.0 and version 2.0 of the Pi 1B.
> 
> Do you have a reference for that, I couldn't find any mention on the official page:
> https://www.raspberrypi.org/documentation/usage/gpio/

The GPIO pinout change was between version 1.0 and version 2.0 of the Pi 1 Model B. This was in my mind as we've just been doing lots of PiTubeDirect testing at ABUG.

These are of course the original 26-pin GPIO headers, so not at all relevant for RGBtoHDMI.

The B+ (which you are using) is the first with a 40-pin GPIO header, and all of these have the same pinout.

Sorry for the confusion!

IanSB, 2020-02-09:
Closing this as multi pi support merged back into dev

closed 2020-02-09 by IanSB