hoglet67/RGBtoHDMI #138: Advanced Deinterlacing Broken on CPLD v2.0-v2.4

hoglet67, 2020-02-05:
Ian,

I've been testing with the Model B, and going through the different CPLD versions that might be in the field.

I've found that the Advanced Deinterlacing doesn't work any more for CPLD 2.x, even though this supports delay, and this is correctly set during calibration.

I tested this a few weeks ago, so I'm pretty sure it's a recent issue.

Looking at the recent commits to capture_line_mode7_4bpp.S, I noticed that in the Multi Pi work you have changed the advanced deinterlacing to always use the PSYNC_EDGE_FAST routines.

I imagine this is the cause, as the double PSYNC read is needed with CPLD 2.x.

Here's a direct link to that change:
https://github.com/hoglet67/RGBtoHDMI/commit/e5eae550ce4aa246ae33dbd7011dcec4b7a4b8f1?w=1#diff-b5ae5553aac9543435414c27e764cfdfL234

Dave

IanSB, 2020-02-05:
I changed it to save some cycles as I thought all the old V1/V2 cplds didn't support advanced deinterlace so it wouldn't matter but obviously that was wrong.
It should be possible to change back to the previous WAIT_FOR_PSYNC_EDGE.

hoglet67, 2020-02-05:
OK, thanks Ian.

I'll change that back, re-test, then push later this evening.

hoglet67, 2020-02-05:
Right, I've changed it back, and the PSYNC related sampling jitter is gone.

However, there is still a minor issue with the advanced deinterlacer on CPLDs v2.x (where the code is now making two PSYNC reads). Every few seconds there will be one line that jitters.

I'm not too fussed about fixing this. Anyone that's really bothered can find a way to upgrade their CPLD.

In the appropriate v1.x and v2.x profile(s), I've changed the default deinterlacer to Simple Motion 4.

Is Simple Motion 4 the best fallback to use here?

Dave

IanSB, 2020-02-05:
> Is Simple Motion 4 the best fallback to use here?

Before advanced deinterlace I used to use simple motion 2 as that was the best compromise between weaving artifacts and "fizzing" on chars when motion stops. (The number indicates the number of fields of "fizzing").

commit 27e3f886 (2020-02-05): Fix #138 (Advanced Deinterlacing Broken on CPLD v2.0-v2.4)

Change-Id: I89186960b46f496e833011383a148c69f0a9f2b7
closed 2020-05-04 by hoglet67