hoglet67/RGBtoHDMI #139: Problems with old BBC cpld versions

IanSB, 2020-02-06:
Testing on a Master 128

I'm getting really bad glitches with CPLD v2.3 in mode 0:
[image]
They seem to be jumping 1 mode0 char / 8 pixels

The horizontal position of the v24 profile in mode 0 is wrong

V30 to V50 show no image at all with very unresponsive menu operation in all modes
V60 to V63 have vertical jumping in mode 7

Also the v30-v62 profile has wrong horizontal offsets for all modes

I think these were all working when I did the initial split into BBC and OBBC

hoglet67, 2020-02-06:
This is with a Pi Zero?

And with a clean SD card + a release.sh build?

hoglet67, 2020-02-06:
I've just done release.sh build, wiped my SD card, and reinstalled.

I can't reproduce any of these issues.

Dave

IanSB, 2020-02-06:
OK I think I figured out the problem:

 I'm using an early 6 bit board which can only run the 3 bit firmware. That behaves itself with current firmware but older firmware inverts the sync when the field sync input is low which it is by default on the 6 bit board. The 3 bit board has field sync high by default. so the sync is getting inverted on a number of old firmware builds with the 6 bit board.
I should be able to fix it by tying the field sync input high.

IanSB, 2020-02-06:
Tying vsync high has fixed it.

hoglet67, 2020-02-06:
Phew!

closed 2020-02-06 by IanSB