hoglet67/RGBtoHDMI #140: Release candidate

IanSB, 2020-02-06:
I've just done some testing with the PC and that seems to be behaving itself.

Do you want to make up a release candidate zip and I'll do a run through of all the machines again and if that looks good we can go for a release on the weekend.

hoglet67, 2020-02-06:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200206_53bd391

hoglet67, 2020-02-06:
My only thought was whether we wanted to prune CPLD versions that are included. I'm in two minds about this, as it's incredibly useful to us to be able to go back and forth.

Plus it's further work....

I fully expect we'll see issues, and need to make other changes over the next few weeks.

hoglet67, 2020-02-06:
One thing I need to check before we announce is that .jed files in the various vhdl directories reflect the latest version. I'm pretty sure some of them are stale.

I'd also like to delete the .xsvf files in the vhdl directories, as these duplicate the ones in the src/scripts/cpld_firmware directory.

Dave

IanSB, 2020-02-06:
> My only thought was whether we wanted to prune CPLD versions that are included. I'm in two minds about this, as it's incredibly useful to us to be able to go back and forth.

I think that might be confusing for new users especially when they are expected to switch between the YUV and RGB cplds on a semi regular basis.

How about putting them in "old" sub-folders in each cpld folder. They wouldn't show up normally in the menus but could be moved out into the main folder if required or you could make it so they showed up by enumerating the "old" sub-folders if debug was enabled in the settings menu.

BTW spoke too soon about the PC, I've just fixed a sync bug that was causing auto switch to get stuck sometimes. I'll send a pull request soon.

IanSB, 2020-02-06:
I've implemented the "old" folder for cplds when debug enabled.

hoglet67, 2020-02-08:
New release candidate:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200208_dda4935

This morning I'm really just made 2 changes:
- create a RGB CPLD v72 that removes the swap bits functionality as suggested
- in CPLD Recovery Mode, automatically bring up the menu, and add further help/warnings

IanSB, 2020-02-08:
A few comments so far:

All the txt files (Profiles & Config.txt etc) are in unix LF format. Did you change that deliberately or was it done via Github? (I get warnings about eol conversions on my windows Github client)
Realistically, most users will be using windows to look at/ edit the txt files so maybe it is better to release in Windows CR/LF format so they can be edited with notepad.

After doing some tests, I think the UK101(TTL) and ZX80(TTL) profiles can be deleted from the RGB(TTL) profiles folder as I realised they will work at TTL levels with the YUV CPLD using the existing monochrome YUV profiles and at least for the UK101 this is preferable as that needs leading edge hsync to eliminate the drift/jitter of the hsync monostable on the trailing edge. (No leading edge option the the RGB CPLD)

They could also be removed from the BBC profiles but there is no alternative YUV option there.

BTW I just damaged another CPLD while testing the UK101. As usual the UK101 was playing up so I lifted the lid to flex the board and the end of an unconnected BNC cable was disturbed and dropped onto the board right on top of the pickup point for the TTL signals I was using for testing. Not really sure what happened but that killed the CPLD. I was powering the converter from the UK101 so there was no powered to unpowered issue.

However I realised I still had the vsync connected straight to +5V on the bbc 6 way TTL connector which I did to fix the issue with old CPLD testing. 
Having CPLD pins tied directly to +5v is probably the cause and may have been the issue with the previous failure as well. (i.e. it works normally but under any adverse conditions it may be the cause of the failure)

hoglet67, 2020-02-08:
> All the txt files (Profiles & Config.txt etc) are in unix LF format. Did you change that deliberately or was it done via Github? (I get warnings about eol conversions on my windows Github client)

Almost all the files in the git repository have unix LF endings, and I'm pretty sure they always have done.

Probably what is happening is that your windows git client is locally converting text files to windows CRLF endings on check out, and back to LF unix line endings on check in.

Can you do:
```
git config -l
```
And see if there is a core.autocrlf property set, and what value it's set to.

> I get warnings about eol conversions on my windows Github client

When do you get these warnings exactlyI'll?

What I've done is update the release script to force the line endings to CRLF line endings in the next release candidate.

> However I realised I still had the vsync connected straight to +5V on the bbc 6 way TTL connector which I did to fix the issue with old CPLD testing.

Yes, having an input directly connected to +5V without a resistor can cause latchup.

This is because:
- the +5V supply will probably come up ahead of the +3V3 supply
- without a resistor, large amounts of current can flow (>200mA is needed to trigger latchup)

> After doing some tests, I think the UK101(TTL) and ZX80(TTL) profiles can be deleted from the RGB(TTL) profiles folder

Is there an argument for keeping them (even if they are sub-optimal) as it may minimise the CPLD reprogramming for someone who mostly uses the Beeb, occasionally uses a UK101? I'm thinking of BigEd specifically here.

hoglet67, 2020-02-08:
I meant to ask, are you able to rework the boards with the dead CPLDs?

And did the RGB_CPLD_v72 sort the issue for you in the 24MHz Mode?

IanSB, 2020-02-08:
> I meant to ask, are you able to rework the boards with the dead CPLDs?

Yes, I have a hot air station and have replaced several cplds over the past year although I didn't rework the first of the two recent failures as it was an issue 1 6 bit which could only run the 3 bit cpld. That was the only one I built before moving on to the issue 2 and it's obsolete so I stripped it for parts to use on another issue 2 instead.

> And did the RGB_CPLD_v72 sort the issue for you in the 24MHz Mode?

Looks like it has fixed it so far...

> When do you get these warnings exactly?

When pushing to github it mentions CR/LF to LF conversion in a dialog box and git config -l gives:
core.autocrlf=true

> Is there an argument for keeping them (even if they are sub-optimal) as it may minimise the CPLD reprogramming for someone who mostly uses the Beeb, occasionally uses a UK101? I'm thinking of BigEd specifically here.

I was thinking about user confusion as it looks like you should use that one for TTL output rather than the YUV one. They could stay in the BBC profile as there is no alternative.
Also getting TTL out of a UK101 isn't that easy as it isn't present normally. The required outputs are actually open collector driving the composite video resistor network.
To get TTL I had to solder a normal 74LS00 on top of the open collector one with it's inputs connected to the other chip and it's outputs bent out for pickup of the TTL signals.
(This mod is required for the Superboard as well I think)

hoglet67, 2020-02-08:
> When pushing to github it mentions CR/LF to LF conversion in a dialog box and git config -l gives:
core.autocrlf=true

I think this is how most windows git clients configure themselves by default. There are pros and cons with each way of doing this. I think it's actually working well, and we're not seeing bogus conflicts or lots os white-space only channged in the commits. So I wouldn't change anything.

It's easy to change the line endings in the release.sh file, so this is what I've done.

> Also getting TTL out of a UK101 isn't that easy as it isn't present normally.

OK, I see now that makes sense to delete the UK101 TTL profiles in both places I think. I'll do that now.

Is the same true for the ZX80?

IanSB, 2020-02-08:
> Is the same true for the ZX80?

Not really as it doesn't need leading edge and the TTL signals are available so the RGB or YUV versions should work equally well.

hoglet67, 2020-02-08:
Ian,

There is a set of files in src/scripts that I don't understand:
```
-rw-r--r--  1 dmb dmb   11623 Feb  6 17:56 'OLD cmdline.txt'
-rw-r--r--  1 dmb dmb    8368 Feb  6 17:56 'OLD config.txt'
-rw-r--r--  1 dmb dmb     311 Feb  6 17:56 'pc timings.txt'
-rw-r--r--  1 dmb dmb      13 Feb  6 17:56  profile_Atom.txt
-rw-r--r--  1 dmb dmb      18 Feb  6 17:56  profile_BBC.txt
-rw-r--r--  1 dmb dmb      26 Feb  6 17:56  profile_OBBC.txt
-rw-r--r--  1 dmb dmb      20 Feb  6 17:56 'profile_RGB(Analog).txt'
-rw-r--r--  1 dmb dmb      18 Feb  6 17:56 'profile_RGB(TTL).txt'
-rw-r--r--  1 dmb dmb      13 Feb  6 17:56  profile_YUV.txt
```
Do these need to be included in the release package, or are type obsolete?

There is also a ReadMe.txt there that is woefully out of date, and probably should be removed to avoid confusion.

IanSB, 2020-02-08:
The OLD files were from before I created the profile system and can be deleted now.
pc_timings was just a reminder about timings used by auto switching.
The profile_* files are what is saved to the root of the SD card when you change each CPLD from it's default profile. They don't need to be included in the release unless you want to make the default in each CPLD something other than the default built into the code.
The readme either needs to be updated or removed.

hoglet67, 2020-02-08:
OK, I'll tidy those up.

Just to be clear, the profile_* files can be deleted from git as well?

I'll delete the ReadMe.txt as I'd prefer all the documentation to be on the wiki.

I'm also wondering if we need to a Profile validation script.

For example, extracting the values of the palette= property, we get:
```
     21 0
      2 1
      2 Atom_MKII_Plus
      2 Colour_Genie_N25
      2 Dragon-CoCo_Full
      4 MDA-Hercules
     10 Mono_(2_level)
      1 Mono_(4_level)
      6 RGB
      5 RGBI_(CGA)
      1 RGBI_(Spectrum)
      1 RGBrgb_(Amstrad)
      1 RGBrgb_(Spectrum)
     14 RrGgBb_(EGA)
      1 Spectrum_48K_9Col
```
Is the numerical form still valid, or do these need to be changes to RGB and RGBI.

The palette=0 values are all in the Colour Test files. The palette=1 values are in the QL profiles.

I was also going to add some Atom CPLD firmware, as I think at least one person with the atom board has done the mod. I won't add it to the recovery folder, as the potential for bricking is too high.

IanSB, 2020-02-08:
>Just to be clear, the profile_* files can be deleted from git as well?

Yes I think so.

>Is the numerical form still valid, or do these need to be changes to RGB and RGBI.

No, it's not valid, they should be changed as the numeric values won't work and I think the QL should be RGB not RGBI anyway as it only has 3 bit RGB out like the BBC.

Actually just remove all the palette=0 & 1 lines because the default is RGB anyway, not sure why they were added in the first place. (The 6 RGB entries above are probably the default files)

IanSB, 2020-02-08:
I'm getting crashes on the Pi zero W
It boots OK but crashes when the menu button is pressed with Data Abort at 02101750

IanSB, 2020-02-08:
OK Can't reproduce this now. I was getting no HDMI output so it might have been screwed up HDMI registers due to bad HDMI contact. I reseated the HDMI connector and it's working for now.

IanSB, 2020-02-08:
Yes that was it, if you boot without a valid hdmi connection you will get a crash.
I think the code assumes sensible values for the HDMI registers and doesn't handle invalid values like 0 very well.
This isn't a show stopper for now but maybe we should handle no HDMI connection a bit better in future.

hoglet67, 2020-02-08:
I just tested self-programming on a modded Atom board, and it works fine.

There is a photo of the mod wires here:
https://github.com/hoglet67/RGBtoHDMI/issues/97#issuecomment-503212565

The only issue was if you start with a blank CPLD you are thrown into the CPLD Recovery Menu, which doesn't have an Atom image.

I'm rather nervous about adding an Atom image to the CPLD recovery menu, because you can badly brick a BBC board by programming the Atom image (and probably visa versa). I think any future Atom builders that are relying on this for bring-up will have to manually copy the image. Do you agree with this?

Are you ready for another release candidate, or shall I do that first thing tomorrow morning?

I'm not planning on anything else this evening.

Dave

IanSB, 2020-02-08:
> Are you ready for another release candidate, or shall I do that first thing tomorrow morning?

Yes I think so, I'll run it through all my machines this evening.

BTW the UK101 is a lot more stable with leading edge sync, it used to need recalibration every 30-60 mins with trailing edge.

hoglet67, 2020-02-08:
Here you go:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200208_55a148e

IanSB, 2020-02-09:
> I think any future Atom builders that are relying on this for bring-up will have to manually copy the image. Do you agree with this?

Yes, definitely not a good idea to put it in the folder by default.

>Here you go:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200208_55a148e

I've tested this latest release candidate with all my machines and not found any issues. I also programmed the 6 bit cpld into a 3 bit board and it successfully detected the wrong CPLD and brought up the recovery menu and there was no damage to the CPLD.

I also briefly tested on a Pi 1A+, Pi3 and Pi4 (The Pi4 is not fully working yet due to SD and HDMI issues but it does give a default 1080p picture with the Beeb)

There are a few untested profiles:
QL
Atom50Hz(TTL)
Nascom II
ZX80 composite
ZX81 composite

The first two I don't have. Most of their settings should be correct but the H and V offsets will need to be finalised.
I have a Nascom but it only has UHF out so I will have to mod it for TTL and composite. 
ZX80 and ZX81 composite require converter boards, which I have, but not yet fitted. The timing for those profiles will be correct but the analog levels will need to be set. (Not really an issue at the moment as no-one except us has analog boards and there will probably be another pcb issue and/or release before that is finalised)

One minor thing:
Mux is not selected on the 3 bit Electron profile so it is using the standard inputs.
However I think it is a good idea to leave that in at least for this release to see if there are any issues reported because the 6 bit doesn't have the option anyway and we will get a good idea if that will be a problem in future.

BTW it looks like PCB way will build 20 boards for $30 plus parts and they use all the usual suppliers like Farnell & Mauser.

hoglet67, 2020-02-09:
> Atom50Hz(TTL)

I'll try to test this today.

IanSB, 2020-02-09:
> I think any future Atom builders that are relying on this for bring-up will have to manually copy the image.

You could create an "old" folder in the recovery folder and put the Atom image in that. Then selecting Debug in CPLD recovery mode will show the Atom image. 
I just tried this and it works OK.

hoglet67, 2020-02-09:
Update on the Atom50Hz(TTL) profiles.

Hmmm, I've forgotton how awful the Acorn colour card is, which is what started me off down the path to Atom colour perfection. It's taken me quite a while to get the card working again, because I had done a couple of mods that needed reversing, and I also needed to make up a new cable and solder this in.

I've had it running for a couple of hours now, and it's very far from perfect. The main issue is there is no Y level clamping, and the threshoulds for video and sync are set with presets. It's pretty much impossible to get it working well in all modes.

There is also a significant glitch (~75ns) on the csync signal after some of the hsync pulses. This is way longer that our glitch filter will cope with, so it gets through.
[image]
[image]
You can attempt to tweak this out by adjusting the sync threshold preset, but it drifts:
[image]
I'm now of the view that we shouldn't be claiming to support this card, at least not in this release.

What I'm thinking of doing is renaming these profiles to remove the 50Hz bit, and re-use them as profiles for Phill's Colour Board, which works far better. This is the board that most people have, and there is at least one person (trixster) that is using this with the original RGBtoHDMI board.

Dave

IanSB, 2020-02-09:
> What I'm thinking of doing is renaming these profiles to remove the 50Hz bit, and re-use them as profiles for Phill's Colour Board, which works far better. This is the board that most people have, and there is at least one person (trixster) that is using this with the original RGBtoHDMI board.

Sounds like the best option.

hoglet67, 2020-02-09:
The Atom profiles in OBBC/BBC/RGB_(TTL) are all updated now, and I have tested every CPLD version. Only v6.0 is a bit off centre, and that's not a version that will be out in the field.

The only issue I've found is that with the 6-bit board (without the mux) the orange semi-graphics end up red, and are a bit noisy. This is expected, as the green level is 50%, which ends up right on the threshold of the CPLD input. Fitting of appropriate value termination resistors helps.

I also renamed the ZX80_(TTL) profile to ZX80 as it was the only profile in the OBBC/BBC/RGB_(TTL) folders that still had the redundant _(TTL) suffix

There's a new release candidate here:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200209_3b8d96b

I'm done I think now.

IanSB, 2020-02-09:
> I'm done I think now

OK I think we can go with that as the release version.

I've been working on the wiki, currently the reference guide and then the tutorial, perhaps you could add the BOM for the 6 bit board as I sent you the info when you were building yours.

hoglet67, 2020-02-10:
> I've been working on the wiki, currently the reference guide and then the tutorial, perhaps you could add the BOM for the 6 bit board as I sent you the info when you were building yours.

I've done my best on the BOM for the 6-Bit board and Analog Frontend:
https://github.com/hoglet67/RGBtoHDMI/wiki/Bill-of-Materials-%286-Bit-Board%29
https://github.com/hoglet67/RGBtoHDMI/wiki/Bill-of-Materials-%28Analog-Board%29

Do check these over if you get a change.

Re: the analog board, some of the parts listed on the schematic are different to what we both have ended up building, so it's probably worth updating the schematic.

Were you planning a further revision of this board prior to release, to address the CL1 track cut / patch wire. 

I think you also mentioned:
> One other change I intend to do for the next issue is to add a schottky diode in the +5v supply from the picoblade connector. That way a standard beeb 6 way DIN can be on the other end of the picoblade connector and it could still be plugged into an Amstrad without causing damage as that has mono video on the +5v pin.

IanSB, 2020-02-10:
> Do check these over if you get a change.

I changed the resistor networks in the 6 bit board to 4.7K to get more safety margin with the Electron but hadn't updated the schematic so I've sent a pull request for that and also edited the wiki.

I also added a screencaps gallery with caps made with screencap scaling set to "on" so they are 1920x1080. Perhaps you could add some more caps from the Atom and Superboard and anything else you have that works with it.

I know it's a bit late but now we are now referring to the boards as 3bit and 6bit in the documentation would it make more sense to change the profile folders and firmware names to match? (This may cause some truncated lines due to increase in length such as the confirmation of firmware update)

i.e. something like
BBC > 3BIT_RGB
OBBC > OLD_3BIT_RGB
RGB(Analog) > 6BIT_RGB(Analog)
RGB(TTL) > 6BIT_RGB(TTL)
YUV > 6BIT_YUV
and maybe prefix the firmware in the same way?

hoglet67, 2020-02-11:
> I know it's a bit late but now we are now referring to the boards as 3bit and 6bit in the documentation would it make more sense to change the profile folders and firmware names to match? (This may cause some truncated lines due to increase in length such as the confirmation of firmware update)

I'd rather get this right, and  I'm inclined to agree, the more constistent we are with naming things, the less confusing it is for end users.

Concerning the profile directory names:
```
BBC > 3-BIT_RGB
OBBC > OLD_3-BIT_RGB
RGB(Analog) > 6-BIT_RGB(Analog)
RGB(TTL) > 6-BIT_RGB(TTL)
YUV > 6-BIT_YUV
```
Firstly, I would be keen to remove some of the punctuation, since I don't think these are every rendered directly in the UI, and these end up needed escaping on Linux which is awkward:
```
3BIT_RGB
6BIT_RGB_TTL
6BIT_RGB_Analog
6BIT_YUV
Atom
OLD_3BIT_RGB
```
Secondly,  I question the current use of the _TTL and _Analog suffixes, where they they are only included where they are necessary for disambiguation. I'm wondering if we should be more explicit:
```
3BIT_RGB_TTL
6BIT_RGB_TTL
6BIT_RGB_Analog
6BIT_YUV_Analog
Atom
OLD_3BIT_RGB_TTL
```
But this is all starting to feel a bit clumsy, so lets try dropping _TTL completely:
```
3BIT_RGB
6BIT_RGB
6BIT_RGB_Analog
6BIT_YUV_Analog
Atom
OLD_3BIT_RGB
```
What I like about this is it's now pretty clear which of these apply when you have the optional analog frontend.

And finally, whether to rename OLD_3BIT_RGB:
```
3BIT_RGB
6BIT_RGB
6BIT_RGB_Analog
6BIT_YUV_Analog
Atom
Legacy
```
Sorry, I'm not being very decisive this morning!

We also present this to the user in the form of interface names, where we turn things about a bit.
```
static const char *frontend_names[] = {
   "TTL RGB (3 bit)",
   "Atom 6847",
   "TTL RGB (6 Bit)",
   "Analog RGB/CVBS (5259)",
   "Analog RGB/CVBS (UA1)",
   "Analog RGB/CVBS (UB1)",
   "YUV (5259)"
};
```
I don't think these necessarily should be consistent with the profile names, they can be longer, and more human readable, but they should probably convery the same information.  So I might be inclinded to change "YUV" to "Analog YUV".

Another question is whether we want to drop support for UA1/UB1, which might then allow the UA1/UB1/5259 to be dropped. (I stripped down my original v1 analog board for parts yesterday). A counter argument is that even if we do drop UA1/UB1, there may be other DACs used in the future.

I've got visitors arriving shortly, so I won't actually be able to do very much today.

I do think now is the right time to get this sort of thing right; it's a low risk change that doesn't require lots of retesting. And we're in the midst of writing documentation, so that's fluid as well.

Dave

IanSB, 2020-02-11:
> I'd rather get this right, and I'm inclined to agree, the more constistent we are with naming things, the less confusing it is for end users.

I've made some of your proposed changes to profile folders + cpld names although I ended up using "Legacy_3BIT" to emphasize which board used those profiles.
I also fixed a hang with h and v aspect set to zero.
(See recent pull request for commits)

Still looking at dropping the old analog board support.
Also I think you wanted to drop the return at end setting from the menu?

hoglet67, 2020-02-12:
I've merged, made a couple of further minor Atom profile changes, and created a Release Candidate 4:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200212_e7f3716

I've also added a few Atom screen shots to the Gallery.

One thing I noticed just now when re-testing some of the legacy atom profiles (specifically CPLD 1.0) is that the horizontal centreing seems to be resolution specific. If the profile is centred at 1600x1200, it's offset a bit horizontally at 1920x1080. I don't see the same effect with the non-legacy Atom profile. I'm not too concerned about this if it only affects the lagacy profiles. The only thing that's different is that this profile uses the Half-Even sampling mode. I'm guessing when the capture size is being adjusted, this is causing a centering error to occur. If this raises any alarm bells with you, then I can create an issue and add some more information.

Dave

hoglet67, 2020-02-12:
> Also I think you wanted to drop the return at end setting from the menu?

I'd like to keep it as an option in the config file. The problem when you change it from the menu is that it then only affects the current profile. If you then end with some profiles with it enabled, and some with it disabled, then changing profiles gets very confusing (try it!).

I can't remember, did we have the notion of global properties, that when change are associate with the default scope rather than the profile scope? If so, then the return position should be one of these.

IanSB, 2020-02-13:
> I can't remember, did we have the notion of global properties

We decided to have default.txt as only manually editable at the time.

One thing that occured to me would be to split the contents of the preferences and settings menu into always global options and always per profile options and re-arrange the menus accordingly, calling them settings and global settings or something similar.
e.g. scanlines, debug, screencap scaling, return, font size and maybe the genlock stuff could be global and the rest per-profile. The global settings could be saved in a custom_default.txt file to override the default.txt

This is quite a radical change so I don't think it's suitable to do right now but it could be implemented in the next release. Ideally I'd like to get this release out by the weekend. There's going to be another one in a month or two anyway with the in band and ntsc artifact code finished so we don't really have to finalise everything now.

I've finished adding descriptions for all menu options in the reference guide.
I wasn't sure exactly what the Half setting in the sampling menu does so perhaps you could add that and maybe proof read them and edit / add anything extra you think is needed. The button options and button configuration also needs to be documented somewhere so can you look at that.

While documenting, I noticed we use R-Y and C &  L for some menu options and YUV elsewhere. Would it make sense to change these to be consistent in YUV and Atom menus. 

Also I noticed that the Dragon, Tandy and Spectrum profiles had green/white borders in 1080p mode (not noticeable in 1600x1200 mode). This is because the Y is inverted in those profiles and it's fixed by altering their border colour settings which  I have done on my branch.

I also added some more screencaps but they are just placeholders for now and should be replaced with something more graphically interesting.

hoglet67, 2020-02-13:
> We decided to have default.txt as only manually editable at the time.

That's actually fine - I'm probably the only person who would want to change this.

> I wasn't sure exactly what the Half setting in the sampling menu does so perhaps you could add that and maybe proof read them and edit / add anything extra you think is needed.

I've just added some text for that.

> While documenting, I noticed we use R-Y and C & L for some menu options and YUV elsewhere. Would it make sense to change these to be consistent in YUV and Atom menus.

Yes, it probably does make sense to make these more consistent, for example:
```
C Filter
L Filter
Subsample C
R-Y PAL switch
```
would change to:
```
UV Filter
Y Filter
Subsample UV
PAL switch
```
>  Ideally I'd like to get this release out by the weekend. 

Sorry, this might have to slip to next week, as we're away for a long weekend break, starting later today.

Dave

hoglet67, 2020-02-13:
> We decided to have default.txt as only manually editable at the time.

That's actually fine - I'm probably the only person who would want to change this.

> I wasn't sure exactly what the Half setting in the sampling menu does so perhaps you could add that and maybe proof read them and edit / add anything extra you think is needed.

I've just added some text for that.

> While documenting, I noticed we use R-Y and C & L for some menu options and YUV elsewhere. Would it make sense to change these to be consistent in YUV and Atom menus.

Yes, it probably does make sense to make these more consistent, for example:
```
C Filter
L Filter
Subsample C
R-Y PAL switch
```
would change to:
```
UV Filter
Y Filter
Subsample UV
PAL switch
```

IanSB, 2020-02-16:
I've been doing some work on the V3 analog board and it's software support.
The results are currently in the "gpio" branch on my fork.

I've been looking at the possibility of using GPIO 0 & 1 (SP_DATA_PIN & SP_CLKEN_PIN) to control the analog switches that switch in the termination resistance. These are obviously used to control the cpld and the DAC but they only 'blip' briefly and can be set to any logic level when not being used. Using an RC filter to filter out the 'blips' works and you end up with a steady logic level representing the termination state.
Also as there are two pins, separate termination control can be used on the Y and UV signals.
(e.g. The Atom 'Y' out should be terminated but not the U & V)

This also has the advantage of freeing up one DAC to use with a possible enhanced version of the interface which might work with the TI99/4a and MSX computers.
 
There is however a slight problem.
GPIOs 0 & 1 are reserved on the Pi for an I2C identity chip so that Pi's can auto configure and it looks like bootcode.bin either re-enables pullups or drives those lines high briefly during bootup which causes termination to be briefly enabled during reset which is not a good idea on weakly driven YUV outputs.

This could be overcome with a very large RC time constant to filter out the large blip on bootup but another alternative would be to use inverted signals and use normally closed analog switches so the gpios going high during bootup would fail safe by turning off termination.

This however would require updated CPLDs with inverted clamp outputs although this is not a bad idea as when the clamp output is floating during version or vsync reading, it is being pulled high externally and that currently turns on the clamp for long periods but with normally closed switches that would turn off the clamp.

The only issue I can see with inverted signals/normally closed switches is that when the CPLD is reprogrammed the two GPIO lines are mostly low when the clock is not active so termination gets enabled during cpld reprogramming and I don't know if it is possible to set those lines high when not actively clocking in data similar to the DAC and CPLD.

defs.h has a #define TERMINATION_INVERTED to switch modes for testing.

It should be possible to mod the V2 boards to use this by changing the analog switch (if the normally closed option is used) and adding the two resistors and capacitors and a couple of wires.

hoglet67, 2020-02-17:
Ian,

There's a new  release candiate here:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200217_19cc272

Regarding the use of GPIO 0 & 1, another thing to test is what state these are in if there is no SD Card present at all.

I think you should try to avoid multiple versions of the analog board in the field, because the interface doesn't have any way of supporting versioning. I'm happy to mod my V2 boards (I have 2 built up), or strip them for components. So if you want to include this termination change in the V3 version, then go ahead and do that. Looking at the situation in China, most SMT Assembly lines seem to have shut down, and only 2 layer boards are being made. So there may be a considerable delay anyway.

We can still make this release public, for people who have existing boards, or who just need the TTL interface.

Dave

hoglet67, 2020-02-17:
Did you look at the AD8802/AD8804 parts - these are 12 channel DACs:
https://www.analog.com/media/en/technical-documentation/data-sheets/AD8802_8804.pdf

Available in a SOIC-20 or TSSOP.

RS have them in stock:
https://uk.rs-online.com/web/p/general-purpose-dacs/8068073/

IanSB, 2020-02-17:
> what state these are in if there is no SD Card present at all.

The pullup / down state is non volatile so without a card on power up it will be pulling or floating according to the last programmed state.

> I think you should try to avoid multiple versions of the analog board in the field

Yes, that's why I'm investigating this idea now, I think I can reliably discriminate the TI 99/4a colours with two extra DACs/comparators using an extended version of the interface (which would likely be external as the extra components might not fit). Making these changes now would keep software and profile compatibility between the two versions although it is a breaking change with the current prototype version of the analog board. The two extra DACs would be the former termination one and the clamp level one so the extended board would have to be DC coupled. (Only the Atom really needs the clamp, all the others should work DC coupled)

I've tweaked the CPLD programming code to make the two GPIOs mostly high so they appear to be a logic 1 after filtering even during programming.  (The code to do that is checked into the gpio branch)

There is still one slight problem: Although a normally closed analog switch that matches the 4066 pinout exists (MAX4611) it doesn't seem to be stocked by the usual sources. All the other freely available normally closed switches use a 16 pin package with different pinout. I have relaid the pcb for that pinout to confirm that it will fit (see recent gpio checkin) but to test this out we will have to dead bug an inverter package on the board to invert the two GPIOs and the clamp signal, feeding the inverted outputs to the control lines on the existing analog switch.

IanSB, 2020-02-17:
> Did you look at the AD8802/AD8804 parts - these are 12 channel DACs:

I hadn't really looked at more DAC channels but the SOIC 20 version is SOIC-W (The wide version so may not fit) and it's £10. The TSSOP is about double the price of the current one.
However I'll have a look to see if there are any other parts that might work.

IanSB, 2020-02-17:
Problem solved I think:
It looks like bootcode.bin only turns on the pullups at boot time rather than actually driving the pins so the weak pullup can be overcome by a strong pulldown (1K) so using inverted signals and normally closed switches are not required which means only minor changes and no CPLD changes.

hoglet67, 2020-02-17:
That's good news Ian.

IanSB, 2020-02-18:
I ended up doing 3 changes:
1. Use GPIO 0 & 1 to control termination via filter networks.
2. swap DAC G & DAC H connections so that the spare is on the end (DAC H) and can be hidden if required.
3. Replace U3 (the analog switch) with a MAX4610. This is pin compatible and has an ON resistance of approx 75 ohms with a 1 volt signal and it's reasonably temperature stable.
(I previously ordered a couple to test)
This means the 75R series resistors aren't required so I replaced them with 10 K pulldowns instead so that the inputs aren't floating when termination is disabled. (It used to glitch when unplugged).

Here's the modded board:
[image]
[image]

Mod details:
On top of the board remove U3 and R4, R5, R6.
Cut the track going to pin 5 of U3 but leave pins 6 and 13 connected
This isolates the G/Y termination control on pin 5 but leaves RB/UV connected together
Cut the track going from C3 to under U4 and also the tracks going to pins 9 & 10 of U4
This disconnects the clamp and spare (old termination control) DAC outputs which isolates the above mentioned pins 6 and 13 as well as pin 10 of U3.

Fit MAX4610 to U3
Bridge R4, R5 & R6 which eliminates the series resistors.
Solder 10K resistors to the above bridges and connect the other sides to Gnd (I used pin 7 of U3)

Solder a wire from Pin9 of U4 to the free end of C3 and then onto Pin 10 of U3
This connects the clamp voltage from DAC:G to C3 and the clamp switch.

On the underside of the board:
Connect 1K resistors from DAT and CLK to Gnd
Connect 10K resistors to the other side of DAT and CLK and then join 10uF capacitors onto the free ends of the resistors.
Connect the free ends of the capacitors to Gnd 
Connect a wire from the capacitor/resistor junction of DAT to pin 5 of U3
This connects the G/Y termination control to the G/Y termination switch
Connect a wire from the capacitor/resistor junction of CLK to pin 6 of U3
This connects the RB/UV termination control to the RB/UV termination switches

I've checked in the updated software to drive these mods into the gpio branch as well as the latest V3 pcb and updated profiles.

I think the only real question remaining is whether having Y only termination is of any real benefit, if not that would save 3 passives and maybe allow that control bit to be used on an enhanced board for something else. 

As a compromise for now, you can't select UV termination only (which would be technically possible) so that bit combination is reserved for some possible future control function.

IanSB, 2020-02-18:
After testing it, I don't thing the separate Y termination is all that useful. I tried terminating the Atom video signal but that resulted in a reduced threshold range due to the reduced voltage and there didn't seem to be any real benefit. The Y signal might have worked better with a longer cable but the U and V would still be unterminated and high impedance so would not really be usable on that long cable.

Anyway I came up with a much better idea to use the other switching signal: I've added another analog switch to switch the Green signal from AC coupling with clamp to DC coupling. The phase shift of the Green signal has always been a concern and it is noticeable on the Amstrad when adjusting the offset and it looks like it is of the order of 1/4 a pixel on 16Mhz pixels although it was still possible to get a good offset value.

At first I didn't think it would fit but then I realised that the passive filter components would not be needed for this signal because when the GPIOs are being used for other things like updating the DAC/CPLD or reprogramming the CPLD, the video signal isn't being looked at so it doesn't matter what state any AC/DC switch is in at that point.

It does matter for the termination as that affects the source and so unwanted termination might cause problems with high impedance sources (It certainly does with my Dragon which corrupts the screen RAM if I terminate the Y output of the 6847)

This also has the benefit that when the setting is in the DC mode the clamp DAC is free to be used for other things like an additional threshold on an enhanced interface.

[image]

IanSB, 2020-02-21:
Well I got a bit carried away and ended up with a new analog design that has everything including the kitchen sink :)

The final V3 design has the 75R resistors squeezed in as well as the 10K pull downs so it can work with the original low resistance TI switch, it has the AC/DC switch as mentioned above, which I have prototyped on my existing analog board and that seems to work OK, and finally it has another 9108 comparator on the underside which gives a minimum of two additional levels of detection for U and V signals using the spare and clamp (when in DC mode) DACs . This requires two more signals to be routed to the CPLD so I have increased the header from 10 way to 12 way and designed a new "8 bit" version of the hat. Both of these boards can be built as the standard 6 bit boards by omitting components and the analog board retains full compatibility with the existing 6 bit hats.
[image]
[image]
[image]

The 8 bit hat has a slight fouling problem between the legs of SW1 and the 12 way header but that can be overcome by cutting the legs of SW1 flush before soldering. Alternatively switching to surface mount switches might be an option.  In theory the new comparator can provide up to 4 extra levels of UV detection but only two can be sent via the header. It might be possible to use all 4 by hand wiring the extras if they were ever needed.

I'll merge the recent small number of changes to support this board back into the dev branch and hopefully that will be it for this release. (The 8 bit modes will require new CPLDs and probably more software changes at some point once we have some new boards to try)

hoglet67, 2020-02-24:
Pull request merged, and another release candidate uploaded:
https://github.com/hoglet67/RGBtoHDMI/releases/tag/20200224_6e0d72e

I've just looked over the new V3 analog design. Looks good to me. The only question I have concerns the schottky diode in the VCC line. The Pi Zero uses a switch mode regulator (PAM2306) so that should be fine with a reduced voltage. But probably worth testing. The Pi 3A+ and Pi 3B+ I think are much more sensitive to undervolting, so may be more problematic.

IanSB, 2020-02-24:
> another release candidate uploaded

Looks good so far, I've updated the wiki to reflect recent changes.

> the schottky diode in the VCC line.

It can always be replaced with a zero ohm link if there are issues or omitted entirely to isolate that pin so I'd rather have the option available.

BTW I just added an alternative connector to the analog board for a six way IDC ribbon cable:
[image]

As the body of the connector is overhanging the pcb it can be pushed through the pcb more than normal to allow for clearance and it also means the ends of the pins act as a support against the camera connector so P6 & P7 don't need to be fitted either. 
It might even be possible to fit a shrouded connector instead of just bare pins as it looks like there might be enough clearance with the overhang. If bare pins are the only option then the diode above is even more important as reversing the connector would reverse the 5V polarity which would probably damage everything connected to it. 
Note this option would not be usable with Pi 1-4 due to fouling issues between the end of the connector and the USB/Ethernet connectors on the Pi so the existing molex could still be fitted if preferred

IanSB, 2020-02-25:
I've created an alternative pcb called "V3 Template" which has 2mm wide alignment tabs on the board (It's in dev on my fork).
Also all the switches and leds have 1mm NPTH in the centre of their locations and currently they are also on the non-template version. In theory if you dropped this pcb on the inside of the plastic case I found previously on ebay it should fit snugly and allow you to drill perfectly aligned 1mm pilot holes in the case for the switches and leds:
[image]

The overall dimensions of the pcb are 0.1mm smaller than the smallest dimensions I got measuring several cases so hopefully it won't bind or be too slack.

There are a couple of issues though:

I put NPTH between the legs of the leds and the design rules checker complains about that so I don't know if that will cause issues with different pcb manufacturers. I know it doesn't cause an issue with JLC as I added a test design without tabs to my last order and it came back OK without any issues:
[image]
 However I have read that some low cost manufacturers don't do NPTH as it requires a second drilling pass and if the led holes were plated they would short the led pins. I tested continuity on several boards to confirm there was no problem.

Also there is a significant  amount of play in the position of the 40 way socket in the PCB. The holes are 1.00mm but the pins are 0.75mm x 0.4 mm. (You can even see that on the 3D model above). So even if the holes were accurate, if the header wasn't exactly dead centre in the holes there would be misalignment in the final switch position. 
There is probably other play with the screw fixings so the tabs themselves might force things back into alignment when the top is fitted or they might jam if the misalignment was too extreme requiring the tabs to be filed off.

Do you think it's worth reducing the hole size in the 40 way connector to reduce the play?
If so what sort of size and can they all be changed at once or only individually?

Also due to the potential problems with the led holes should I remove them from the standard (i.e. non-template) version of the board?

hoglet67, 2020-02-25:
Re: the LED alignment hole, some thoughts:
- I would avoid NPT holes, as some manufactures simply ignore this!
- you could move the pads apart a bit to make room for a PT hole, it will still be easy to fit the LED
- or don't bother with LED alignment holes, and instead make sure the LEDs are exactly inline with the switch centres and a defined distance apart. This is what I did on the Atom Video to HDMI board.

Re: the 40 way connector, I think reducing the play would be useful. A pin of 0.40x0.75 gives a 0.85 diagonal, so reducing the hole size to 0.9mm would help.

Note: with JLCPCB hole tolerance is +/- 0.08mm, so in theory you might get a very tight fit, but given how slack things are at the moment, I think you'll be fine at 0.9mm.

In the KiCad footprint editor you'll need to create a custom footprint by saving the existing footprint to a local project library. You then edit one pad manually, right click/pads/copy pad properties, select all the other pads by draging a rectangle around them, the right-click/pads/paste pad properties.

IanSB, 2020-02-26:
> I would avoid NPT holes, as some manufactures simply ignore this!

Actually I think JLC ignore NPTH as well.
I noticed that the sides of the non plated holes including the four mounting holes were blackened rather than clean fibreglass like the edges of the board (You can see this with the earlier boards as well) so I think the holes were actually plated but then the plating was etched away later because there are no copper pads associated with the holes and thus the etch resist mask so the UV etch resist didn't harden to protect them.

I reduced the holes on the 40 way to 0.9mm and also tweaked the LED pads slightly by moving them apart 0.1mm, reducing their size by 0.2mm and reducing the alignment hole by 0.2mm to 0.8mm. After that the hole no longer overlaps the pads and there is 0.12mm clearance either side. This still isn't enough to stop the rules checker from complaining but it's better than before.
(I'm a bit reluctant to increase the led hole spacing any more as I've been mounting the leds tight to the board and any lead splay due to wider hole spacing would stop that)

I removed the alignment holes and tabs and put the LED pads back to their original size on the standard version.

As mentioned on the forum, I'm thinking of doing a max SMD version with surface mount resistor networks, LEDs and switches to use with any manufacturing run. Using surface mount switches would eliminate the fouling with the 12 way TTL connector and the optional 6 way connector on the analog board but my main concern with that is the alignment accuracy of the switches as they are only resting on surface pads rather than positively held by holes.
Do you have any thoughts on that?

IanSB, 2020-02-26:
I had a much better idea for fitting a 6 way IDC header:
[image]
I used a 10 way header for the above test as I don't have any 6 way ones yet but it's a perfect fit vertically and even provides support against the bottom edge of the case.

IanSB, 2020-02-29:
I've started on the tutorial for creating a new profile. At the moment, it's just how to setup the geometry menu but while writing I noticed the need for some additional helper info in the menus, mainly to do with the pixel aspect ratio so I've added that to the source status and also the helper status line in the geometry menu.
Do you want to add that to the current release candidate?
Also could you look at the tutorial page and let me know if you think anything should be clarified further for the geometry menu.

hoglet67, 2020-02-29:
> Do you want to add that to the current release candidate?
I don't see why not, as it's just additional status info.

> Also could you look at the tutorial page and let me know if you think anything should be clarified further for the geometry menu.
I'll take a look sometime next week.

At the moment I'm deep in a rabbit hole, trying to make the BeebFPGA output true progressive rather than interlaced video in Mode 7.

Dave

IanSB, 2020-03-03:
I noticed you merged the recent pull request, but hold off on a new release candidate as there is another minor improvement to follow with the sync handling which means the sync helper info works properly

IanSB, 2020-03-11:
I've built up an 8 bit board and it works OK (in 6 bit mode) and the software can also detect an 8 bit board is present using the pull down resistor on the data line.
The extra 2 bits can be used in two different ways:
With the analog interface they can be used to provide an extra level on the RB / UV inputs (the vsync input already provides an additional level on the G / Y input). This means full 2 bits per pixel analog support with 64 colours max. 
This mode just requires some minor CPLD changes to include 3 input to 2 bit encoders although those changes probably won't fit in the standard 6 bit CPLDs so some features will have to be stripped out for that to work in 8 bit variants of the CPLDs (perhaps removing 24 Mhz mode 7 support in RGB? and Spectrum 48K in YUV?)
 
In TTL mode the extra 2 bits can provide full 256 colour capture although that will look a bit messy with the menus on an 8 bit frame buffer and scan lines won't work.
Not sure if switching to a 16 bit frame buffer would be feasible but I think the menu issue could be worked around by stripping the top bit off during capture when the menu is on screen.
Again this would require a new 8 bit variant of the RGB CPLD which could capture 8 bits per pixel.

At the moment four pixels are captured at 6BPP over two psync edges.
If the 8 bit pixel data was packed as 8 bits 1st pixel + 4 LSbits 2nd pixel for the first psync edge followed by 4MSbits 2nd pixel + 8 bits third pixel for the second psync edge then you would get three pixels per two psync edges.

As the max pixel clock is ~28Mhz with 4 pixels, that implies a max clock of ~21Mhz with 3 pixels which is still in a usable range.

IanSB, 2020-09-08:
I think this can be closed now as it related to the previous run of release candidates

closed 2020-09-08 by IanSB