hoglet67/RGBtoHDMI #169: [Wish] Integrate FF_OSD

BwanaFr, 2021-01-07:
The [FF_OSD project](https://github.com/keirf/FF_OSD) is used to output Flashfloppy screen to video signal.
Flashfloppy can be used on many computers like Amiga, Amstrad CPC, PC...

The communication between FF_OSD and Flashfloppy is done using an I2C bus.
It can be great to implement this feature in RGBtoHDMI by receiving I2C data from Flashfloppy and outputting OSD on HDMI signal.

How complicated implementing this feature can be ?

IanSB, 2021-01-09:
>The communication between FF_OSD and Flashfloppy is done using an I2C bus.

There are a couple of spare GPIOs but they are not connected to I2C hardware so they would have to be bit banged but the main problem is that the CPU is 100% busy reading video from the GPIOs so can't spare any time to implement I2C receive protocol.
There will be times during H and V sync periods when data could be received but there would have to be some way to arbitrate that.

IanSB, 2021-01-23:
@BwanaFr 

I got that working on c0pperdragon's Amiga version of RGBtoHDMI
See:
https://github.com/c0pperdragon/Amiga-Digital-Video/issues/26

BwanaFr, 2021-01-23:
Thanks a lot! This is really what I needed (I'll keep the BluePill for now).

closed 2021-01-23 by BwanaFr