hoglet67/RGBtoHDMI #170: HDMI screen "pickiness"

Sch-LikA, 2021-01-07:
I just built a small batch of RGBtoHDMI, that is a really amazing project ! First a huge thanks.

At first I thought I screwed up my build since nothing showed on screen. After lots of testing on 7 different screens :

- Dell 2407 over hdmi-to-dvi
- Dell 2410 over hdmi and "hdmi-to-dvi"
- Samsung 23h9x over hdmi-to-dvi
- Nexdock (would be so cool) hdmi
- standalone RTD2660H controller + 11' screen hdmi
- HP LP1965 hdmi-to-dvi
- Dell U2413 hdmi

Only the last two work. At least they work fine :-)

Anything I can do to help support the other ones ?

hoglet67, 2021-01-08:
I can't find a reference on the internet to a samsung 23h9x. Can you provide links to the specs for this?

There seems to be compatibility issues between the NexDock and the Pi:
https://www.raspberrypi.org/forums/viewtopic.php?t=245153

Have you tried booting with SW1 pressed? This should force 60Hz output.

What source are you using?

Sch-LikA, 2021-01-08:
Sorry wrong Samsung model, it is a 2343BW.

I have a "2nd release" Nexdock which works fine with about anything I ever threw at it (Raspberry 0,1,2,3 & 4, MiSTerFPGA, Smartphones). I just tried with SW1 pressed, not better, and tried connecting another laptop on the USB in case the Nexdock used USB to detect activity, no better result.

Just tried with SW1 pressed and got my 2410 working, great. I am now using Auto@50-60Hz (it passes the 50Hz test btw ...) I will try the other screens later this week-end. Would Auto@50-60Hz be a better default choice ?

My source is a C128 (DCR) 80 columns, no TTL board. Works fine, I tried the "VDC Mode Mania" "demo", which does a lot of tricks to get higher resolutions & colors on screen, I had to set "Auto Switch" to off since the interlaced modes where troubling rgb-to-hdmi sync. It works then, but it is quite strange to see a TFT flicker :-)

Thanks for the support !

IanSB, 2021-01-09:
>Just tried with SW1 pressed and got my 2410 working, great. I am now using Auto@50-60Hz (it passes the 50Hz test btw ...) I will try the other screens later this week-end. Would Auto@50-60Hz be a better default choice ?

I'm not sure what's going on there. The default resolution of "Default@EDID" is the same as "Auto@50-60Hz" except that it tries to be safer and always sends 60Hz unless the EDID indicates that 50Hz will work whereas Auto@50-60Hz always sends 50Hz for 50Hz sources which means you will get blank screens on monitors that don't support 50Hz when using that.

Did you change the resolution setting to something else before running your monitor tests? (holding SW1 sets it back to "Default@EDID" as well as temporarily forcing 60hz)

Also be aware that a lot of old Dell monitors will work with 50Hz input and thus pass the test but they do an internal 50Hz to 60Hz conversion and send video to the LCD at 60Hz which makes their 50Hz support useless as you end up with the same judder on smooth motion as you would get when sending 60Hz out of the Pi instead. 

My Dell 2707 and 2007 both do this but they do actually produce a display so I'm surprised you are getting no output with your 2407 which is from the same year.

HP monitors of the same vintage like your LP1965 and the LP2065 are much better as they actually work at 50Hz with no internal conversion.

> I tried the "VDC Mode Mania" "demo",

Non standard modes are not yet fully supported on the C128 but at least some of them now work.
There is a beta test version on my github fork which you can try with "VDC Mode Mania"
https://github.com/IanSB/RGBtoHDMI/releases/tag/7eef881

This build also has a "HDMI mode" setting that allows the HDMI output mode to be changed which might also have an effect on some of the non-working monitors although you will have to change the setting on a working monitor to start with.
Note changing from DVI compatible will stop it working with the HP LP1965 and probably some of your other DVI monitors

Sch-LikA, 2021-01-10:
Hi again,

I just retried my monitors, with other cables this time, and yes, they all work. Sorry for the wrong alarm. The only problem remaining is the Nexdock, it still refuses to show any image.

I will try your new build next and run "VDC Mode Mania" !

Sch-LikA, 2021-01-10:
OK amazing stuff. Your beta firmware works fine on my Dell 2410.

- Interlaced modes are still flickering but sync seems stable by default, much better already
- vdc-fli renders amazingly well
- vdc-hfli still has issues on my setup, sync is not stable and image cannot be seen correctly
- vdc-himono gives "auto switch : no profile matched" and does not work well
- Big-text-pal & Big-text-ntsc gets recognized ok but shows a kind of wave effect on the right side of the screen, still readable 

I will get my CM8833 (mkI) I just repaired out to compare a few colors, but it is much better now. To avoid flicker, would some kind of Amiga 2024 "high persistence phosphor" emulation help (no idea how to do that maybe mix the two frames in a buffer and render around 50hz ?) Thanks again !!

IanSB, 2021-01-13:
>"VDC Mode Mania"

Unfortunately I don't have a Commodore 128 (yet) so I'm relying on feedback from third party testers

>  Interlaced modes are still flickering but sync seems stable by default, much better already

The problem with those two interlaced modes (VDC-IHFLI 266 & VDC-ITFLI 315) is that they output progressive syncs with interlaced video so effectively they are alternating progressive frames for a dithering effect and they won't be weave deinterlaced because there is no way to detect that is what they are doing. If they had interlaced syncs that would be detected and they would be weave deinterlaced. It might be possible to have a forced deinterlace mode but then the pair order of the woven lines would be random each time the mode was selected.

> vdc-himono gives "auto switch : no profile matched" and does not work well

Not sure what mode you mean, it supports VDC-IMONO 749i & VDC-IM800 651i which I've been told work OK and are deinterlaced correctly as they output interlaced syncs. It looks like you are using another mode so what version of the demo are you running? 

> Big-text-pal & Big-text-ntsc gets recognized ok but shows a kind of wave effect on the right side of the screen, still readable

This probably means the line length is set incorrectly in the geometry menu as that will cause vertical noise bars so try adjusting that setting. When adjusting, you are looking for a result that is either all clean or all noise, not alternating noisy and clean bars and after you get a result, tweak the sampling phase in the sampling menu to get rid of any remaining noise and save the profile.
If you get a good result, please let me know the values you changed.

Sch-LikA, 2021-01-15:
I am using V1.01 of Vdc Mode Mania. VDC-IMONO is 720x700 interlace monochrome, refresh at ~21Hz according to the demo. Interestingly when I press button 1 the image looks half ok behind, but when out of the menu I see the led blinking and the image appears interlaced flickering with the led.

I just set "line length" to 1017 and screen is perfect in both Big-text-* modes. I played with sampling phase, all but 3 and 4 work fine. So 1017 as line length and anything but 3 or 4 as sampling phase works fine on my setup !

That stuff is amazing. I just built an analog module for my CBM-II, I'll try next. Thanks for your time !

Sch-LikA, 2021-01-16:
oh, btw this is on a PAL C128DCR.

IanSB, 2021-01-16:
>I just set "line length" to 1017

To confirm that is the correct setting: the original noisy value was 1016, does setting it to 1018 produce noise as well?

>VDC-IMONO is 720x700 interlace monochrome, refresh at ~21Hz according to the demo

Can you post a screencap of the Source Summary menu in the info menu when running that mode.
(Go into that menu then press the up and down buttons and that will screencap the menu image to the SD card as a PNG)

IanSB, 2021-01-20:
@Sch-LikA 

> I am using V1.01 of Vdc Mode Mania.

My other tester has been using V1.11 of VDC Mode Mania and it works with that (apart from VDC-HFLI_403 which has very strange results in the logs and may have out of spec H or V sync pulses)

That appears to be downloadable here:
https://csdb.dk/release/?id=161195
Can you try that version?

Sch-LikA, 2021-01-21:
Hi ! I just tried Vdc Mode Mania 1.11 

Mode 1 & 2 (VDC-IHFLI & VDC-ITFLI)  : All ok
Mode 3 : Pictures have a bad vertical alignment, picture starts at ~1/3 top and there is a black line in the middle, like in this capture : 
[image]

Mode 4 - VDC-HFLI 403 : Gets detected but does not sync
Mode 5 : Works fine !

Here is the capture of VDC 1.01 IMONO mode :
[image]

Text modes : I confirm 1017 is the correct line length, 1016 and 1018 produce noise.

IanSB, 2021-01-21:
@Sch-LikA 
>Here is the capture of VDC 1.01 IMONO mode

The current version of IMONO mode has a PAL line length of 1024 but your version has an NTSC line length of 1016 instead.

I think I've figured out what the problem is with the other modes so there should be a fix soon

IanSB, 2021-01-22:
@Sch-LikA 

Here is an updated beta 14 which might work properly in all VCD modes:

https://github.com/IanSB/RGBtoHDMI/releases/download/98d62b35/beta14.zip

IanSB, 2021-01-26:
@Sch-LikA 

Now up to beta 16 which according to my other tester works with all VCD Mode mania:

https://github.com/IanSB/RGBtoHDMI/releases

One option that requires manual override is NTSC Big text because it shares the same timing as regular NTSC.
To select that you have to change Auto Switch to off and then select the NTSC text sub-profile (not profile).
I may implement some sort of popup menu for circumstances like this where two or more sub-profiles share the same timing.

Sch-LikA, 2021-01-29:
Just tried b16, Vdc Mode Mania 1.01 works fine in all modes.

Vdc Mode Mania 1.11 still has the same issue in mode 3 (the astronaut), and mode 6 (VDC-IM800) flickers and does not sync. All the rest works fine !

I even had another issue resolved after forcing a resampling, I built a 4 option roms switcher, and Basic8 was having issue in text mode. It does now work fine. I'll try some Basic8 graphic modes later.

Thanks a lot, huge progress.

IanSB, 2021-02-01:
@Sch-LikA 

Please post source summary screencaps for those that don't work.

Sch-LikA, 2021-02-16:
Hi ... sorry for delay, here are the screenshots :

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

And finally text mode of Basic8 does blink too :

[image]

Btw I was working on Atari ST's lately, and was wondering if the monochrome hires "kindof" vga (640x400, 72hz vert /31.5khz hor) could be supported by RGBtoHDMI ?

IanSB, 2021-02-17:
@Sch-LikA 

>Btw I was working on Atari ST's lately, and was wondering if the monochrome hires "kindof" vga (640x400, 72hz vert /31.5khz hor) could be supported by RGBtoHDMI ?

Already works, see:
https://github.com/c0pperdragon/Amiga-Digital-Video/issues/6

Sch-LikA, 2021-02-17:
Amazing work guys, thanks !

victokra, 2023-05-24:
Hi, tokra here the author of "VDC Mode Mania". I just got an RGB2HDMI and am trying my modes with it. I had success with all of them but ITFLI, which is kind off picky and results in a flashing screen. Still investigating this.

> The problem with those two interlaced modes (VDC-IHFLI 266 & VDC-ITFLI 315) is that they output progressive syncs with interlaced video so effectively they are alternating progressive frames for a dithering effect

This is news to me :) I did not consciously do anything else with these two modes than with the monochrome-modes. They both set the VDC-register 8 to "interlaces video & sync". Maybe adding colour to the mix makes the VDC do strange things? Would be interested to go deeper into this if someone can explain what exactly is and should be happening.

Anyway, the wrongly placed picture of VDC-FLI in V1.11 is due to a small screw-up of mine I realized only years later. Other than earlier versions, I must have copied a line wrong and now it load the settings for the wrong mode and as such sets reg 7 (VSYNC-position) wrong. It is probably the easiest to just use V1.13 from my homepage which fixes this issue and speeds up the screen-copy and screen-clean a bit since it now uses assembler-routines. Get it here: https://www.tokra.de/c128/vdcmodemania-v113.zip

IanSB, 2023-05-25:
@victokra 
>I had success with all of them but ITFLI, which is kind off picky and results in a flashing screen. Still investigating this.

Flashing screen either means there isn't a profile that matches the timing or the selected profile has a capture size that exceeds the size of the video frame. (See the help menus in the latest release)

>>The problem with those two interlaced modes (VDC-IHFLI 266 & VDC-ITFLI 315) is that they output progressive syncs with interlaced video so effectively they are alternating progressive frames for a dithering effect

> This is news to me :) I did not consciously do anything else with these two modes than with the monochrome-modes. They both set the VDC-register 8 to "interlaces video & sync". Maybe adding colour [...]

The main problem is that I don't have any Commodore 128 hardware so debugging issues like this is not easy.

What does the source summary show for those modes (i.e are they being detected as progressive or interlaced?)
Go into the info menu and select source summary while those modes are being displayed. 
You can screencap the menu by pressing SW2 & SW3 together.

>I have attached my profiles for Mode Mania and the newer modes.

Can all these mode files be included in the single Commodore 128 profile folder?
(i.e. do they all have different timings so they can be auto selected?)

IanSB, 2023-05-25:
@victokra 
>These are the screenshots of the menus. 

Progressive / interlace detection is done by first measuring the average line duration over 100 lines of video then measuring the time for two frames / fields. This value is then divided by the line duration (with suitable rounding) to find the number of lines for two frames /  fields. 
If this number is odd then the video is interlaced else it is progressive.
Could you use the info menu and select the save log and EDID option while displaying a problem mode then post log.txt from the SD card as this should show more details about those measurements.

victokra, 2023-05-25:
Here you go with the log. Hope you can make something out of it. Searching for IHFLI and ITFLI should bring you to the affected sections.

[Log.txt](https://github.com/hoglet67/RGBtoHDMI/files/11567593/Log.txt)

IanSB, 2023-05-25:
@victokra 

```
Autoswitch match: VDC-IHFLI_266 (8) = 63182, 63817, 266, 3
Lines per frame = 266, (266)
Actual frame time = 16891206 ns (non-interlaced), line time = 63500 ns

Autoswitch match: VDC-ITFLI_315 (12) = 63680, 64320, 315, 3
Lines per frame = 315, (315)
Actual frame time = 20160203 ns (non-interlaced), line time = 64000 ns
```
The time for two frames/fields is only logged when interlace is detected and in the above cases the value for only one frame is logged instead when progressive is detected but this value is merely the two frame value divided by 2 so it is easy to reproduce the original value by multiplying by two
```
16891206 * 2 = 33782412 / 63500 = 532.0064 so 532 lines
20160203 * 2 = 40320406 / 64000 = 630.0063 so 630 lines
```

As these are both even numbers I don't think they can be interlaced signals as there has to be half a line extra per field for interlace. Maybe some combination of VDC register settings is stopping interlaced sync from being generated?

Do the other interlaced modes work OK?

victokra, 2023-05-25:
Yes, both VDC-IMONO (720x700) and VDC-IMONO800 are recognized as interlaced and displayed correctly. I assume it is the color-activation that messes up the interlace. The VDC was known to be a buggy chip. I found another interlace color-program (IPAINT) that shows exactly the same problem. Anyways, thanks for the explanation. That gives me a good starting point to investigate and experiment.

Maybe it could be made an option to force interlace and pick top / bottom field order?

IanSB, 2023-05-26:
@victokra 

Here is a proposed Commodore 128 profile set with your changes plus I also set the frame buffer to double height in all profiles as that is usually required for variable scanlines and may also help with interlaced display

Delete the existing Commodore_128-80 folder in /profiles and replace with the attached folder.

Also delete the Commodore_128-80 folder in /saved_profiles as that will override the new values in /profiles

I have set the video to progressive for VDC-IHFLI_266 & VDC-ITFLI_315 but you might want to change back to interlaced to see if the double height framebuffer setting makes any difference.

If the above looks OK I'll add it to the next release, if not please post your corrections.
[Commodore_128-80.zip](https://github.com/hoglet67/RGBtoHDMI/files/11570283/Commodore_128-80.zip)

victokra, 2023-05-26:
Ok, these profiles work just as well, but did not improve the interlace-recognition. I investigated a little more why the monochrome modes are recognized correctly and the color ones are not. Basically the difference is in the character-height register (reg 9) which has to be set for the character-height - 1. While for monochrome-interlaced-modes these are set to a height of 7 (odd) for the color modes they are set to 4 or 6 (even). I've checked that odd values always lead to interlace being detected correctly and even values always don't. I can see no really good reason for this though. As said the VDC was known to be buggy.

The number of lines is calculated by (reg 4 +1) * (reg 9 + 1).
For IHFLI this is: (132 +1) * (3 +1) = 532 interlaced
ITFLI: (104 +1) * (5 +1) = 630 interlaced
IMONO: (106 +1) * (6 + 1) = 749 interlaced
IM800: (92 + 1) * (6 +1) = 651 interlaced

However for the color-modes I **have to** set to the register to even values as each field has to have an even number of lines for color. E.g. with 4 I get the following colors:

AAAAAAAA
BBBBBBBB
AAAAAAAA
BBBBBBBB

CCCCCCCC
DDDDDDDD
CCCCCCCC
DDDDDDDD

A nice setup. But with 5 the setup would look like this:

AAAAAAAA
BBBBBBBB
AAAAAAAA
BBBBBBBB
AAAAAAAA

CCCCCCCC
DDDDDDDD
CCCCCCCC
DDDDDDDD
CCCCCCCC

So, less color-granularity (2 colors per 5 lines instead of 2 per 4 lines). I've actually tried such a setting and it was correctly recognized, but it just does not make much sense :)

Only option seems to force interlace for even numbered line count and to choose field order.

IanSB, 2023-05-28:
@victokra 

Have you tried adjusting register 5 (Vertical Total Fine Adjust) ?
http://www.ffd2.com/fridge/chacking/c=hacking2.txt

If you have one of those cheap USB logic analysers it might be interesting to look at the relationship between the H and V sync pulses on two adjacent frames in progressive, normal interlaced and bad interlaced modes.

victokra, 2023-05-28:
Yes, but sadly contrary to original documentation register 5 only works in NON-interlaced mode. Since I saw the first examples of interlace on the VDC back in 1992 I've always wondered why they always set register 5 to the same value as register 9 (character height). It really makes no sense from a technical point of view. But only if both registers have the same value interlaced display is working at all. You would think that lower values in register 5 would be the fine adjust of the vertical total number of lines, but then interlace is just lost completely and only one field is displayed (also on CRTs). The fine adjust in reg 5 only works as documented in NON-interlace mode and in interlace-mode does not add any lines to VerticalTotal at all. I've also checked the few "professional" interlace software from the past (like IPaint and Graphic Booster) and they all set reg 5 to the same value as reg 9 - with the above mentioned side-effect that you get an even total number of lines with the wrong combination which the RGB2HDMI does not recognize as Interlace. Which worked fine on the Commodore-monitors I use (1901 and 1084), they don't care about the missing half extra line in each interlaced-field.

To get "correct" interlace the value in register 9 multiplied by the total number of lines in register 4 must be odd. This is the case for my VDC-IMONO (107*7=749 total vertical lines) and VDC-IM800 (93*7=651 total vertical) and that's why auto-deinterlace with the RGB2HDMI. Xander Mol did a video of VDC Mode Mania with his HDMI-capture devices which you can check here:

https://www.youtube.com/watch?v=toeOYcWU-6o

The color-modes flicker similar to as they would on a CRT because the interlace is not recognized. The monochrome-interlace modes starting a 3:54 are rock sold in comparison.

I still chalk this up to the VDC being a buggy chip. Reading about its history its a wonder it works as well as it does.

I do not have a logic analyzer to do further investigations, but will ask around :) Thanks for your suggestions.

IanSB, 2023-05-28:
@victokra 
>Those worked fine on the Commodore-monitors I use (1901 and 1084), they don't care about the missing half extra line in each interlaced-field.

If they are sending progressive syncs then they shouldn't really work fine on an analog monitor as it would draw both sets of images in the same place rather than interleaving them but on such a small monitor with the flicker it can be difficult to see that.

The only other possibility is that these non standard modes are doing something else with the syncs which is not detected by current RGBtoHDMI code but as mentioned above the only way to determine that is to use a logic analyser or digital storage oscilloscope to look at H & V syncs on two adjacent fields
i.e. something like this:
https://www.ebay.co.uk/itm/314272582941
(There are many sellers)

victokra, 2023-05-28:
I only can say for sure that I visually ascertained on my Commodore 1901 which field came first back when I programmed the demo back in 2011. If I flipped the field order around in the display routine the picture looked wrong.

Sorry, I have no idea about logic analyzers and would not even know how and where to connect them and what to look for when running the software or how to set it up.

I agree that getting to the bottom of this quirk is the final goal, but would a manual override be possible in the meantime?

IanSB, 2023-05-28:
@victokra 
>If I flipped the field order around in the display routine the picture looked wrong

Then there must be something different between the syncs in each field. The only thing I can think of is that the vsync is shifted by half a line in one field which would make the first field one line longer than the second field.
e.g. IHFLI would be 266.5 & 265.5
It would still be possible to discriminate between fields if that was the case (which is what your analog monitor appeared to be doing).
Here is a version of kernelrpi.img that forces interlace to true for all inputs. (This may mess up other modes)

[kernelrpi-force-interlace.zip](https://github.com/hoglet67/RGBtoHDMI/files/11586956/kernelrpi-force-interlace.zip)

Overwrite your existing copy on the SD card and reboot
Note you will have to change the video type from progressive back to interlaced in the two problem profiles for this to have any effect.

There are several possible outcomes from this change:
1. no effect
2. it works!
3. It works but the fields are reversed
4. It works but the fields are offset by a line or two
Post a screencap if it makes a difference
If this does make a difference then I could add a force interlace setting to the Vsync Type to always assume interlace just for those profiles. (The current interlace setting just allows interlace or progressive as detected)

victokra, 2023-05-29:
2. It works!

[image]
[image]

Not sure why the capture of ITFLI is that stretched. It looks fine on the display. Original resolutions are 640x480 for the Terminator-pic and 640x576 for the train.

So, this indeed seems to be the solution for now. Thanks for your efforts! Hope to see this implemented in the next release/beta.

IanSB, 2023-05-29:
@victokra 
>2. It works!

That's great although it confirms the output is very non-standard as no video signal like that should have different line numbers in each field.

Anyway here is an updated set of files:
[C128_even_interlace.zip](https://github.com/hoglet67/RGBtoHDMI/files/11593931/C128_even_interlace.zip)

Replace kernelrpi.img and the Commodore_128-80 profile set on your SD card and reboot
Make sure to delete any saved profiles as they will override the new ones.

This version has a new setting in Vsync Type of "Force Interlaced" which forces interlaced detection always for that profile.
The new Commodore_128-80 profile has the VDC-IHFLI & VDC-ITFLI sub-profiles updated to use that setting. (I also renamed them as VDC-IHFLI_532i & VDC-ITFLI_630i)

If you can test that and confirm everything is working as expected then I'll commit that to the repository for the next beta.

BTW I've been working with c0pperdragon to do an RGBtoHDMI adapter for the Commodore 64 video. It uses a small plug in board which sits under the VIC II to pick up the digital signals going to the VICII chip and it then sends them out of the C64 on the UHF modulator's phono in an encoded format so that an RGBtoHDMI can plug into that and decode the data to make a completely noise free bit perfect image. 
Do you think there would be interest in a C128 variant of that?

victokra, 2023-05-29:
Thanks. Here's my version

[Commodore_128-80.zip](https://github.com/hoglet67/RGBtoHDMI/files/11594403/Commodore_128-80.zip)

I made the following changes:

- adjusted the vertical offset for VDC-IHFLI_532i.txt
- deleted Cmd_128_60Hz_262.txt - the ROM-sets by Commodore only produces 264 lines in NTSC. Not sure where the 262 lines alternative came from. Feel free to keep it if there's another good reason.
- added Cmd_128_50Hz_320.txt - there was a ROM-revision by Commodore in 1986 which adjusted PAL-settings for VDC to be more correct (15625 Hz and 312 lines). However all machines before that had slightly off PAL-timing (1016 line-length = 15748 Hz, 320 lines). I think about 50% off all PAL-C128 out there will be the older revision.
- renamed Cmd_128_50Hz.txt  to Cmd_128_50Hz_312.txt to reflect above change
- picked palette RGBI for VDC-Spectrumania-4in1_410.txt - this one requires black yellow, was set to RGBI (CGA) before

Regarding c0pperdragon: I understand this basically reproduces the VIC-chip's output on its own? Does the mod require the HF-modulator? A lot of people have removed theirs from their machine for other picture-improving hardware (on C64 and C128) to remove "jailbars". If just the jack is used that should be fine. C128 folks would be just as interested I think. Not sure if the extra registers of the VIC-IIe in the C128 matter (test-bit and 2 Mhz-mode).

IanSB, 2023-05-30:
@victokra 
>I made the following changes:

Thanks for your updates.

>deleted Cmd_128_60Hz_262.txt - the ROM-sets by Commodore only produces 264 lines in NTSC. Not sure where the 262 lines alternative came from. Feel free to keep it if there's another good reason.

The NTSC version of PETSCII robots produces 262 line video and it was added to support that, see: https://github.com/hoglet67/RGBtoHDMI/issues/266

The old c0pperdragon board was a multi board YUV adapter, this new board only works with RGBtoHDMI and just fits under the VICII, monitors the data bus and generates a signal called "lumacode" (a four level analog mono video signal at double the normal clock rate) which can be output on a phono connector such as the RF one (RF output disconnected). An analog RGBtoHDMI can decode that back into a C64 image. (It's a way of getting bit perfect video out of the computer on a single wire).

[image]
[image]
[image]
[image]

victokra, 2023-05-30:
@IanSB 

> The NTSC version of PETSCII robots produces 262 line video and it was added to support that, see: #266

Ah, I see. I've checked the PAL-version of that game and that still does 312 lines. The author must have chosen 262 since it's closer not NTSC.

> c0pperdragon [...] new board only works with RGBtoHDMI
> An analog RGBtoHDMI can decode that back into a C64 image. (It's a way of getting bit perfect video out of the computer on a single wire).

That's nice, though likely not for me since I got a digital RGBtoHDMI only (the one from Texelec). My C128 has a jailbar-fix and near-perfect VIC-picture already, though not HDMI...

Another issue that may fit this topic: Applying what I learned I created a 960x540-mode for the VDC (so half horizontal and vertical of FullHD) and it works nicely. However when the output resolution is set to 1920x1080 it does not scale horizontally as it should. See pictures  of source summary and geometry menu below. Do you see any reason why it would not work scale horizontally?

[image]
[image]

Strange enough when I set the output resolution to 1920x1200 it scales correctly horizontally, but of course then I have a small black border and the top and bottom and this is not the native resolution of my display device.

[image]

I only get the scaling to apply when output res is set to 1920x1080 if I cange the geometry menu.

[image]

Any idea what might be going on here?

IanSB, 2023-05-31:
@victokra 
>Any idea what might be going on here?

Yes, the integer scaling was originally designed to assume 4:3 sources and scaled to fill that available area on widescreen panels.

I later introduced the H and V pixel aspect settings which allowed integer scaling to specify the aspect ratio of the LCD pixel block to be used for one source pixel which constrained the size of the resulting image and at the same time allowed a wider than 4:3 image to be displayed which was useful for displaying 768x576 cropped to 768x540 etc. 
However due to the scaling calculations that worked out at 912 pixels for a 1920x1080 and 960 pixels for 1920x1200

I have tweaked the code to use the original width of the display with integer scaling as the H and V pixel aspect should constrain the image to 4:3 for such sources so please try this:

[kernelrpi-widescreen.zip](https://github.com/hoglet67/RGBtoHDMI/files/11607451/kernelrpi-widescreen.zip)

Note that if either of the H and V aspect settings are set to 0 then it will fill the screen as much as possible. Similarly if the H and V aspect are an impossible combination like 8:1 that will also just fill the screen.

Some of the existing profiles have H and V aspect set to 0 so they should be tweaked to fix a sensible aspect ratio to avoid any issues with this change.
Note when you adjust them, a line at the top shows the current actual aspect ratio after rejecting invalid or zero settings so if that looks correct then set the values to match

Interpolated scaling has a "4:3" and "fill all" settings so you would have to select the appropriate one for those scaling modes.

victokra, 2023-05-31:
Hooray! That works great. I checked all the profiles again. Here are the few that have been updated. I have included the one for the upcoming 960x540-mode which may prompt me to release a new version of VDC Mode Mania with the "special mode" for the RGB2HDMI. I will probably wait until the recent changes become official or at least an official beta.

Thanks for this project. I did not realize how cool the RGB2HDMI is until I was told at the Revision-demoparty to get one :)

[Commodore_128-80.zip](https://github.com/hoglet67/RGBtoHDMI/files/11611108/Commodore_128-80.zip)
[image]
[image]

timmcross, 2023-05-31:
> BTW I've been working with c0pperdragon to do an RGBtoHDMI adapter for the Commodore 64 video. It uses a small plug in board which sits under the VIC II to pick up the digital signals going to the [...]

IMHO, I would love to give this a try for both the C64 and C128!

xahmol, 2023-05-31:
> BTW I've been working with c0pperdragon to do an RGBtoHDMI adapter for the Commodore 64 video. > > (..)
> Do you think there would be interest in a C128 variant of that?

Would certainly also be interested, provided that it is an easy install for a hardware handicapped person like me. Does it need soldering or something like that? (Think it did not)

IanSB, 2023-06-02:
@victokra 
@timmcross 
@xahmol 
>Does it need soldering or something like that? (Think it did not)

It can in theory be just a plug in solution. I think c0pperdragon is selling his board with crocodile clips so it can be connected to the modulator phono without soldering but that might be a problem with the breadbin C64 because the modulator lid on at least some of those models has a tab soldered to the body so getting the lid off needs a soldering iron unless the tab can be cut with some wire cutters. Also the very early breadbin model with discrete clock circuitry needs an additional socket to lift the board clear of some nearby components.

An alternative might be to use a cable with a phono dangling out the back of the C64 but that is less elegant than re-using the modulator phono. You would have to contact the sellers to see if they would offer that option.

Note you also need an analog RGBtoHDMI, either a standard one or the lower cost "lumacode" one designed for use mainly with this style of adapter 

see
https://github.com/c0pperdragon/c0pperdragon.github.io/wiki/VIC-II-dizer

also with lumacode RGBtoHDMI and case but no Pi zero:
https://www.ebay.co.uk/itm/235009314007?hash=item36b7a58cd7:g:Gi0AAOSwlIVkX0Zl

this one includes lumacode RGBtoHDMI with case, raspberry Pi & sd card)
https://www.sellmyretro.com/offer/details/63609

victokra, 2023-06-06:
@IanSB 

I'm still experimenting with this fine device and am not sure if this is a bug or by design.

I have the 720x700 interlace mode. This is how I set it up on RGB2HDMI:

[image]

H Pixel Aspect = 2 and V Pixel Aspect = 3 means this is scaled to 1440 x 1050. But Min and Max V Height are both 350 as they are obviously counting the lines of each field and not both fields combined? However the real resolution in interlace is 700 lines and when this is scaled to 1050 this is factor of 1.5 and it shows in the picture I think

[image]

Some pixels have double height, some single height.
If I set Min V Height to 270 the picture is cut off but looks better:

[image]

All pixels have the same height again. It may be hard to see zooming in on the picture I've done with my phone.
For reference here are the capture-pictures from the device for the two settings.

[image]
[image]

Here the first picture has a height of 720 and as such does not show any distorted pixels. But when I resize this to 1080 in IrfanView for example I get a similar effect as on the 1080p-display.

So, what to make of this? Should interlaced-scaling work differently?

IanSB, 2023-06-06:
@victokra 

>I'm still experimenting with this fine device and am not sure if this is a bug or by design.
I have the 720x700 interlace mode. This is how I set it up on RGB2HDMI:
H Pixel Aspect = 2 and V Pixel Aspect = 3 means this is scaled to 1440 x 1050. But Min and Max V Height are both 350 as they are obviously counting the lines of each field and not both fields combined? However the real resolution in interlace is 700 lines and when this is scaled to 1050 this is factor of 1.5 and it shows in the picture I think

This behaviour is by design but wasn't really intended for use with interlaced sources.

Fractional integer scaling (an integer plus a half) is allowed vertically or horizontally when double height or double width capture is enabled. (The final scaling factor is shown in the source summary page)
Double height/width means that two pixels are written to the Pi's frame buffer in the appropriate direction for each pixel captured. This has two purposes:
1. It improves scaling quality when interpolated scaling is used because for small sources like 320x200 there is less difference between the frame buffer size (e.g. 640x400) and the final display size so the image is not as soft as interpolating 320x200 directly.
2. Double height frame buffers are required for scanlines as there are two vertical pixels in the frame buffer for each source pixel.

Doubling the height and width of the frame buffer like that initially had a restricting effect on integer scaling because the frame buffer was double it's original size.

e.g. A source like a commodore 64 which is 320x200 could be integer scaled to a 1024x786 monitor with 3x scaling in both directions for 960x600 with some border.  If double height & width were both enabled, the frame buffer would be 640x400 and that doesn't integer scale at all on a 1024x768 monitor so your would be left with a 640x400 image with very large borders or have to use interpolated scaling instead.

However I realised that using 0.5 steps in integer scaling would still produce a good image because although alternate pixels would be different sizes, they would be the same colour so it wouldn't be noticeable. When the scanlines option is enabled, although the height of all the scanlines would be different to the height of all the source pixels that wouldn't be objectionable either.
e.g. in my above example of scaling 640x400 to 1024x768 with 1.5x scaling the pixels would be alternately 1 and 2 LCD pixels wide/high.

However, this doesn't work so well with interlaced sources. They make use of the double height frame buffer to store both fields as the real height of the image is twice the height in geometry but the data is different in each pair of lines so you get uneven pixels.
There is no way around this as you can't integer scale a 700x700 image to a 1920x1080 screen with the correct aspect ratio as  simply doesn't fit so you would either have to specify the H & V pixel aspect ratio of 1:2 or 2:2 to get even pixels or use interpolated scaling.
I suppose I could disable the half integer scaling when an interlaced source is detected but you can already override it as above by specifying a different aspect ratio and there are issues as the BBC micro can generate progressive images with interlaced sync and I wouldn't want to disable the half integer scaling under such conditions.

BTW 2:3 (i.e. 2 : 1.5) scaling would work on a 4K monitor

victokra, 2023-06-06:
Thanks for the explanation. In the end it is a decision what looks best considering all constraints. I.e. do I want to display every pixel in the correct aspect ratio, but live with a small-window-boxed area? Or can I live with half-integer-scaling if the resulting shown picture is larger. Or is it possible to reduce the V height Min size just slightly, so better scaling options are availabe.

I've decided on a case-by-case basis now. The color-mode ITFLI had 1.5 scaling because it was 576 pixels high. I've reduced min-size to 540 and while loosing 36 lines it now looks much better.

On monochrome the 1.5 scaling looks ok and does not cause as much problems as with color. So I've left this for 720x700 and applied it to 800x600, so this fills a larger portion of the screen now than before.

My "approved" updated 2 profiles attached. Would be nice if you updated those 2 in the main-branch as well.

[Commodore_128-80-updated-profiles.zip](https://github.com/hoglet67/RGBtoHDMI/files/11670701/Commodore_128-80-updated-profiles.zip)

IanSB, 2023-06-06:
@victokra 

>My "approved" updated 2 profiles attached. Would be nice if you updated those 2 in the main-branch as well.

OK I've committed that as well. Note for now this is in the dev branch on my fork where the betas are released. It will eventually find its way to the main branch here in the next stable release.
https://github.com/IanSB/RGBtoHDMI/commits/dev

IanSB, 2023-07-25:
@victokra 
>added Cmd_128_50Hz_320.txt - there was a ROM-revision by Commodore in 1986 which adjusted PAL-settings for VDC to be more correct (15625 Hz and 312 lines). However all machines before that had slightly off PAL-timing (1016 line-length = 15748 Hz, 320 lines). I think about 50% off all PAL-C128 out there will have the older revision ROM.

I just added a Commodore 128 to my collection and this has the 320 line ROM.
However the sub-profile that often gets selected is Text-PAL_320.txt (Which has 1017 clocks per line) rather than Cmd_128_50Hz_320.txt (which has 1016 clocks per line).
The above is the only difference between them (for detection purposes) and that is too close together to be reliably differentiated so I have had to remove Text-PAL_320.txt to ensure that the standard mode works all the time.
The same is true of Cmd_128_60Hz_264.txt and Text-NTSC_264.txt so I have had to remove the latter as well.

This means your text demos using the above will not be displayed properly as they will be slightly mis-sampled.
There are a several ways around this:
I could add those as manually selectable profiles
You could modify the demos to use 1016 clocks (the standard profiles might need their max size increased)
You could change the number of lines in those demos by at least 2 (e.g. 322 and 266) which would allow updated profiles to be differentiated.

BTW my C128 had a bizarre fault:
It all worked except the 80 column output which had characters with bits 1 & 3 stuck high. I first thought this was a RAM fault so I changed VDC RAM U23 but that didn't help. I then though that the VDC might have failed but I noticed that running my fingers along the VDC pins would change the video on the screen which corresponded to touching DD1 & DD3 so they appeared to be floating and sure enough there was no continuity between those VDC lines and the DRAM.
After adding a couple of links it all started working properly. It might have been like that from new but never noticed if the 80 column mode was never used.

victokra, 2023-07-25:
I noticed the mis-detection for the text-modes as well. These are really just add-ons to the demo, not of the utmost importance, but your post got me thinking about them again.

Effectively these modes where just designed to show how it is possible to fill our more screen area on a regular display-device.

When I originally did the demo in 2012 I wanted to keep the original Commodore-ROM-settings as much as possible, that's why they have 320 lines for PAL and 264 for NTSC. However these values were never really as close to the PAL or NTSC-standards as they could be. My intention is to make them as close to NTSC and PAL as possible, so most TVs and monitors will display them correctly.

The clock cycles need to be a multiple of 9 (coming from increasing the char-size to 9) for the wider text-screen-size. So for NTSC they can only be 1017 and for PAL they could also be 1026 to be close enough to the original horizontal frequencies for NTSC and PAL.

But then I have 262 lines on NTSC: 16.000.000 / 262 / 1017 = 60.05 Hz
and 312 lines for PAL: 16.000.000 / 312 / 1026 = 49.98 Hz
or 315 lines for PAL: 16.000.000 / 315 / 1017 = 49.94 Hz

But all these again lead to the problem of conflicts with the other (standard) sub-profiles. If I use a different number of lines this might break the modes on other display-devices which expect signals close to the PAL/NTSC-standard.

So, all things considered, option 1 " I could add those as manually selectable profiles" seems to be the most sensible solution. Is it possible to mark profiles that way, so that they can only be selected manually? I did not find such an option when fiddling around witht he modes in June.

IanSB, 2023-07-25:
@victokra 
>Is it possible to mark profiles that way, so that they can only be selected manually?

There is an option to switch between two 'timing sets' for each profile (a timing set is all the settings in the geometry and sampling menu)
This is required for some computers that output using two different pixel clocks like the bbc micro which uses 12Mhz for teletext mode and 16Mhz for other modes. The Apple II GS also switches between 14.3Mhz and 16.3Mhz pixel clocks.
With some computers it is possible to automatically select either set 1 or 2 and those options can be chosen using the Auto Switch option. This also has a manual mode and that can be used to manually switch between the two sets. When selected you can either switch with the menu option that appears or press SW3 to toggle.

Here is a profile set with the 320 & 264 modes having set1 = normal text and set2 = demo text
[Commodore_128-80.zip](https://github.com/hoglet67/RGBtoHDMI/files/12165544/Commodore_128-80.zip)

Note that the set1/2 setting is saved so you can accidentally leave it defaulting to set2 after calibrating

I don't think this is the ideal solution as you will have to keep pressing SW3 during the demos so another option might be to have a separate VDC Demo profile with the demo versions of the 320 and 262 profiles (You could permanently use that profile set if you have the later ROMs with 312 & 262 line standard modes).

Not sure what to call such a profile, any suggestions?

Do you have a link to the latest versions of your demos as I should be able to run them now.

BTW c0pperdragon has implemented a version of his lumacode board mentioned above for the C128 which is currently being tested.

victokra, 2023-07-25:
So, the "VDC Demo profile" would have the two sub-sets for the two text-modes? Would like to try that myself. I'd call it "VDC Mode Mania Text profile" or "VMM Text profile".

I will send you the link to the latest VDC Mode Mania as I only want to release this once at least the new beta of RGB2HDMI-firmware that supports it is out. VGA Mania you can download her:

https://www.tokra.de/c128/vdcvgamania-v101.zip

IanSB, 2023-07-26:
@victokra 
I have released Alpha 59F here:
https://github.com/IanSB/RGBtoHDMI/issues/18#issuecomment-1650738215

Please download and try it.
There are two profiles:
Commodore_128-80
Commodore_128-80_(VDC_Text_Demos)

I have made an experimental change to the (VDC_Text_Demos) version so it should auto switch between the two 320 line and two 264 line sub-profiles. (i.e. support everything)
However this relies on the C128 16Mhz crystal being very accurate and I don't know if that is the case.

There is about 1000PPM difference between the 1016 and 1017 modes so I have reduced the Clock Tolerance setting in those 4 profiles from 5000PPM to 500PPM. As long as both the Pi zero and C128 crystals are accurate this will work. I have tried many Pi zeros and they do seem to be accurate but the C128 is more unknown.
e.g. at the moment when the standard 320 line mode is selected my PPM error is about 64PPM and switching to the 320 demo sub-profile the error changes to -908PPM which is greater than the 500 threshold so it will select the correct sub-profile.
The C128 uses an oscillator module rather than a discrete crystal circuit and they are usually trimmed for accuracy but more systems need to be tested to confirm if this is viable.
Worse case if there are inaccurate oscillators then the profile won't get selected at all and the video will flash as it can't find a match.
What sort of PPM errors do you get on your C128? (In the info menu)

IanSB, 2023-07-26:
@victokra 
I've just been trying your demos and found some problems:
1. Spectrumania
Standard Pixel width mode works
Double pixel width mode doesn't work as there is no profile to match it (I had to add one)
That mode is outputting 320 line video with line length 1024 rather than 1016
Are you partially modifying the VDC registers as that will mean the final output timing will depend on which ROM (312 or 320) is fitted?
Would this be an issue in NTSC mode as both the 262 & 264 profiles use 1016 clock cycles?
Can't test the 4in1 as I don't have 64K (would that be any different with different ROMs?)

Here is the additional profile:

[Cmd_128_50Hz_320_1024.txt](https://github.com/hoglet67/RGBtoHDMI/files/12172582/Cmd_128_50Hz_320_1024.txt)

2. VGA mode mania
The profiles are recognised correctly but I have tweaked the "FB size" setting to Double Height + Width for both of them so the menus are more readable:

[VDC-VGA-HFLI_513.txt](https://github.com/hoglet67/RGBtoHDMI/files/12172740/VDC-VGA-HFLI_513.txt)
[VDC-VGA-Text-Mono_521.txt](https://github.com/hoglet67/RGBtoHDMI/files/12172742/VDC-VGA-Text-Mono_521.txt)

The text menu is visible but the images are garbage:
[image]
[image]

3. VDC Mode Mania
The big text mode profiles are correctly selected with the reduced Clock tolerance setting but the images are also garbage:

[image]
[image]

Any idea what the problem is?
It could be a hardware issue so I will check all the address lines on the VDC DRAM but can you recommend any test programs for the C128?

tkracke, 2023-07-26:
Will try the PPM error later tonight. @xahmol should be able to test this as well.

Regarding palettes: VDC Mode Mania and VDC VGA Mania were made for RGBI (brown) and VDC Spectrumania was made for CGA (dark yellow). I do not approve of using the "wrong" palettes for the demos ;-)

IanSB, 2023-07-26:
@tkracke
>Regarding palettes: VDC Mode Mania and VDC VGA Mania were made for RGBI (brown) and VDC Spectrumania was made for CGA (dark yellow). I do not approve of using the "wrong" palettes for the demos ;-)

It's the other way around RGBI = dark yellow & CGA = brown.

I can leave all the VDC profiles with the correct palette and just change the palette for the standard text modes.

However there is a slight problem, these are the standard modes:
Cmd_128_50Hz_312
Cmd_128_50Hz_320
Cmd_128_60Hz_262
Cmd_128_60Hz_264

I have also added the following to cover Spectrumania:
Cmd_128_50Hz_320_1024
Cmd_128_60Hz_262_1024
Cmd_128_60Hz_264_1024
(They could be renamed to VDC-Spectrumania...)

These ones could be left permanently with the RGBI / yellow palette.

However when running Spectrumania on a PAL 312 line ROM there is no difference in the timing so Cmd_128_50Hz_312 would be selected and the palette will depend on which profile set is selected by the user.

So the two options are for the above Spectrumania profiles to always have the yellow palette with 312 line being the odd one out or for those profiles to track the user CGA/RGBI choice which would be consistent behaviour irrespective of ROM or PAL/NTSC.

Which would you prefer?

victokra, 2023-07-26:
Ok, this is the screenshot of the source summary right after power on:

[image]

I noticed in the 320 Big Text profile (choice 8 in VDC Mode Mania) the aspect ratio is 1:2 while 2:3 would look much better usually (at least at 1920x1080 which I am using).

victokra, 2023-07-26:
For testing purposes I quickly amended VDC Mode Mania so you can choose in which ROM-setting it starts (by switching the 3 affected registers):

[vdc mode mania-t.zip](https://github.com/hoglet67/RGBtoHDMI/files/12175855/vdc.mode.mania-t.zip)

This is for the SD2IEC-version, just start this instead of the usual program, it will present with the 3 options: NTSC, PAL-320 and PAL-312.

victokra, 2023-07-26:
>  track the user CGA/RGBI choice which would be consistent behaviour irrespective of ROM or PAL/NTSC

This option seems best. The demo itself tells you it would prefer dark yellow. So if you run it with brown you will notice and have the chance to change profiles then. Also the demo features (poor) software-emulation for dark yellow. So running this with brown active is not totally wrong.

xahmol, 2023-07-26:
> @xahmol should be able to test this as well.

I am in holiday now and not around any Commodore for another two weeks.... so can not test on short notice.

IanSB, 2023-07-26:
@victokra 
OK here are my proposed profile sets:
[Commodore_128_profiles.zip](https://github.com/hoglet67/RGBtoHDMI/files/12177373/Commodore_128_profiles.zip)

As discussed they are:
Commodore_128_CGA
Commodore_128_CGA_(Alt)
Commodore_128_RGBI
Commodore_128_RGBI_(Alt)

The palettes are only changed in the sub-profiles with names beginning C128_

The Clock tolerance PPM thresholds for all profiles are set to 3000 except the following in the Standard (non-Alt) profiles where they are set to 500:

C128_50Hz_320
C128_50Hz_320_1024
Text-PAL_320
C128_60Hz_264
C128_60Hz_264_1024
Text-NTSC_264

The "Text" profiles are removed in the (Alt) sets all thresholds are set to 3000 so the only thing that won't work properly with the Alt sets are those two text modes.

I also disabled the genlock in the following modes as they are below 50Hz so will never lock (They can be made to lock but then the output might not be displayed on most monitors)
VDC-HFLI_426
VDC-IM800_651i
VDC-IMONO_749i

Finally something strange happens when I select option 3 (VDC-FLI) in modemania:
All the other modes output the correct sync timing immediately they are selected during the black screen loading period but the above mode outputs some set of pulses that is being interpreted as intermittent loss of sync (the LED D2 which is normally off lights to indicate loss of sync and it is flashing) and this makes the menus unusable during that period.
Once the image is displayed the correct sync timing is output.
Is this expected behaviour?

victokra, 2023-07-27:
Gave it a quick test, seems to work fine. Why are the (Alt)-sets needed still? Is this just for PPM-tolerance? If it is confirmed that the pixel-clock of the C128 is stable enough I would rather remove them, as this might confuse users. Also, maybe CGA and RGBI could add "brown" and "dark yellow" to the the profile-name for clarity as even I get those mixed up...

Regarding VDC-FLI: This is expected behaviour. Long answer: The 8x1 pixel mode is a hack not supported originally by the VDC. If you set the registers to 8x1-mode it will just produce a black screen. The trick is to set the mode to 8x2 in the VBLANK-area and only set it to 8x1 once the raster is outside the VBLANK. This requires an assembler loop that watches the VBLANK and I run this only once the picture has been loaded from disk fully. So, while loading the mode is not fully set up and will have more lines.

[22 more replies]