hoglet67/RGBtoHDMI #175: Analog - Commodore CBM-II (Pet-II) not supported

Sch-LikA, 2021-01-15:
I just plugged my Commodore CBM-II B128-80 to my analog board. Loaded YUV firmware (not 100% clear in the doc btw) and tested profiles.

The best working one is the "Apple II", 60Hz, but I get a flickering screen with half fonts.

The CBM-II is outputing 80 columns luma+sync "composite" signal, 60hz (it is an American 110v-60hz machine plugged into a 220v-50Hz to 110v step-down converter). What can I tweak to get a working setup ?

Thanks a lot !!

IanSB, 2021-01-16:
Because it does pixel perfect sampling rather than general purpose sampling, the converter needs a profile which descibes the source signal in some detail with things such as the pixel clock, the number of pixel clock cycles per line (line length), number of lines per field and other things like the size and position of the active video within the frame.
Also for analog sources it needs info about the voltage levels of the signal.

Setting up a new profile is (mostly) decribed in the wiki:
https://github.com/hoglet67/RGBtoHDMI/wiki/Tutorial-on-Adding-a-New-Profile
I haven't completely finished that but there should be enough info for you to get a working image by adjusting things in the geometry menu and maybe the sampling menu.

I would start with a profile that gives best results such as the Apple II one above and then modify that, saving the resulting configuration. You might need to adjust the analog sampling levels to optimise their values but the Apple II ones are set up for 1vp-p 75R terminated video.

If you then drill down into the Saved_Profiles folder on the SD card you will find your saved Apple II config
Rename that to an appropriate name and move it to the equivalent folder in the Profiles folder.
Reboot and then you should be able to select your new profile from the Profiles option.

If you get a good result please post it here and I'll add it to the next release

Sch-LikA, 2021-01-21:
I am super close to having a profile working for my CBM-II. This is the closest I get :
[Commodore-CBM-II_60Hz.txt](https://github.com/hoglet67/RGBtoHDMI/files/5852605/Commodore-CBM-II_60Hz.txt)

Mostly works when adjusting values, but when getting back to normal mode, screen flickers regularly.

Here is a capture on the screen with some info : 
[image]

The info I got are : only one video mode : 80x25 lines of text, characters are 8x8 pixels (8x14 for the model with integrated monitor, this seems weird) > 640x200 pixels ?

When showed on a CRT, characters are around 1/2.

Datasheet of the videochip is [here](https://archive.org/details/crtc6545/mode/2up)  : 

I fired my gbs-control board, using yuv to vga I get a good output. I got the debug infos from it, I get (text after > is the variable name in the code since I cannot make sense of the acronyms) :

```
...
h: 428 v: 523 PLL:5 A:7b7b7b S:8f.10.00 H-   I:00 D:0584 m:1 ht:2345 vt: 246 hpw: 185 u:  0 s:ff S:14 W:-61
...

HT / scale   : 2402 512   > VDS_HSYNC_RST / VDS_HSCALE
HS ST/SP     : 16 144      >  VDS_HS_ST / VDS_HS_SP
HB ST/SP(d)  : 2322 340 > VDS_DIS_HB_ST / VDS_DIS_HB_SP
HB ST/SP     : 2248 208  > VDS_HB_ST VDS_HB_SP
------
VT / scale   : 750 683    > VDS_VSYNC_RST / VDS_VSCALE
VS ST/SP     : 1 6           > VDS_VS_ST / VDS_VS_SP
VB ST/SP(d)  : 750 26    > VDS_DIS_VB_ST / VDS_DIS_VB_SP
VB ST/SP     : 14 16        > VDS_VB_ST / VDS_VB_SP
IF VB ST/SP  : 8 10         > IF_VB_ST / IF_VB_SP
CsVT         : 246             > GBS::STATUS_SYNC_PROC_VTOTAL
CsVS_ST/SP   : 4 1         > getCsVsStart / getCsVsStop
```

Anything obvious for you ? Any setting I should tackle ?

Thanks !!

IanSB, 2021-01-21:
@Sch-LikA 
>Any setting I should tackle

Assuming the pixel clock is set to 18Mhz in the menu and is actually 18Mhz in the hardware, the PPM error is too great which means the line length setting also in the geometry menu is wrong, adjust until the PPM value gets closer to zero and you see all clear or all noise on the screen