hoglet67/RGBtoHDMI #179: What does JPINV1 do?

jan-hofmeier, 2021-01-27:
Hi,

I am currently designing a version for the Video Slot of the Amiga B2000 (anyone else interested?), because I want to extend it like it was intended by Commodore. 
I am looking at the designs by @c0pperdragon and I have a few questions I couldn't find an answer to:

I am wondering what JPINV1 in the "Small" Version does. 
If I understand the Logic with the XORs correctly, it is supposed to generate a rising edge for the Transceivers on the rising and the falling edge of CDAC, to double the clock frequency.
So inverting CDAC shouldn't have an effect in my eyes. Do you have any details on this?
I assume U1C is there to delay the signal, is this correct?

The V2 Version has a Jumper to either use CDAC or 7M inverted. From what I could make out the 7M is needed for Super Denise. But why does it get inverted? For delay?
As the Small Version doesn't have the 7M option I would assume that it doesn't work with Super Denise?

As I want to support Super Denise I need the 7M, which isn't available on the Video Slot. But the Video Slot provides /C1 and /C3, which allows my to generate the 7M signal from there by XORing them. Do you think this is a problem?

c0pperdragon, 2021-01-27:
Yes, the XOR is there to generate a short pulse to let the flip-flops latch the data on the rising and the falling clock edges. 
And yes, the U1C adds a tiny bit of delay to the PiCLK line. This is to make sure there is a sufficiently long setup-time for the data bits going into the Pi (the sampling loop on the Pi relies on this specific behaviour).

The inverter step on the V2 board is there to compensate for the one pixel by which the display would be offset when using the SuperDenise and the 7MHz clock input. 
The only case where this makes an actual difference is when using a 1280x1024 monitor. The default configuration is fine-tuned to fit the standard Worbench with pixel precision. Without this inverter, additional software adjustments would be necessary to again get the display to the correct position.

The Small board needs to be connected to the correct clock source (CDAC or 7MHz) during install and the optional inverter does the same thing as explained above.

jan-hofmeier, 2021-01-27:
Thanks for your fast response!

My mistake was that I somehow had in mind that PiCLK was the also the pulse (I don't konw why, because it's so obviously wrong). 
So yeah this all now makes sense to me. Thanks.

So I should be fine using the XOR of /C1 and /C3 as this gives me 7M inverted.

jan-hofmeier, 2021-01-28:
Oh one last question @c0pperdragon 
Why do V1 and V2 use an extra 3.3V regulator and nit just the 3.3v provided by the Pi, like the small one does?