hoglet67/RGBtoHDMI #182: No Issue just thanks

SuloxyA500, 2021-02-04:
If this post does not belong here, please delete it.

I just want to thank you for your work.
I installed it on the pi zero and connected it to my A500.
It works like a charme, right out of the box, no problems at all, even on my 4k display!

Best Regards
Suloxy

abrugsch, 2021-02-20:
Same here. Much thanks for this great project!

bluearcus, 2021-02-28:
Absolutely... this has transformed the output options for several of my machines already, and more still to come!

bkodenkt, 2021-06-17:
Absolute game changer for connecting my Amigas to current TVs and displays. Thank for this!

jdryyz, 2022-01-24:
Agreed! Fantastic project.

jdryyz, 2022-01-27:
I have to add that I also discovered that my A2000 RGB2HDMI system was two versions behind. Now that I have updated to the latest, the image quality is even better! It solved something I was going to ask about and then added features I was interested in as well. So awesome!