hoglet67/RGBtoHDMI #185: Make auto NTSC mode switch on non-monochrome UV for YUV profiles

bluearcus, 2021-02-14:
Auto NTSC artefact detection is done on Color burst. Where the interface is running in YUV mode, can assessment of UV values not be used to automatically enable / disable the mode based on monochrome or colour output?

IanSB, 2021-05-22:
@bluearcus 

This has been implemented in beta26 but has only had limited testing so can you give it a try:
https://github.com/IanSB/RGBtoHDMI/releases

IF U and V are zero it will automatically switch artifacting on when the Auto artifact option is selected which is now the default on the CoCo 1 & 2 profiles. It is not the default on the Dragon but could be made so if you think it would benefit.

The NTSC artifact colours have been tweaked as well
As usual, wipe the SD card first, don't overwrite existing files as the palettes will not get updated.

IanSB, 2021-08-04:
Addressed in https://github.com/IanSB/RGBtoHDMI/commit/2835bf044c6e9f85cc2d2e28ceafaabe0660da44

closed 2021-08-04 by IanSB