hoglet67/RGBtoHDMI #188: White screen ?!

francouai, 2021-02-22:
This project is crazy. Thank you to the team.
I'm re-using my Amiga500+ with Vampire since both are now HDMI.

I was playing with the game "Jim Power" and suddenly, my screen started to be white like snowing from the bottom then fully white.
I rebooted my A500+ and getting directly this same white screen.. 🥵
I was thinking to a corrupt config or bad config so I make a fresh micro SD card. Nothing changed. 
Denise chip is still working fine via the RGB video output port, so no problem at all with it.
But picture is white/corrupted via the HDMI port

you can see the problem here: 
https://youtu.be/FXDz4afmoY4

Any idea will be welcomed .

francouai, 2021-02-22:
due to unknown reason, my RPI burned.. very strange because it worked only 20mns.
I changed the RPI to a brand new one and it is all fixed.

benur75, 2021-02-22:
Good evening Francouai, I looked at your video. Don't you forget to remove your kickstart chip ? When using a vampire, you have to remove the KS chip.
bye

francouai, 2021-02-22:
Hi Benur75; my KS chip has been removed long time ago.

IanSB, 2021-03-14:
@benur75 

One possibility that just occurred to me:
PIN 18 (GPIO 24) on the Pi is used to provide a text overlay input to support FFOSD. 
When this pin is taken high it makes the screen output white and that is used to draw white text on the screen.
The Pi is normally configured to pull this input low but if that isn't working on your problem Pi then the input would be floating and randomly switch high resulting in a white screen or low resulting in normal operation.
You could work around that by soldering a 47K resistor from Pin 18 to ground.

See this thread for more details:
https://github.com/c0pperdragon/Amiga-Digital-Video/issues/26