hoglet67/RGBtoHDMI #190: Test pattern for R1 board and A500 rev 5

turbidvisions, 2021-03-08:
On boot of my A500 rev 5 with the original RGBtoHDMI adapterboard and a Pi Zero W, I see only a test pattern with four squares (red, yellow, cyan and blue).  Monochrome video through the composite connector works fine.  I've tried both the latest stable release (20200504_d19df1c) and the most recent prelease (20210131_20ce5f0) with the same results.

Thanks!

IanSB, 2021-03-12:
@turbidvisions 

Assuming that the SD card is written correctly, the most likely cause is that the PiCLK signal is missing. If you built the board yourself you need to make sure the link is fitted and check the 74LVC86 soldering and if possible look at the PiCLK signal on Pin 11 (GPIO17) with a scope.

To check the Pi is booting up, you can connect a USB TTL serial port to the TXD/RXD lines on the Pi (115200 Baud) and view the debug log in a terminal program.

kp73, 2021-03-12:
I get the same test pattern then a black screen (no response from pressing a momentary switch connected to the board), also a rev 5 motherboard A500.  I'm using this board from ebay https://www.ebay.co.uk/itm/RGB-to-HDMI-adapter-board-for-OCS-Commodore-Amiga-500/114719151179?hash=item1ab5cb544b:g:MU8AAOSwF6dgBrPQ.   I have a USB to TTL serial lead somewhere so I'll give that a try.

kp73, 2021-03-13:
I got this from the debug log, it all seemed reasonable;

`     Actual 100 lines = 6400126 ns
          Clock error = 35 PPM
 Error adjusted clock = 14187079 Hz
 Target PLL frequency = 2383429272 Hz, prediv = 1, PER = 4
 Actual PLL frequency = 2383429272 Hz
        GPCLK Divisor = 7
      Lines per frame = 313, (312.999)
Actual frame time = 20032350 ns (non-interlaced), line time = 64001 ns
Window: H=63680 to 64318, V=19931840 to 20131534, S=Comp, HW=4788, HT=9000
Width or Height differ from last FB: Setting dummy 64x64 framebuffer
Overscan L=240, R=240, T=0, B=0
Initialised Framebuffer
Size: 720x270 (requested 720x270)
Framebuffer address: DE000000
Modified display list word at 00000664 = 4F407801
RAM benchmark: Main memory = 798924 ns, Screen memory = 25515415 ns
Detected screen size = 1920x1080
Pitch=1440, width=720, height=270, sizex2=0, bpp=16
chars=89, nlines=270, hoffset=100, voffset=28, ncapture=-1
palctrl=0, samplewidth=5, hadjust=0, vadjust=0, sync=0x4
detsync=0x4, vsync=1, video=1, ntsc=0, border=0, delay=1
*PLL
Locked`

So I thought to try another monitor, and it just worked, debug output is very similar;

`     Actual 100 lines = 6400115 ns
          Clock error = 33 PPM
 Error adjusted clock = 14187103 Hz
 Target PLL frequency = 2383433360 Hz, prediv = 1, PER = 4
 Actual PLL frequency = 2383433360 Hz
        GPCLK Divisor = 7
      Lines per frame = 313, (313)
Actual frame time = 20032352 ns (non-interlaced), line time = 64001 ns
Window: H=63680 to 64318, V=19931840 to 20131534, S=Comp, HW=4795, HT=9000
Width or Height differ from last FB: Setting dummy 64x64 framebuffer
Overscan L=240, R=240, T=0, B=0
Initialised Framebuffer
Size: 720x270 (requested 720x270)
Framebuffer address: DE000000
Modified display list word at 00000994 = 4F407801
RAM benchmark: Main memory = 798882 ns, Screen memory = 25520973 ns
Detected screen size = 1920x1080
Pitch=1440, width=720, height=270, sizex2=0, bpp=16
chars=89, nlines=270, hoffset=100, voffset=28, ncapture=-1
palctrl=0, samplewidth=5, hadjust=0, vadjust=0, sync=0x4
detsync=0x4, vsync=1, video=1, ntsc=0, border=0, delay=1
Locked
*PLL`

I'll try and figure out why it isn't working with the first monitor.  Any ideas?

kp73, 2021-03-13:
More complete log from the monitor with the black screen..

RGB to HDMI booted
Simple board detected
Using PLLA as the sampling clock
PLLA: 2400.000000 ANA1 = 00144000
PLLA: PDIV=1 NDIV=62 CTRL=0002103e FRAC=524288 DSI0=256 CORE=2 PER=256 CCP2=256
PLLB: 2000.000024 ANA1 = 00144000
PLLB: PDIV=1 NDIV=52 CTRL=00021034 FRAC=87382 ARM=2 SP0=256 SP1=256 SP2=256
PLLC: 2400.000000, ANA1 = 00144000
PLLC: PDIV=1 NDIV=62 CTRL=0002103e FRAC=524288 CORE2=256 CORE1=256 PER=2 CORE0=2
PLLD: 2000.000024 ANA1 = 00144000
PLLD: PDIV=1 NDIV=52 CTRL=00021034 FRAC=87382 DSI0=256 CORE=4 PER=4 DSI1=256
PLLH: 1485.000000 ANA1 = 0000000c
PLLH: PDIV=1 NDIV=77 CTRL=0002104d FRAC=360448 AUX=256 RCAL=256 PIX=1 STS=528383
PLLA: 2400.000000 ANA1 = 00144000
PLLA: PDIV=1 NDIV=62 CTRL=0002103e FRAC=524288 DSI0=256 CORE=256 PER=4 CCP2=256
CPU speed detected as: 1000 Mhz
EMMC: BCM2708 controller power-cycled
CPLD update file not detected
Keycount = 0
CPLD  Design: Simple
CPLD Version: 0.0
Loading file /Palettes/Atari_800_NTSC.bin
Loading file /Palettes/Atari_800_PAL.bin
Loading file /Palettes/Atom_MKII.bin
Loading file /Palettes/Atom_MKII_Full.bin
Loading file /Palettes/Atom_MKII_Plus.bin
Loading file /Palettes/Colour_Genie_N25.bin
Loading file /Palettes/Colour_Genie_S24.bin
Loading file /Palettes/Colour_Genie_S25.bin
Loading file /Palettes/Commodore_64.bin
Loading file /Palettes/Commodore_64_Rev1.bin
Loading file /Palettes/Dragon-CoCo.bin
Loading file /Palettes/Dragon-CoCo_Emu.bin
Loading file /Palettes/Dragon-CoCo_Full.bin
Loading file /Palettes/MDA-Hercules.bin
Loading file /Palettes/Mono_(2_level).bin
Loading file /Palettes/Mono_(3_level).bin
Loading file /Palettes/Mono_(4_level).bin
Loading file /Palettes/Mono_(6_level).bin
Loading file /Palettes/RGB.bin
Loading file /Palettes/RGBI.bin
Loading file /Palettes/RGBI_(CGA).bin
Loading file /Palettes/RGBI_(Spectrum).bin
Loading file /Palettes/RGBrgb_(Amstrad).bin
Loading file /Palettes/RGBrgb_(Spectrum).bin
Loading file /Palettes/RrGgBb_(EGA).bin
Loading file /Palettes/Spectrum_48K_9Col.bin
Loading file /Palettes/TI-99-4a.bin
Loading file /Palettes/Tea1002.bin
Loading file /Palettes/Test_4_Lvl_B_or_U.bin
Loading file /Palettes/Test_4_Lvl_G_or_Y.bin
Loading file /Palettes/Test_4_Lvl_R_or_V.bin
FOUND RESOLUTION: 640x480
FOUND RESOLUTION: 720x576
FOUND RESOLUTION: 800x480
FOUND RESOLUTION: 800x600
FOUND RESOLUTION: 1024x600
FOUND RESOLUTION: 1024x768
FOUND RESOLUTION: 1280x1024
FOUND RESOLUTION: 1280x720
FOUND RESOLUTION: 1280x800
FOUND RESOLUTION: 1360x768
FOUND RESOLUTION: 1440(1920)x1080
FOUND RESOLUTION: 1440x1080
FOUND RESOLUTION: 1440x900
FOUND RESOLUTION: 1600x1200
FOUND RESOLUTION: 1600x900
FOUND RESOLUTION: 1680x1050
[... 68 lines cut ...]
    clkinfo.clock_ppm = 5000 ppm
    Nominal 100 lines = 6399900 ns
     Actual 100 lines = 6400134 ns
          Clock error = 36 PPM
 Error adjusted clock = 14187061 Hz
 Target PLL frequency = 2383426276 Hz, prediv = 1, PER = 4
 Actual PLL frequency = 2383426276 Hz
        GPCLK Divisor = 7
      Lines per frame = 313, (313)
Actual frame time = 20032388 ns (non-interlaced), line time = 64001 ns
Window: H=63680 to 64318, V=19931840 to 20131534, S=Comp, HW=4733, HT=9000
Width or Height differ from last FB: Setting dummy 64x64 framebuffer
Overscan L=240, R=240, T=0, B=0
Initialised Framebuffer
Size: 720x270 (requested 720x270)
Framebuffer address: DE000000
Modified display list word at 00000664 = 4F407801
RAM benchmark: Main memory = 798813 ns, Screen memory = 25488075 ns
Detected screen size = 1920x1080
Pitch=1440, width=720, height=270, sizex2=0, bpp=16
chars=89, nlines=270, hoffset=100, voffset=28, ncapture=-1
palctrl=0, samplewidth=5, hadjust=0, vadjust=0, sync=0x4
detsync=0x4, vsync=1, video=1, ntsc=0, border=0, delay=1
*PLL
Locked

IanSB, 2021-03-14:
@kp73 

That monitor probably doesn't like the non-standard 50Hz output coming from the Pi zero in auto detect mode or alternatively it doesn't like the DVI compatible HDMI signal.

To check for the first case, hold down the menu button during power up and that will temporarily set the output to 60Hz and you might get an image to allow you to change the menu settings.
If that doesn't work then you will have to try changing the HDMI mode on the main menu from another monitor.

You can force a more standard 50Hz mode by changing the resolution to 1920x1080 and refresh to 50Hz in the main menu.

Even then it will still be slightly off frequency due to the genlock so it that doesn't work the next option to try is disabling the genlock in the settings menu but that will mean the output is not locked to the input and you will get frame drops or repeats. (Save the config after changing this setting)

kp73, 2021-03-14:
@IanSB thanks very much for the help, it was 50Hz output.  Switched to 60 and it's all good with the old monitor.  @turbidvisions is it possible that the same thing is causing your issue?  Apologies anyway for hijacking your issue a bit, at least you know that at least one other rev 5 Amiga 500 works, and the 50Hz output isn't specific to the rev 5 Amiga motherboard.

turbidvisions, 2021-03-17:
@kp73, @IanSB Thanks for your help!  I'll definitely try playing with output frequency and report back.

shred, 2021-03-20:
I found this issue because I had the same problem with the latest stable release (20200504_d19df1c). I tested on different monitors, but only got a 60 Hz picture with four colored rectangles. The RasPi also didn't react on button presses.

However, it was resolved for me after I updated to the most recent prelease (20210131_20ce5f0).

turbidvisions, 2021-03-20:
Still no luck on my side, trying again the 20210131_20ce5f0 release, but this time capturing the debug log with looks similar to that of @kp73 except for large reported clock error:

          Clock error = 611664 PPM
WARN: PPM error too large, using nominal clock
 Error adjusted clock = 14187580 Hz

[debug_log.txt](https://github.com/hoglet67/RGBtoHDMI/files/6175851/debug_log.txt)

IanSB, 2021-03-21:
@turbidvisions 

The old stable release doesn't support the Amiga board so you must use the latest dev release 20210131_20ce5f0 or later.

The debug log shows detected video lines as 103uS long with 102uS hsync pulses instead of  64uS & 4.7uS respectively. It also shows the number of lines per frame detected as 316 instead of 313 which indicates very messed up sync signals so there maybe a problem with the board construction. 

You need to check the soldering around the pixel clock and sync signals as they pass through the board and if possible look at those signals with an oscilloscope.