hoglet67/RGBtoHDMI #195: Feature request:  Configure OSD position

derkommutator, 2021-03-25:
I'm using RGB2HDMI in an installation where part of the screen is obscured by a bezel.  This is no problem for the video itself, because it can be freely positioned via the min/max settings.  But the OSD menu is cut off on the left hand side.  I can see enough to navigate the menus with a bit of effort, but it would be nice if there were a way to specify an offset for the OSD.

IanSB, 2021-04-16:
I'll look at adding something in a future version but in the meantime you may be able to achieve the same result by editing the config file.

Load config.txt from the SD card into a text editor and add the following lines at the bottom of the # Common video settings block:

overscan_left=10
overscan_right=10
overscan_top=10
overscan_bottom=10

and save the file. This will reduce the active screen area by 10 pixels all around and you can experiment by varying the sizes further.

reboot the Pi after re-inserting the SD card.

derkommutator, 2021-04-17:
Ah, I hadn't realised those settings exist.  Honestly those are a better approach to my situation anyway, since all four borders of the screen are occluded and will never be used for anything.  Better to make them completely off limits.  Thanks!

derkommutator, 2021-04-26:
Unfortunately, this workaround does not work, because while it does reduce the active screen area for the input, it does not affect the menu positioning.  I'll reopen this issue so that it may remain a feature request for later consideration.  It's not a dealbreaker problem for me, just makes working the menus a little more ... challenging.  ;)

derkommutator, 2021-04-26:
(Ignore open/close there.  I managed to confuse myself (not the first time).  This workaround does the job.

closed 2021-04-26 by derkommutator

Gun-neR, 2025-07-11:
Updated, but opps... I probably should have left it alone :P  Back to the flashing screen with GunFright VDC and can't seem to fix it.  I will open up a new issue with details.