hoglet67/RGBtoHDMI #196: pixel sparkles on A2000 V1

gold11-carca, 2021-03-26:
HI,
I assembled a V1.0 version for Amiga 2000 and I have problems
bright pixel only on the black when he is lined with white,
A number of pixel sparkles.
With an A500 with version V2 the same cable and the same monitor
The image is clean. a solution ?

LinuxJedi, 2021-03-27:
Hi,

Make sure the jumper is set correctly (Denise / Super Denise) based on the model you have. I have also observed one A2000 rev 4.4 that needed to be jumpered to Denise for a Super Denise to get rid of sparkles.

Other people have recommended:

- Use 74VHC for U1/U5 instead of 74LVC
- Add a 47pF capacitor between GPIO pins 9 and 11

I've also opened PR #197 which is a design I have found more reliable.

LinuxJedi, 2021-03-27:
See also: c0pperdragon/Amiga-Digital-Video#41

closed 2021-06-17 by IanSB