hoglet67/RGBtoHDMI #200: FF OSD on RGBtoHDMI CPLD version by LinuxJedi

solarmon, 2021-04-17:
Hi,

I suppose this is really a question for @IanSB

I'm currently testing @LinuxJedi 's A500 RGBtoHDMI CPLD board (https://github.com/LinuxJedi/AmigaRGBtoHDMI)

Whilst it is working,  I am also trying to get FF OSD working on it. I can get it working using c0pperdragon 's RGBtoHDMI board, but it seems the LinuxJedi RGBtoHDMI CPLD board does not seem to work.

Using the same FF OSD setup and the the same Raspberry Pi, I'm connecting to Pin 18 (GPIO24) and using the sync signal from Denise on pin 32 - this works with the c0pperdragon RGBtoHDMI board, but not the LinuxJedi RGBtoHDMI CPLD board.

[image]

Does there need to be any extra/special setup/configuration to support FF OSD when using the CPLD version of the Raspberry Pi  firmware?

IanSB, 2021-04-17:
FF_OSD isn't currently supported in CPLD mode on the release software.
One potential problem is that the FF_OSD GPIO (24) is also used to program the CPLD so there might be a conflict.
I have now implemented it in order to test if there are any problems. This also contains other fixes for single button mode.
You will still need the 1K resistor to stop the FF_OSD signal interfering with CPLD programming.

Download the beta18.zip from my fork and let me know if it works OK for you:
https://github.com/IanSB/RGBtoHDMI/releases/tag/3313630

Note this is a standard software release so you will need to edit the file /profiles/default.txt and change to "single_button_mode=1" to force single button mode and also change the profile line to "Amiga" in the main menu
To test a clean install, delete the file:
/cpld_firmware/Delete_This_File_To_Erase_CPLD.txt
This will force the CPLD to be erased on the next reboot and so you can test recovery mode.
It would also be useful to test CPLD programming when there is FF_OSD text on the screen.

solarmon, 2021-04-18:
@IanSB 

Thanks! I'm out all day, so I'll try the beta later on.

BTW, what other button modes are there? What GPIO pins could be used for more buttons? I'm going to do a re-spin of the CPLD board and can add more buttons to give the user more options. Controller the menu with more buttons will be a better user experience.

LinuxJedi, 2021-04-18:
> BTW, what other button modes are there? What GPIO pins could be used for more buttons? I'm going to do a re-spin of the CPLD board and can add more buttons to give the user more options. Controlling the menu with more buttons will be a better user experience.

The default is three button mode. So you have:

* SW1 - GPIO16 (what we currently use)
* SW2 - GPIO26
* SW3 - GPIO19

They are pulled high so short to ground to activate. You then have up, down and select buttons. The buttons also do more things when the menu is off, such as SW1 turns scanlines on/off.

I originally designed my CPLD version to have the three buttons but reduced it to one when IanSB told me it was possible so that it was familiar to people who used the c0pperdragon design. My A2000 CPLD design (currently untested) gives the option of using either 1 button or 2 button mode.

solarmon, 2021-04-18:
@IanSB 

I tried beta18, and the first time I tried it on a brand new SD card, deleting the "Delete_This_File_To_Erase_CPLD.txt" file, I initially saw the screen that says it is trying to recover and was rebooting, but then the screen was just blank and flickering every now and then, and every so often I could see a white text but could not read it quickly enough.

I tried again and deleted everything on the SD card and boped over the beat18 files again, without deleting the "Delete_This_File_To_Erase_CPLD.txt" file. This time I could see the white text menu that asked to select the CPLD profile, which I did and chose the 12-bit RGB 92 one. It rebooted and once I then changed the profile from Amstrad to Amiga, it was all stable again.

I connected up FF OSD and it was indeed working! Woohoo!

Thank you! Now that I know FF OSD is working (albeit still wavy/jaggy) I can add that to the remix board I'm working on, along with the three buttons.

solarmon, 2021-04-19:
@IanSB Question regarding the buttons again. Would the same GPIO pins work as extra buttons in the original RGBtoHDMI (non CPLD) design?

IanSB, 2021-04-19:
@solarmon 
Yes, the non-CPLD design can be switched into 3 button mode but it is untested

The single_button_mode setting in default.txt is inverted for simple (c0pperdragon) mode so when editing it: 
single_button_mode=0 means 3 buttons for CPLD and 1 button for non-CPLD
single_button_mode=1 means 1 button for CPLD and 3 buttons for non-CPLD

There is a slightly updated beta19 here:
https://github.com/IanSB/RGBtoHDMI/releases/tag/fddfad7

This has slightly lower CPU usage when FF_OSD is not being used so can you check the FF OSD still works.

solarmon, 2021-04-20:
@IanSB I can confirm that FF OSD is still working with beta19

Regarding the buttons, maybe in the future, a rotary encoder mode could also be supported? This would make it even easier to navigate the menu.

LinuxJedi, 2021-04-20:
> Regarding the buttons, maybe in the future, a rotary encoder mode could also be supported? This would make it even easier to navigate the menu.

I like that idea. It would make things like profile editing when you need to change large numbers much easier.

IanSB, 2021-04-20:
@solarmon 

A rotary encode might be possible but it would have to use the GPIOs for the up & down buttons

IanSB, 2021-06-07:
Fixed in this release:

https://github.com/hoglet67/RGBtoHDMI/releases/tag/20210529_777ea4f

closed 2021-06-07 by IanSB