hoglet67/RGBtoHDMI #202: Sam Coupe Support?

Croccy22, 2021-04-20:
I totally understand the Sam is a very niche computer but having a HDMI solution for it would be amazing,

The Sam has a euroconnector interface on the back of and according to the the technical manual which can be found here: https://sam.speccy.cz/systech/sam-coupe_tech-man_v3-0.pdf (Page 6). The Sam has Red0&1, Green0&1, Blue0&1 and a Bright pin exposed via this connector alongside the Csync and CPU Clock signals.

I have done very minimal reading so far on how this project works but I assume the above means it is a 6bit RGB signal but with a bright pin to double the amount of colours similar to how the Spectrum did it?

Would there just be a way to wire these signals into the current hardware design and have a chance of it working or is it going to require someone with a lot of knowledge to work out how to interpret the signals?

IanSB, 2021-04-20:
Yes, it looks like it should work as you have 7 TTL bits available on that connector.

To add support, a custom 7 bit palette would need to be created that has the right RGB values for each of the 128 possible colour combinations (There are some palettes already in the system which support various 3, 4, and 6 bit colour tables but not 7 bit).
Once a palette has been created you would have to wire the 7 bits into the TTL header on the main 12 bit board to match the bit order in the palette.
A custom timing profile would also need to be created and you will need to find out the pixel clock rate for the various modes and other info like the number of pixel clock cycles per video line (including hsync period) although this can be determined visually if necessary.

You can manually create a palette table and add it to the SD card and there is some info on how to do that on the stardot thread but if you can find details of the RGB values for those 128 possible combinations, most likely from emulator source code I'll add the palette to the next release. If you can find out the other pixel clock info I should be able to create a good starting point for a profile but you would have to make the final tweaks on actual hardware.

Croccy22, 2021-04-21:
Sounds very promising. I have started doing some looking around at various resources. The SimCoupe Emulater has the source availble on Github and I have found a section all about timing (https://github.com/simonowen/simcoupe/blob/master/Base/SAM.h) which I don't know if that would be of any use. I've not found the palette information yet although I have found some references so I guess it is in there somewhere. There all seems to be quite a bit of info about the palette here: https://en.wikipedia.org/wiki/SAM_Coup%C3%A9

First things first I need to get some PCBs ordered and some components and get one of the 12bit boards made up :)

IanSB, 2021-04-21:
@Croccy22 
Yes that emulator source has all the info needed, the palette info is in SAMIO.cpp

IanSB, 2021-04-22:
@Croccy22 
>Sounds very promising

OK That's all implemented and speculative Sam Coupe profiles have been added for both the RGB and BBC CPLD versions.
This should be included in the next release due sometime in the coming week

At a minimum these will need adjustment of the horizontal and vertical offsets in the geometry menu to centre up the image plus an auto calibration.

>First things first I need to get some PCBs ordered and some components and get one of the 12bit boards made up :)

When you have built up your board, you will need connect to the 12 way RGBtoHDMI P2 header as follows:
RED1, GREEN1, BLUE1 on the euroconnector to R3, G3, B3 on P2
RED0, GREEN0, BLUE0 on the euroconnector to R2, G2, B2 on P2
BRIGHT on the euroconnector to R1 on P2
C SYNC on the euroconnector to Sync on P2
+5v, 0v on the euroconnector to +5v, 0v on P2
This leaves just Vsync and B1 unconnected on P2

If you get a good result, please post your final saved profile for inclusion in any future release

Croccy22, 2021-04-22:
Wow, thank you so much for this. I will update as soon as I can get the board built up. There will be an entire community really grateful for this :)

IanSB, 2021-05-06:
@Croccy22 

FYI the beta20 release (and later) on my fork has the initial Sam Coupe support:
https://github.com/IanSB/RGBtoHDMI/releases

Croccy22, 2021-05-07:
Thanks for the update. I have all the components here, just waiting for boards to arrive :)

Croccy22, 2021-05-18:
Hi, the boards arrived today. Put one together and connected it like you said with your latest firmware. And IT WORKS!!!! The picture quality is amazing!! There are no weird artifacts or anything. Possibly the white colour may have a pink tinge, but that may well be my TV picture mode. Thank you so much for doing this!! I'm going to design a PCB now with the euroconnector on it with a header to connect the RGBtoHDMI board, then put it in a case. The entire Sam community owes you one :)

Croccy22, 2021-05-18:
And the pinkness was my TV so all good :)

IanSB, 2021-05-18:
@Croccy22 

Good to hear it works.
Did you have to adjust the h and v offsets in the geometry menu to centre the image? If so, could you post the saved configuration file which will be in a sub folder in \Saved_Profiles.
Also could you post a couple of screencaps (press middle button - files will be on the SD card), perhaps one of the boot screen and one that looks graphically interesting and I'll add them to the gallery.

Croccy22, 2021-05-18:
I have now adjusted the vertical offset so the screen is perfectly centred :)

I can't actually get a screen grab of Sams boot screen because the Sam I am currently using has the trinity boot from (SD Card interface). But I have grabbed some screenshots from Prince of Persia and Lemmings because everyone loves those games!

The captures along with the Saved Profile can be grabbed from here: https://www.northdevonretroarchive.co.uk/wp-content/uploads/2021/05/RGBtoHDMI-SAM.zip

Croccy22, 2021-06-21:
Just a quick update. Everything so far has been working perfectly on my Sam. I have recently finished the design of an interface board which plugs directly into the Sam's euroconnector and presents the connections required for the RGBtoHDMI board. I've then also designed a 3D printed case to house it all in but this needs a re-design now as the new circuit board is smaller than my prototype.

Thanks again for all the work everyone involved has put into project

[image]