hoglet67/RGBtoHDMI #203: TI-99/4A status

Sch-LikA, 2021-05-07:
Hi again !

I built a "full" board including U7 to test TI-99/4A support. It is not far from working, I just have strange black artifacts but other than that it looks kinda good !

What is there left to have a fully working setup and how could I help ?

Thanks !

IanSB, 2021-05-07:
@Sch-LikA 

The analog board works by thresholding the incoming video signal at several user set levels so that any voltage above a preset level is a logic 1 and below it's a logic 0 and this is how it is able to extract a noise free image from a relatively noisy analog signal.
However it only works if there is a big enough voltage difference between the two levels so that it is not affected by noise or variations in the output levels between machines.

This seems to work well with most supported systems but it is very marginal with the the TI-99/4A as the levels from the TMS9929A video chip are just a little too close together. This results in the need to fine tune the voltage levels for each machine and reliability depends on how much noise is in the signal and how much those voltage levels drift with temperature.

I have included some test palettes to aid with setting up the voltage levels and you should try the following:
(When setting voltage levels it is a good idea to increase and decrease the levels until the image is incorrect and set the final value half way between those two settings)

1. Select the U7-TI-99-4A profile and ensure the TI startup screen is displayed:
Note the screen will probably blank after a while when you are making the adjustments but you can recall with QUIT (press FCTN and = keys)
[image]

2. Go into the palette menu and change the palette to "Test 4Lvl G or Y"
3. Go into the sampling menu and adjust the "Y Hi", "Y Mid" and "Y Lo" voltages until you get the following image:
Note the noisy blocks can be either of the colours shown but noisy may be better. (The noise isn't shown on the final palette)
Also The Hi voltage must be higher than the Mid voltage which must be higher than the Lo voltage.
[image]

4. Go into the palette menu and change the palette to "Test 4Lvl R or V"
5. Go into the sampling menu and adjust the "UV Hi", "UV Mid" and "V Lo" voltages until you get the following image:
Again The Hi voltage must be higher than the Mid voltage which must be higher than the Lo voltage.
[image]

6. Go into the palette menu and change the palette to "Test 4Lvl B or U"
7. Go into the sampling menu and adjust the "U Lo" voltage and maybe "UV Hi" & "UV Mid" until you get the following image:
Note the "UV Hi" and "UV Mid" settings are shared between "Test 4Lvl B or U" and "Test 4Lvl R or V" so if you have to adjust them to get a clean image you will need to go back to step 4 and check to see if it is still OK
[image]

 
8. Go into the palette menu and change the palette back to TI 99-4A
9. Go into the sampling menu and manually adjust the sampling phase for a noise free image (Auto calibrate isn't reliable due to the marginal signals).
10. Save the configuration in the main menu

You may have to repeat all the above to optimise but I can't guarantee that you will get a noise free solution as things are just too marginal. Also you may find that you get a good result when warmed up but it is a bit noisy at first when switched on from cold.

I think the only other options to improve this would be to modify the TI 99/4A to reduce any noise by giving the video chip it's own separate power supply and maybe connecting directly to the chip.

BTW the same video chip is used in MSX-1 machines so it might work with them as well but I haven't tested that.

Let me know how you get on.

Sch-LikA, 2021-05-26:
Wow !! These are what I call great instructions ! I have a 99% flicker-free setup now, thanks a lot ! Took some time to find the correct settings, but wow it looks awesome !

[image]
[image]

Never saw such a crisp image from any TI-99/4A !

Thanks again !!!

closed 2021-05-26 by Sch-LikA