hoglet67/RGBtoHDMI #205: The Amiga variable Scanlines are gone with the CPLD Beta 19/20 Firmaware

SirCathal, 2021-05-08:
@IanSB 
Hi Ian. I use the new CPLD Adapter inside my A1000 and all my sparkling pixel problems are gone.

But the wonderful option for the variable scanlines that you added to the firmware are gone.
There is no Amiga Variable Scanline Profile and the old ones don't work.
 
I can enable scanline in the menu, but the level can't be changed. The menu entry is there, but after entering the menu it cannot be changed. 
Manually adding the scanlinelevel to the configuration file dos not help

Can you please bring that feature back?
I really love soft scanlines around 80%

IanSB, 2021-05-08:
You could enable it manually by setting the overclock values and turning on double height framebuffer (FB Size) in the geometry menu. (You have to turn on double height framebuffer for variable scanlines to work but for the 12 bit Amiga modes that also requires the overclock settings)

In any case I've added the profile to Beta21:
https://github.com/IanSB/RGBtoHDMI/releases

SirCathal, 2021-05-10:
@IanSB 
Thank you, the Beta21 works fine with the variable scanline profile.
There is only one strange thing.
As long as I am in the Menu, I get some green dot patterns in the Amiga picture behind the menu.
When I quit the menu everything is perfect. so this is not a real problem, but I want to mention it.
I did not have that with Beta 20, but with Beta20 I hat no scanlines activated.

IanSB, 2021-05-10:
@SirCathal 

>As long as I am in the Menu, I get some green dot patterns in the Amiga picture behind the menu.

I have one of LinuxJedi's CPLD boards but I'm not seeing that problem.
Can you post a photo of the dots?
Do you get the dots with the standard Amiga profile?
What scanline intensity are you using?
Try adjusting the sampling phase in the sampling menu.
Also try adjusting the overclock settings.

IanSB, 2021-05-10:
@SirCathal 

It might be an overheating issue due to the overclocking required for variable scanlines. Do you have a heatsink on your Pi zero? If not, I suggest adding one.

closed 2021-05-24 by IanSB