hoglet67/RGBtoHDMI #206: HDMI-monitor blinking / display not usable as long as the Amiga (3000) system is not in workbench mode.

LouisAtGH, 2021-05-12:
Hello,
I combined The RGB to HDMI card with a Amiga 3000 having kickstart 3.1 ROM 40.068.  It only  working in workbench mode, however not in other situations. My findings are:
-	The display is flashing each second during the amiga boot-sequence card as long as the workbench is not yet started;
-	The same behavior if the boot sequence fails;
-	Or the system is booted without any floppy- or hard- disk;
-	Or the system is started using a game disk.
During the “flashing” about 1 seconds on and about 1 seconds off the Amiga is running properly and you can even control the Amiga (but it is very hard due to the intense flicker)
And strange enough, if I activate the RGBtoHDMI control menu …….. the flicker stops and the Amiga screen behaves normal. The flicker returns as soon as I leave the (Raspberry) control menu.
I tried to fix the problem by changing settings in the Raspberry setup and also, trying the problem by using a different Denise jumper setting and switching A3000 flicker fixer on and off. Nothing works. I am back to defaults (50Hz Amiga). 
These findings are observed using latest software (20210322_f771e51 Release). I own the board for just two weeks and do not have any experience with earlier software versions. 
I am sure it can be fixed in software, since the simple thing as calling the Raspberry setup menu is enough to let the problem disappear.
I hope this behavior triggers a bell and someone can fix it.
Sincerely,

Louis

IanSB, 2021-05-12:
Try the latest beta:
https://github.com/IanSB/RGBtoHDMI/releases

LouisAtGH, 2021-05-13:
Thanks!

 �

That solves the problem !!

 �

 �

Louis

 �

 �

https://github.com/IanSB/RGBtoHDMI/releases

Zucky16, 2021-05-13:
Want to add my experience as well. Very similar situation and results.

System: Amiga 500 NTSC/PAL - I'm in Canada, so NTSC standard and 60Hz
Using RGBtoHDMI in NTSC all good no problems at all
Switch to PAL and screen flicker/blink with black screen every 1 second or so, made no difference booting from disk or SCSI same result
Press the menu button on the RGBtoHDMI, screen refresh is confirmed at 50Hz and is stable including faded Amiga 500 output in background, so long as the menu is on the screen all is good
Leave menu, back to flashing again
No amount of config seemed to help the matter at all

Following the direction of using the latest beta solved it, perfect PAL modes can switch to NTSC and back again, no problems
**_Installed Beta21 and everything is great._**

Thanks to all the devs and people putting time into this project it is a miracle!

IanSB, 2021-05-13:
This is caused by the machine having an NTSC crystal fitted but being configured to output a PAL signal. Because the crystal is the wrong frequency for PAL all the line and frame timings are wrong so it didn't match the profile. I added an extra profile with these slightly wrong timings.

IanSB, 2021-06-07:
Fixed in this stable release:

https://github.com/hoglet67/RGBtoHDMI/releases/tag/20210529_777ea4f

closed 2021-06-07 by IanSB