hoglet67/RGBtoHDMI #208: Amiga 2000 CPLD Calibration Problem

ILAHWWINC, 2021-05-22:
With my A2000 CPLD board and firmware beta25 i have flickering screen in hires interlaced screenmode. 

I can fix this using the auto calibration (save settings afterwards) and get a perfect picture. But after reset / power cycle the picture is flickering again. 

From this situation a reboot of the RGB2HDMI with the option in the settings 
removes the flickering.

What should i do to get a flickerfree screen without rebooting?

IanSB, 2021-05-22:
Can you use the save log option when you have the flicker and also when it is OK and post both log.txt files. Also a screencap of the problem would be useful.

IanSB, 2021-05-22:
I found a possible cause but as I don't have any suitable hardware I can't test it so please try the attached file:
[kernelrpi-interlace-change.zip](https://github.com/hoglet67/RGBtoHDMI/files/6527299/kernelrpi-interlace-change.zip)
Unzip and overwrite the existing kernel file on the SD card and reboot the Pi zero

ILAHWWINC, 2021-05-23:
@IanSB 
This kernal works for me :-). Many thanks.

IanSB, 2021-05-23:
@ILAHWWINC 

That change has been incorporated into the latest beta 27:
https://github.com/IanSB/RGBtoHDMI/releases

ILAHWWINC, 2021-05-23:
Very fast help. Problem solved.

closed 2021-05-23 by ILAHWWINC

LinuxJedi, 2021-05-23:
Can confirm this fixed the same issue for another user of my A2000 boards.